espressif/esp-adf #963: Lyratd MSC 2.2 "pipeline_recording_to_sdcard" sound noise

jinyz05, 2023-03-08:
1. adf master, idf v4.4
2. menuconfig with lyratd msc 2.2 and wav encoder
3. the rec.wav recorded about 1.7MB size
4. fetch and play the wav from sdcard, i got noise sound;

Log following:
I (28) boot: ESP-IDF v4.4.4-148-g4c2afac355-dirty 2nd stage bootloader
I (28) boot: compile time 13:17:21
I (29) boot: chip revision: v1.0
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (40) qio_mode: Enabling default flash chip QIO
I (45) boot.esp32: SPI Speed      : 80MHz
I (50) boot.esp32: SPI Mode       : QIO
I (55) boot.esp32: SPI Flash Size : 4MB
I (59) boot: Enabling RNG early entropy source...
I (65) boot: Partition Table:
I (68) boot: ## Label            Usage          Type ST Offset   Length
I (76) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (83) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (90) boot:  2 factory          factory app      00 00 00010000 00100000
I (98) boot: End of partition table
I (102) boot_comm: chip revision: 1, min. application chip revision: 0
I (109) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=5ba78h (375416) map
I (221) esp_image: segment 1: paddr=0006baa0 vaddr=3ffb0000 size=01f3ch (  7996) load
I (224) esp_image: segment 2: paddr=0006d9e4 vaddr=40080000 size=02634h (  9780) load
I (230) esp_image: segment 3: paddr=00070020 vaddr=400d0020 size=36758h (223064) map
I (297) esp_image: segment 4: paddr=000a6780 vaddr=40082634 size=117fch ( 71676) load
I (330) boot: Loaded app from partition at offset 0x10000
I (330) boot: Disabling RNG early entropy source...
I (341) psram: This chip is ESP32-D0WD
I (342) spiram: Found 64MBit SPI RAM device
I (342) spiram: SPI RAM mode: flash 80m sram 80m
I (345) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (352) cpu_start: Pro cpu up.
I (356) cpu_start: Starting app cpu, entry point is 0x400815bc

I (0) cpu_start: App cpu up.
I (855) spiram: SPI SRAM memory test OK
I (863) cpu_start: Pro cpu start user code
I (863) cpu_start: cpu freq: 240000000
I (863) cpu_start: Application information:
I (866) cpu_start: Project name:     recording_to_sdcard
I (872) cpu_start: App version:      v2.5-4-ga535ca9-dirty
[image]

I (878) cpu_start: Compile time:     Mar  7 2023 13:20:48
I (884) cpu_start: ELF file SHA256:  50105f7383587a8a...
I (890) cpu_start: ESP-IDF:          v4.4.4-148-g4c2afac355-dirty
I (897) heap_init: Initializing. RAM available for dynamic allocation:
I (904) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (910) heap_init: At 3FFB2AF8 len 0002D508 (181 KiB): DRAM
I (917) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (923) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (929) heap_init: At 40093E30 len 0000C1D0 (48 KiB): IRAM
I (936) spiram: Adding pool of 4095K of external SPI memory to heap allocator
I (944) spi_flash: detected chip: generic
I (948) spi_flash: flash io: qio
I (953) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (968) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (976) RECORD_TO_SDCARD: [ 1 ] Mount sdcard
I (1482) RECORD_TO_SDCARD: [ 2 ] Start codec chip
W (2482) SPI: MCS ret:0,Status:1
I (2483) RECORD_TO_SDCARD: [3.0] Create audio pipeline for recording
I (2483) RECORD_TO_SDCARD: [3.1] Create fatfs stream to write data to sdcard
I (2490) RECORD_TO_SDCARD: [3.2] Create i2s stream to read audio data from codec chip
I (2500) RECORD_TO_SDCARD: [3.3] Create audio encoder to handle data
I (2506) RECORD_TO_SDCARD: [3.4] Register all elements to audio pipeline
I (2513) RECORD_TO_SDCARD: [3.5] Link it together [codec_chip]-->i2s_stream-->audio_encoder-->fatfs_stream-->[sdcard]
I (2525) RECORD_TO_SDCARD: [3.6] Set music info to fatfs
I (2530) RECORD_TO_SDCARD: [ * ] Save the recording info to the fatfs stream writer, sample_rates=44100, bits=16, ch=2
I (2542) RECORD_TO_SDCARD: [3.7] Set up  uri
I (2547) RECORD_TO_SDCARD: [ 4 ] Set up  event listener
I (2553) RECORD_TO_SDCARD: [4.1] Listening event from pipeline
I (2559) RECORD_TO_SDCARD: [4.2] Listening event from peripherals
I (2566) RECORD_TO_SDCARD: [ 5 ] Start audio_pipeline
I (2574) RECORD_TO_SDCARD: [ 6 ] Listen for all pipeline events, record for 10 Seconds
I (3628) RECORD_TO_SDCARD: [ * ] Recording ... 1
I (4628) RECORD_TO_SDCARD: [ * ] Recording ... 2
I (5631) RECORD_TO_SDCARD: [ * ] Recording ... 3
I (6687) RECORD_TO_SDCARD: [ * ] Recording ... 4
I (7896) RECORD_TO_SDCARD: [ * ] Recording ... 5
I (8896) RECORD_TO_SDCARD: [ * ] Recording ... 6
I (9896) RECORD_TO_SDCARD: [ * ] Recording ... 7
I (10896) RECORD_TO_SDCARD: [ * ] Recording ... 8
I (11898) RECORD_TO_SDCARD: [ * ] Recording ... 9
I (12899) RECORD_TO_SDCARD: [ * ] Recording ... 10
W (12912) RECORD_TO_SDCARD: [ * ] Stop event received
I (12912) RECORD_TO_SDCARD: [ 7 ] Stop audio_pipeline
E (12912) AUDIO_ELEMENT: [wav] Element already stopped
E (12915) AUDIO_ELEMENT: audio_element_process_running
E (12924) AUDIO_ELEMENT: [file] Element already stopped
W (12930) AUDIO_PIPELINE: There are no listener registered
W (12936) AUDIO_ELEMENT: [file] Element has not create when AUDIO_ELEMENT_TERMINATE
W (12944) AUDIO_ELEMENT: [i2s] Element has not create when AUDIO_ELEMENT_TERMINATE
W (12953) AUDIO_ELEMENT: [wav] Element has not create when AUDIO_ELEMENT_TERMINATE

jinyz05, 2023-03-08:
update with latest firmware using "tw_upload_dsp_firmware(-1)",  the wav file sound better, but also get noise;

jinyz05, 2023-03-08:
[update]
speech-recognition demo got no response;
player mp3 demo works, but the volume is too low;

yujong-lee, 2023-05-02:
I am experiencing the exact same problem.

MikeTolkachev, 2023-09-04:
I am experiencing the exact same problem with volume. 
I found strange behaviour. If you run the example mp3 player and then disconnect and reconnect one connector between boards A and B, then the sound volume will be high. After pressing the reset button, the sound will become quiet again.
Connector image: <img>https://ibb.co/0DDnBHT</img>