espressif/esp-adf #978: ESP-ADF Compatible with IDF 5.0.1?

DerPicknicker, 2023-04-03:
Hello, 

I use the ESP32-S3 for an Project with the newest IDF Version. But as you mentioned in the Github the Espressif Audio Development Framework is currently not usable with the newest IDF Version. Do you already have an Timeframe when the Audio Development Frame is compatible with the ESP IDF V5? 

Or is it already compatible but the Github Documentation is not updated yet? 

Best regards

DerPicknicker, 2023-04-10:
@jason-mao … Push … any updates regarding compatibility ?

Builty, 2023-04-11:
I also want to know this. I am doing a project with 5.0.1 and need the audio support. I don't want to roll back to old IDF if the new ADF is close. 

Please give a release timeline.

DerPicknicker, 2023-04-11:
Yeah I feel you. Here was mentioned in a couple of weeks. But this was in feb 2023... But still no updates on the ADF side..... 

https://github.com/espressif/esp-adf/issues/952#issuecomment-1445903783

jason-mao, 2023-04-11:
Hi guys, Thank you for your patience. We are still working on this, the latest status is that the modificaitons final review is on going, it will take one or two weeks.

DerPicknicker, 2023-04-11:
Thank you for your updates.

zafeer-birde, 2023-04-26:
@jason-mao Is this activity now complete ? What timelines are we looking at now ?

MoritzSchwoerer, 2023-05-08:
Any news about this topic?

DerPicknicker, 2023-05-09:
@jason-mao … any updates..?

jason-mao, 2023-05-10:
Hi guys, thank you for being so patient. This feature support on c1e884871872bc748a0248746fcba4980170356f

closed 2023-05-10 by jason-mao