espressif/esp-adf #980: Can 'v2.5/examples/protocols/voip' make support ESP32-LyraT too?

EdwardRyu, 2023-04-06:
Hi,

Our company is working on the upgrade SW from ADFv2.4 to ADFv2.5.
Our product is based on ESP32-LyraT.
And we had implemented VOIP using the example of release/v2.4/examples/advanced_examples)/voip/.
Now we are planning to upgrade VOIP SW to use the example of v2.5/examples/protocols/voip along with ADFv2.5.

When looking into Compatibility tables in below link, however, protocols/voip doesn't seem to be supported by ESP32-LyraT.

 https://github.com/espressif/esp-adf/blob/v2.5/examples/README.md#compatibility-of-examples-with-espressif-audio-boards

In case of ADFv2.4, examples/advanced_examples)/voip supports both ESP32-LyraT and ESP32-LyraT-Mini.

https://github.com/espressif/esp-adf/blob/release/v2.4/examples/README.md

Questions are :

Q1. What does make this limit in case of the product based on ESP32-LyraT.
Q2. Can you make  possible for 'v2.5/examples/protocols/voip' to support HW based on ESP32-LyraT as well?

Thank you.

ahhfzhang, 2023-04-13:
Hi @EdwardRyu 

The VoIP example in ADF V2.5 supports ESP32-LyraT, but most VoIP applications require AEC feature, and the AEC effect of ESP32-LyraT is not very good. When using one codec for ADC&DAC, we recommend using ES8311, which has internal playback channel return capabilities. This will result in better processing performance for the AEC algorithm.

Thanks.

EdwardRyu, 2023-04-13:
Thank you, Michael for your update.
Okay, I will try with ES8311 to see if it works fine.

closed 2023-06-06 by EdwardRyu