espressif/esp-adf #981: report an i2c_bus.c mutex bug

HikerPan, 2023-04-07:
Sorry, my english is poor.

My ESP-IDF version is V4.4.4
In my project, I use two i2c port.
I2C_NUM_0 for ES8311 I2C, It's define by ESP-ADF.
And I also have an touch chip use I2C_NUM_1.
The below code is part of my touch chip source code.
I need to init it and deinit it at some times.
And while my code deinit the I2C_NUM_1 and at the same time i use I2C_NUM_0.
The programe crashed.
`
void app_touch_i2c_init(int scl, int sda)
{
i2c_config_t es_i2c_cfg = {
.mode = I2C_MODE_MASTER,
.sda_pullup_en = GPIO_PULLUP_ENABLE,
.scl_pullup_en = GPIO_PULLUP_ENABLE,
.master.clk_speed = 100000,
};

es_i2c_cfg.scl_io_num = scl;
es_i2c_cfg.sda_io_num = sda;
touch_i2c = i2c_bus_create(I2C_NUM_1, &es_i2c_cfg);
}

void app_touch_i2c_deinit()
{
i2c_bus_delete(touch_i2c);
touch_i2c = NULL;
}
`

I check the crash information.
And found the _busLock is NULL at i2c_bus.c
assert failed: xQueueSemaphoreTake queue.c:1545 (( pxQueue ))
So I think the next code may be the root cause.
When i call deinit. it will call this delete and destory _busLock.
But I2C_NUM_0 still need to use it.
So, thre crash happened.

`
esp_err_t i2c_bus_delete(i2c_bus_handle_t bus)
{
    I2C_BUS_CHECK(bus != NULL, "Handle error", ESP_FAIL);
    i2c_bus_t *p_bus = (i2c_bus_t *) bus;
    if (--p_bus->ref_count == 0) {
        i2c_driver_delete(p_bus->i2c_port);
        i2c_bus[p_bus->i2c_port] = NULL;
        audio_free(p_bus);
        mutex_destroy(_busLock);
        _busLock = NULL;
    }
    return ESP_OK;
}
`
chang to：
`
esp_err_t i2c_bus_delete(i2c_bus_handle_t bus)
{
    I2C_BUS_CHECK(bus != NULL, "Handle error", ESP_FAIL);
    i2c_bus_t *p_bus = (i2c_bus_t *) bus;
    if (--p_bus->ref_count == 0) {
        i2c_driver_delete(p_bus->i2c_port);
        i2c_bus[p_bus->i2c_port] = NULL;
        audio_free(p_bus);
        // mutex_destroy(_busLock);
        // _busLock = NULL;
    }
    return ESP_OK;
}
`

TempoTian, 2023-04-07:
yes, After review code, This is an issue.
we always use only one I2C bus to connect multiple device, so have not meet such issue yet.
You can remove the delete bus action to keep resource to make function work as a workaround.
We will fix it soon.

HikerPan, 2023-04-10:
@TempoTian Thanks for your replay.

closed 2023-04-10 by HikerPan