espressif/esp-adf #987: error: implicit declaration of function 'esp_blufi_close'; did you mean 'esp_wifi_stop'?

TAMHAN, 2023-04-12:
## Environment

- Audio development kit:      [ESP32-LyraT]
- Audio kit version (for ESP32-LyraT/ESP32-LyraT-Mini/ESP32-S3-Korvo-2): [v4]
- [Required] Module or chip used:  [ESP32-S2-WROVER]
- [Required] IDF version (run ``git describe --tags`` in $IDF_PATH folder to find it):
    // v4.4-191-g83daa6dabb

tamhan@TAMHAN18:~/esp4/esp-adf/examples/advanced_examples/wifi_bt_ble_coex$ cd $IDF_PATH
tamhan@TAMHAN18:~/esp4/esp-idf$ git describe --tags
v4.4.4

- [Required] ADF version (run ``git describe --tags`` in $ADF_PATH folder to find it):
    // v2.3-171-gaac5912

tamhan@TAMHAN18:~/esp4/esp-idf$ cd $ADF_PATH
tamhan@TAMHAN18:~/esp4/esp-adf$ git describe --tags
v2.2-92-g59e5e02

- Build system:         [idf.py]
- [Required] Running log:         All logs from power-on to problem recurrence
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` in your project folder to find it):
    // 1.22.0-80-g6c4433a
- Operating system:     [Linux]
- Using an IDE?: [No]
- Power supply:         [external 5V]

## Problem Description

// Detailed problem description goes here.
I attempt to use Bluetooth in an ADF project, and it fails with a BluFi error. This happens both in my own project and in the example ~/esp4/esp-adf/examples/advanced_examples/wifi_bt_ble_coex.

### Expected Behavior
Attempt to compile example ~/esp4/esp-adf/examples/advanced_examples/wifi_bt_ble_coex and get binary file.

### Actual Behavior
Compile fails

### Steps to Reproduce

tamhan@TAMHAN18:~/esp4/esp-adf/examples/advanced_examples/wifi_bt_ble_coex$ idf.py build
Executing action: all (aliases: build)
Running ninja in directory /home/tamhan/esp4/esp-adf/examples/advanced_examples/wifi_bt_ble_coex/build
Executing "ninja all"...
[0/1] Re-running CMake...
-- Building ESP-IDF components for target esp32
Processing 1 dependencies:
[1/1] idf (4.4.4)
-- DEBUG: Use esp-modbus component folder: /home/tamhan/esp4/esp-idf/components/freemodbus.
-- Project sdkconfig file /home/tamhan/esp4/esp-adf/examples/advanced_examples/wifi_bt_ble_coex/sdkconfig
Loading defaults file /home/tamhan/esp4/esp-adf/examples/advanced_examples/wifi_bt_ble_coex/sdkconfig.defaults...
/home/tamhan/esp4/esp-adf/examples/advanced_examples/wifi_bt_ble_coex/sdkconfig.defaults:8 CONFIG_BTDM_CONTROLLER_MODE_BTDM was replaced with CONFIG_BTDM_CTRL_MODE_BTDM
/home/tamhan/esp4/esp-adf/examples/advanced_examples/wifi_bt_ble_coex/sdkconfig.defaults:9 CONFIG_BTDM_CONTROLLER_BLE_MAX_CONN was replaced with CONFIG_BTDM_CTRL_BLE_MAX_CONN
/home/tamhan/esp4/esp-adf/examples/advanced_examples/wifi_bt_ble_coex/sdkconfig.defaults:10 CONFIG_BTDM_CONTROLLER_BR_EDR_MAX_ACL_CONN was replaced with CONFIG_BTDM_CTRL_BR_EDR_MAX_ACL_CONN
/home/tamhan/esp4/esp-adf/examples/advanced_examples/wifi_bt_ble_coex/sdkconfig.defaults:11 CONFIG_BTDM_CONTROLLER_BR_EDR_MAX_SYNC_CONN was replaced with CONFIG_BTDM_CTRL_BR_EDR_MAX_SYNC_CONN
/home/tamhan/esp4/esp-adf/examples/advanced_examples/wifi_bt_ble_coex/sdkconfig.defaults:13 CONFIG_BTDM_CONTROLLER_BLE_MAX_CONN_EFF was replaced with CONFIG_BTDM_CTRL_BLE_MAX_CONN_EFF
/home/tamhan/esp4/esp-adf/examples/advanced_examples/wifi_bt_ble_coex/sdkconfig.defaults:14 CONFIG_BTDM_CONTROLLER_BR_EDR_MAX_ACL_CONN_EFF was replaced with CONFIG_BTDM_CTRL_BR_EDR_MAX_ACL_CONN_EFF
/home/tamhan/esp4/esp-adf/examples/advanced_examples/wifi_bt_ble_coex/sdkconfig.defaults:15 CONFIG_BTDM_CONTROLLER_BR_EDR_MAX_SYNC_CONN_EFF was replaced with CONFIG_BTDM_CTRL_BR_EDR_MAX_SYNC_CONN_EFF
/home/tamhan/esp4/esp-adf/examples/advanced_examples/wifi_bt_ble_coex/sdkconfig.defaults:17 CONFIG_BTDM_CONTROLLER_PINNED_TO_CORE was replaced with CONFIG_BTDM_CTRL_PINNED_TO_CORE
/home/tamhan/esp4/esp-adf/examples/advanced_examples/wifi_bt_ble_coex/sdkconfig.defaults:18 CONFIG_BTDM_CONTROLLER_HCI_MODE_VHCI was replaced with CONFIG_BTDM_CTRL_HCI_MODE_VHCI
/home/tamhan/esp4/esp-adf/examples/advanced_examples/wifi_bt_ble_coex/sdkconfig.defaults:19 CONFIG_BLE_SCAN_DUPLICATE was replaced with CONFIG_BTDM_BLE_SCAN_DUPL
/home/tamhan/esp4/esp-adf/examples/advanced_examples/wifi_bt_ble_coex/sdkconfig.defaults:20 CONFIG_SCAN_DUPLICATE_BY_DEVICE_ADDR was replaced with CONFIG_BTDM_SCAN_DUPL_TYPE_DEVICE
/home/tamhan/esp4/esp-adf/examples/advanced_examples/wifi_bt_ble_coex/sdkconfig.defaults:21 CONFIG_SCAN_DUPLICATE_TYPE was replaced with CONFIG_BTDM_SCAN_DUPL_TYPE
/home/tamhan/esp4/esp-adf/examples/advanced_examples/wifi_bt_ble_coex/sdkconfig.defaults:22 CONFIG_DUPLICATE_SCAN_CACHE_SIZE was replaced with CONFIG_BTDM_SCAN_DUPL_CACHE_SIZE
/home/tamhan/esp4/esp-adf/examples/advanced_examples/wifi_bt_ble_coex/sdkconfig.defaults:23 CONFIG_BTDM_CONTROLLER_FULL_SCAN_SUPPORTED was replaced with CONFIG_BTDM_CTRL_FULL_SCAN_SUPPORTED
/home/tamhan/esp4/esp-adf/examples/advanced_examples/wifi_bt_ble_coex/sdkconfig.defaults:24 CONFIG_BLE_ADV_REPORT_FLOW_CONTROL_SUPPORTED was replaced with CONFIG_BTDM_BLE_ADV_REPORT_FLOW_CTRL_SUPP
/home/tamhan/esp4/esp-adf/examples/advanced_examples/wifi_bt_ble_coex/sdkconfig.defaults:25 CONFIG_BLE_ADV_REPORT_FLOW_CONTROL_NUM was replaced with CONFIG_BTDM_BLE_ADV_REPORT_FLOW_CTRL_NUM
/home/tamhan/esp4/esp-adf/examples/advanced_examples/wifi_bt_ble_coex/sdkconfig.defaults:26 CONFIG_BLE_ADV_REPORT_DISCARD_THRSHOLD was replaced with CONFIG_BTDM_BLE_ADV_REPORT_DISCARD_THRSHOLD
/home/tamhan/esp4/esp-adf/examples/advanced_examples/wifi_bt_ble_coex/sdkconfig.defaults:27 CONFIG_BLUEDROID_ENABLED was replaced with CONFIG_BT_BLUEDROID_ENABLED
/home/tamhan/esp4/esp-adf/examples/advanced_examples/wifi_bt_ble_coex/sdkconfig.defaults:28 CONFIG_BLUEDROID_PINNED_TO_CORE_1 was replaced with CONFIG_BT_BLUEDROID_PINNED_TO_CORE_1
[... 231 lines cut ...]

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here. It should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plaintext here for us to search the error log. Or attach the complete logs and leave the main part here if the log is *too* long.
```

## Other Items If Possible

- [ ] sdkconfig file (Attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**Note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

TAMHAN, 2023-04-12:
[sdkconfig.zip](https://github.com/espressif/esp-adf/files/11215983/sdkconfig.zip)

TAMHAN, 2023-04-12:
I managed to get it compiling by changing this: 

tamhan@TAMHAN18:~/esp4/esp-adf/play_mp3_experiment$ gedit /home/tamhan/esp4/esp-adf/components/esp_peripherals/lib/blufi/wifibleconfig.c

        case ESP_BLUFI_EVENT_RECV_SLAVE_DISCONNECT_BLE:
            ESP_LOGI(WIFI_BLE_TAG, "BLUFI close a gatt connection");
            //esp_blufi_close(g_wifi_ble_config->ble_server_if, g_wifi_ble_config->ble_conn_id);
            break;
            
            
            

tamhan@TAMHAN18:~/esp4/esp-adf/play_mp3_bluetooth$ gedit /home/tamhan/esp4/esp-adf/components/wifi_service/blufi_config/blufi_config.c
            
        case ESP_BLUFI_EVENT_RECV_SLAVE_DISCONNECT_BLE:
            ESP_LOGI(TAG, "BLUFI close a gatt connection");
            //esp_blufi_close(cfg->ble_server_if, cfg->ble_conn_id);
            break;

jason-mao, 2023-06-09:
@TAMHAN I saw your ADF is v2.3, would please update to latest ADF version to check? I build ADF v2.5 with IDF v4.4.4 and v4.4.5 no same issue in there.

TAMHAN, 2023-07-08:
Hello, sorry for being slow. This is a bit problematic, my client would not appreciate - the code works for us. But still, big thanks for your willingness to help!

closed 2023-09-05 by jason-mao