espressif/esp-adf #992: How to play audio from different sources like spiffs stream / a2dp stream as different modes ?

eltxaditya, 2023-04-25:
Hi !
I wanted to have two these modes in my device which can be switched with a push button
1. Mode 1 where files are played from flash memory spiffs stream(bluetooth is off)
2. Mode 2 where bluetooth is turned on and a2dp stream is used
I found examples for both, but I would like to know the best route to switch between these.
What confuses me is that do I need to make two pipelines and play one at a time or should the whole pipeline and other components like event listener etc be deleted altogether and initialized a fresh to start a new pipeline?
Is there a standard route to achieve this.
Any help appreciated.
Thanks

jason-mao, 2023-05-11:
@eltxaditya You can refer the https://github.com/espressif/esp-adf/tree/master/examples/korvo_du1906/components/audio_player

eltxaditya, 2023-05-11:
Thanks a lot.