espressif/esp-adf #995: build error

brilliantWorks, 2023-05-16:
hi there~

Audio development kit: Custom board
Module or chip used: ESP32
ADF version : v2.5

examples/get-started/play_mp3_control 
When I build, I get an error like below.
I just selected "customboard" in menuconfig.
If you do not select "customboard" in menuconfig, no issues will occur.

----------------------------------------------------------------------------------------------------------
.obj   -c /opt/Espressif/ESP32/esp-adf-v2.5/components/audio_hal/driver/es8388/es8388.c
/opt/Espressif/ESP32/esp-adf-v2.5/components/audio_hal/driver/es8388/es8388.c: In function 'es8388_init':
/opt/Espressif/ESP32/esp-adf-v2.5/components/audio_hal/driver/es8388/es8388.c:43:22: error: 'BOARD_PA_GAIN' undeclared (first use in this function)
     .board_pa_gain = BOARD_PA_GAIN,                         \
                      ^~~~~~~~~~~~~
/opt/Espressif/ESP32/esp-adf-v2.5/components/audio_hal/driver/es8388/es8388.c:340:41: note: in expansion of macro 'ES8388_DAC_VOL_CFG_DEFAULT'
     codec_dac_volume_config_t vol_cfg = ES8388_DAC_VOL_CFG_DEFAULT();
                                         ^~~~~~~~~~~~~~~~~~~~~~~~~~
/opt/Espressif/ESP32/esp-adf-v2.5/components/audio_hal/driver/es8388/es8388.c:43:22: note: each undeclared identifier is reported only once for each function it appears in
     .board_pa_gain = BOARD_PA_GAIN,                         \
                      ^~~~~~~~~~~~~
/opt/Espressif/ESP32/esp-adf-v2.5/components/audio_hal/driver/es8388/es8388.c:340:41: note: in expansion of macro 'ES8388_DAC_VOL_CFG_DEFAULT'
     codec_dac_volume_config_t vol_cfg = ES8388_DAC_VOL_CFG_DEFAULT();
                                         ^~~~~~~~~~~~~~~~~~~~~~~~~~
[1142/1228] Building C object esp-idf/audio_hal/CMakeFiles/__idf_audio_hal.dir/driver/es7210/es7210.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

texruska, 2023-05-29:
Your custom board is missing some definitions that seem to be required (for now). My minimal working example looks like this:

```
#ifndef _AUDIO_BOARD_DEFINITION_H_
#define _AUDIO_BOARD_DEFINITION_H_

extern audio_hal_func_t AUDIO_NEW_CODEC_DEFAULT_HANDLE;

#define AUDIO_CODEC_DEFAULT_CONFIG(){                   \
        .adc_input  = AUDIO_HAL_ADC_INPUT_LINE1,        \
        .dac_output = AUDIO_HAL_DAC_OUTPUT_ALL,         \
        .codec_mode = AUDIO_HAL_CODEC_MODE_BOTH,        \
        .i2s_iface = {                                  \
            .mode = AUDIO_HAL_MODE_SLAVE,               \
            .fmt = AUDIO_HAL_I2S_NORMAL,                \
            .samples = AUDIO_HAL_48K_SAMPLES,           \
            .bits = AUDIO_HAL_BIT_LENGTH_16BITS,        \
        },                                              \
};

#endif
```

texruska, 2023-05-29:
This is related to https://github.com/espressif/esp-adf/pull/1008

For now you can solve this problem by having your `board_def.h` to contain this following definition:

```#ifndef _AUDIO_BOARD_DEFINITION_H_
#define _AUDIO_BOARD_DEFINITION_H_

#define BOARD_PA_GAIN 0

#endif
```

closed 2023-06-07 by espressif-bot