espressif/esp-adf #996: audio gap when using esp_audio_seek

robbietree8, 2023-05-16:
## Environment

- Audio development kit:      none
- Audio kit version (for ESP32-LyraT/ESP32-LyraT-Mini/ESP32-S3-Korvo-2): none
- [Required] Module or chip used:  ESP32-PICO-V3-02
- [Required] IDF version (run ``git describe --tags`` in $IDF_PATH folder to find it):
    v4.4.3-210-g932b6699ac
- [Required] ADF version (run ``git describe --tags`` in $ADF_PATH folder to find it):
    v2.4.1-12-g81bac5f
- Build system:         idf.py
- [Required] Running log:         All logs from power-on to problem recurrence
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` in your project folder to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch5) 8.4.0
- Operating system:    macOS
- Using an IDE?: VSCode
- Power supply:         Battery

## Problem Description

I have one mp3 file, and I want to loop play it without delay.

Currently, I'm using `esp_audio_seek` to seek to beginning when playing time is closed to end.

But unfortunatelly, I still see the gap, from the log, almost 500ms

### Expected Behavior

### Actual Behavior

### Steps to Reproduce

### Code to Reproduce This Issue

```cpp
static void esp_audio_state_task (void *para)
{
    QueueHandle_t que = (QueueHandle_t) para;
    esp_audio_state_t esp_state = {0};
    while (1) {
        int time = 0;
        int duration = 0;
        esp_audio_time_get(player, &time);
        esp_audio_duration_get(player, &duration);
        ESP_LOGI(TAG, "[ * ] End of time:%d ms, duration:%d ms", time, duration);

        xQueueReceive(que, &esp_state, 1000 / portTICK_PERIOD_MS);
        ESP_LOGI(TAG, "esp_audio status:%x,err:%x\n", esp_state.status, esp_state.err_msg);

        if(duration > 0 && duration - time < 5000) {
            ESP_LOGI(TAG, "seek to start");
            esp_audio_seek(player, 5);
        }
    }
    vTaskDelete(NULL);
}
```

## Debug Logs

```
I (57302) app_music: seek to start
I (57302) ESP_AUDIO_CTRL: Enter media_ctrl_seek_by_time procedure, src:0, st:RUNNING, pos:(135060 byte), t:(5 seconds)
I (57312) ESP_AUDIO_CTRL: Seek with RUNNING status, total bytes:1624586
W (57322) AUDIO_ELEMENT: OUT-[Audio_forge] AEL_IO_ABORT
W (57322) AUDIO_ELEMENT: OUT-[DEC_mp3] AEL_IO_ABORT
W (57332) AUDIO_ELEMENT: OUT-[Audio_forge] AEL_IO_ABORT
W (57332) MP3_DECODER: output aborted -3
W (57332) AUDIO_ELEMENT: OUT-[Audio_forge] AEL_IO_ABORT
I (57342) MP3_DECODER: Closed
W (57352) AUDIO_ELEMENT: OUT-[Audio_forge] AEL_IO_ABORT
I (57352) ESP_AUDIO_TASK: Received last pos: 1533928 bytes
I (57352) AUDIO_FORGE: audio forge closed
I (57362) ESP_AUDIO_TASK: Recv Element[DEC_mp3-0x3f800e7c] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_STOPPED
I (57382) ESP_AUDIO_TASK: Received last time: 56548 ms
I (57382) ESP_AUDIO_TASK: Recv Element[OUT_iis-0x3f80116c] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_STOPPED
I (57392) ESP_AUDIO_TASK: Recv Element[Audio_forge-0x3f8012e0] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_STOPPED
W (57322) AUDIO_ELEMENT: OUT-[IN_spiffs] AEL_IO_ABORT
I (57412) ESP_AUDIO_TASK: Recv Element[IN_spiffs-0x3f800bb8] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_STOPPED
W (57422) ESP_AUDIO_TASK: Ignore the element's status[AEL_STATUS_STATE_STOPPED] with seeking process
I (57432) ESP_AUDIO_TASK: Recv Element[IN_spiffs-0x3f800bb8] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_STOPPED
I (57442) AUDIO_ELEMENT: [IN_spiffs] AEL_MSG_CMD_RESUME,state:5
I (57452) AUDIO_ELEMENT: [DEC_mp3] AEL_MSG_CMD_RESUME,state:1
I (57462) AUDIO_ELEMENT: [OUT_iis] AEL_MSG_CMD_RESUME,state:1
I (57462) I2S_STREAM: AUDIO_STREAM_WRITER
I (57462) AUDIO_ELEMENT: [Audio_forge] AEL_MSG_CMD_RESUME,state:1
I (57472) AUDIO_FORGE: audio_forge opened
I (57472) ESP_AUDIO_TASK: Recv Element[OUT_iis-0x3f80116c] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_RUNNING
I (57462) MP3_DECODER: MP3 opened
I (57502) ESP_AUDIO_TASK: Recv Element[Audio_forge-0x3f8012e0] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_RUNNING
I (57472) SPIFFS_STREAM: File size is 1624586 byte, pos:135060
I (57642) ESP_AUDIO_TASK: Recv Element[IN_spiffs-0x3f800bb8] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_RUNNING
I (57752) ESP_AUDIO_TASK: Recv Element[DEC_mp3-0x3f800e7c] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_RUNNING
W (57752) ESP_AUDIO_TASK: Ignore the element's status[AEL_STATUS_STATE_RUNNING] with seeking process
I (57762) ESP_AUDIO_CTRL: Exit media_ctrl_seek_by_time procedure, src:0, st:RUNNING, pos:135060, ret:0
I (57772) ESP_AUDIO_CTRL: Total:1624586, bps:216000, duration:60029 ms, line 1718
```

## Other Items If Possible

- [ ] sdkconfig file (Attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**Note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

jason-mao, 2023-05-26:
@robbietree8 We will provide a new example to demonstrate how to do that.

robbietree8, 2023-05-26:
thanks @jason-mao very looking forward to it.

robbietree8, 2023-06-01:
@jason-mao any update?

maojxsir, 2023-06-09:
@robbietree8 We are working on this. It will be available next week.

jason-mao, 2023-06-09:
Related issue https://github.com/espressif/esp-adf/issues/699

jason-mao, 2023-06-14:
@robbietree8 The example is visible on 1c3b61a3462111427764be4fbb5a05080b994981

closed 2023-06-14 by jason-mao

robbietree8, 2023-06-15:
@jason-mao thanks, I have tried two pipelines solution, and it works good.

But I'm currently using esp audio api due to the requirement of coexistence with bluetooth music, is there any solution for this scenario?

maojxsir, 2023-06-15:
@robbietree8 `esp audio` is not support without gap loop playback. As your requirements, I suggest use the element and pipeline to implement, local music refer the `examples/player/pipeline_loop_playback_without_gap`, then add one bluetooth pipeline to connect to I2S stream element when you want to playing.

robbietree8, 2023-06-15:
@maojxsir @jason-mao 
There is something I missed in my last comment.

I'm using ble and classic bluetooth together, and I'm following this [example](https://github.com/espressif/esp-adf/tree/master/examples/advanced_examples/wifi_bt_ble_coex), in this example, it uses `esp audio` api.

So, what's your suggestion for this scenario?

jason-mao, 2023-06-15:
For audio playback you can based on `pipeline_loop_playback_without_gap` and  add BT stream to implement.  If you want to play BT audio, disconnect  ringbuffer of I2S_Writer from the local MP3 element output, then connect the  input ringbufer of  I2S_Writer to the  BT stream output. https://github.com/espressif/esp-adf/blob/1c3b61a3462111427764be4fbb5a05080b994981/examples/player/pipeline_loop_playback_without_gap/main/main.c#L161-L162, 

Like this you can switch between each other.

alanmosely, 2023-07-18:
Hi @jason-mao , thanks for `examples/player/pipeline_loop_playback_without_gap`, I've implemented it and continuous playback works but there is still a tiny but still noticeable gap when using ogg or mp3 files - the audio files I've tested with have no gaps at start or end of file.

My log output below shows everything working as expected. Would you expect this example to be able to achieve completely continuous playback?

```
W (91361) FATFS_STREAM: No more data, ret:0
I (91361) AUDIO_ELEMENT: IN-[file] AEL_IO_DONE,0
I (91521) AUDIO_ELEMENT: IN-[dec] AEL_IO_DONE,-2
I (91741) VORBIS_DECODER: Closed
I (91741) audio_player: [ * ] The Pipeline[0] has been finished. Switch to another one
I (91741) AUDIO_ELEMENT: [file-0x3de18e7c] Element task created
I (91751) AUDIO_THREAD: The dec task allocate stack on external memory
I (91761) AUDIO_ELEMENT: [dec-0x3de1b070] Element task created
I (91761) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:359, MEM Total:2012759 Bytes, Inter:72619 Bytes, Dram:72619 Bytes
I (91771) AUDIO_ELEMENT: [file] AEL_MSG_CMD_RESUME,state:1
I (91781) FATFS_STREAM: File size: 296371 byte, file position: 0
I (91791) AUDIO_ELEMENT: [dec] AEL_MSG_CMD_RESUME,state:1
I (91791) VORBIS_DECODER: VORBIS codW (91791) AUDIO_ELEMENT: [file] Element already stopped
W (91791) AUDIO_ELEMENT: [file] Element already stW (91801) AUDIO_ELEMENT: [dec] Element already stopped
I (91801) CODEC_ELEMENT_HELPER: The element is 0x3de1b070. The reserve data 2 is 0x0.
I (91821) VORBIS_DECODER: a new song playing
```

alanmosely, 2023-08-07:
How do you get audio file durations using this example, as no equivalent to `esp_audio_duration_get` seems to be available for elements and pipelines?