espressif/esp-adf #997: ESP ADF + IDF 5 with Custom Board Config

DerPicknicker, 2023-05-17:
Hello, 

first of all I would thank you @jason-mao and your team for the release of the latest release of ADF. It seems that everything works as expected. But now to my Problem. I don't have any Audio Development Board (Lyrat..). And I don't want to use them. I only have ESP32 and ESP32-S3 NodeMCU Boards:
ESP32: 
[image]
ESP32-S3:
[image]

I want to use them in combination of IDF and ADF. But for me it's not clear how to create a custom board config. And yes I read dozen of websites and the official documentation. 

@jason-mao : 
Is it possible to provide me the custom Board Files or even better add them to the GithubRepo. To have a starting point for developer who aren't using the Developer boards provided by Espressif... 

Best regards,

DerPicknicker, 2023-05-18:
> @DerPicknicker how about [play_mp3_control](https://github.com/espressif/esp-adf/tree/master/examples/get-started/play_mp3_control) example, it already contains custom board files.

Thank you for the Hint! But how to adapt it to the Nodemcu Boards..? Is there any Documentation about the progress? I don't want to use the Touch Pins in the ADF way. I want to have full control and do it only in IDF. I need the ADF only for decoding audio the rest should be done in IDF. 

For me the the missing part is: 
- How to adapt the Custom Board to the ESP32 / ESP32-S3 without any external Hardware attached
- How to use only certain parts of ADF with IDF in combination 

@robbietree8 Do you get my Point? 

Hope you have some more information 💯

robbietree8, 2023-05-18:
@DerPicknicker 

Touch pins is not necessary, you can use gpio/adc button to control audio.

For the second part:

For me, there is not so much burden to use adf, just simply include cmake file from adf path `include($ENV{ADF_PATH}/CMakeLists.txt)`, and only use it to deal with audio part. for the rest, idf is enough.

DerPicknicker, 2023-05-18:
@robbietree8 ... 

Thank you for your support. 

> For me, there is not so much burden to use adf, just simply include cmake file from adf path `include($ENV{ADF_PATH}/CMakeLists.txt)`, and only use it to deal with audio part. for the rest, idf is enough.

Yes this is already done. So for this I don't need a custom board config right? But what's about the Pin Configuration? 
Here I need to select a board but no board fits to my need. In my Code I use I2S and own PIN numbers. How does the Board selection in the screenshot below affect this? 
[image]

MANY THANKS!

DerPicknicker, 2023-05-19:
@robbietree8 .. 

After some reading I got the main concept (or I guess I got it..) .. I hope this is the last question: 
Do I need a custom board config IF I set the I2S Pins manually? Or does the custom not affect my program code if I call a function initI2S(23,33,34); (the numbers are the GPIO Pins)?

Or do I need to create a custom board config for a naked esp32/S3 to run properly? 

Best regards 👍

DerPicknicker, 2023-05-28:
@robbietree8 PUSH ;) ... 

I tried to get this Example running: https://github.com/espressif/esp-adf/tree/master/examples/player/pipeline_http_mp3 
I connected an I2S amplifier to the ESP32 Board. This doesn't work I didn't hear anything. I tested different Pins and I validated with an Arduino Library that my Hardware is working fine. 

So I tested now several pins with the Arduino code: https://github.com/schreibfaul1/ESP32-audioI2S/blob/master/examples/I2Saudio/I2Saudio.ino

I tested these Pins but sadly no Audio comes out of my Speaker. 

#define I2S_DOUT      22 // 22,25 works
#define I2S_BCLK      26 // 26,27 works
#define I2S_LRC       25  // 25,26 works

All these Pins mentioned above works with the Arduino Code without any Problems so my I2S Hardware is working fine. 

I tested also an custom Code which inits the I2S with the DEFAULT settings: 
```
i2s_stream_cfg_t i2s_cfg = I2S_STREAM_CFG_DEFAULT();
i2s_cfg.type = AUDIO_STREAM_WRITER;
i2s_stream_writer = i2s_stream_init(&i2s_cfg);
```

Any Hints from your Side...?

texruska, 2023-05-28:
It would help if you posted your code to a repo/jist so that we could read through it in full

DerPicknicker, 2023-05-29:
@texruska ... 

I can share my code but it's based on the example I linked in my reply above you. So I think it should be the best if we can find out why the example is not working on my esp32 board... I can guarantee that my i2s hardware is working fine. I tested it with the Arduino library and different pins..

DerPicknicker, 2023-05-29:
Hello, 

here is my custom code which based on the espressif example. Let me know if you found something out. Be aware that you need to change your WiFi password to get the code running. 
[mp3_radio_4.2.zip](https://github.com/espressif/esp-adf/files/11588806/mp3_radio_4.2.zip)

@texruska hope this helps and thank you for your time.

TempoTian, 2023-05-29:
It seems that after send data to I2S, sound can be played.
In common configuration adf need a codec which configure through i2c, and data is exchanged through I2S.
If you use only I2S, you can choose Custom board and only define I2S pin.
You can add such a function and change pin to your board setting and check whether it works or 
`esp_err_t get_i2s_pins(i2s_port_t port, i2s_pin_config_t *i2s_config)
{
    AUDIO_NULL_CHECK(TAG, i2s_config, return ESP_FAIL);
    if (port == I2S_NUM_0) {
        i2s_config->bck_io_num = GPIO_NUM_16;
        i2s_config->ws_io_num = GPIO_NUM_7;
        i2s_config->data_out_num = GPIO_NUM_6 ;
        i2s_config->data_in_num = GPIO_NUM_15;
        i2s_config->mck_io_num = GPIO_NUM_5;
    }  else {
        memset(i2s_config, -1, sizeof(i2s_pin_config_t));
        ESP_LOGE(TAG, "i2s port %d is not supported", port);
        return ESP_FAIL;
    }

    return ESP_OK;
}
`

texruska, 2023-05-29:
You didn't share with us the custom board config, which is the key to the problem I expect

TempoTian, 2023-05-29:
[image]

Select this board it will use custom board

DerPicknicker, 2023-05-29:
Hello @TempoTian & @texruska .. thank you for the information. 

And yes @TempoTian i don't need i2c I want to send the audio directly via i2s. With your code snippet I will try to create a custom board config. 

BTW: Do I need two configurations if I want use the second i2s peripheral for IDF Code..? 

Many thanks👍🏻

TempoTian, 2023-05-29:
If you have two I2s you can set you I2S pin inside get_i2s_pins and assign the pin setting when `if (port == I2S_NUM_1)`
And change the configuration to use port 1 by overwrite the default setting
```
i2s_stream_cfg_t i2s_cfg = I2S_STREAM_CFG_DEFAULT();
i2s_cfg.i2s_port= 1;
```

DerPicknicker, 2023-05-29:
@TempoTian .. Yes this is clear for me but DO I need this or this is optional and not required to get the code running? I have a code which uses i2s not for audio for custom hardware this code works fine in IDF without ADF added. My main question is do I need to configure this custom board for non ADF code as well 😉..?

texruska, 2023-05-29:
ADF gets the I2S pins from the board definition, so unless you're using the same pins for I2S as an existing one you'll have to make a "custom" board

Non-ADF code can do whatever it likes, there are many ways to solve this problem and ADF has chosen to use this pattern

DerPicknicker, 2023-05-29:
> ADF gets the I2S pins from the board definition

Thank you for the information. Are you working also with ADF? Can I come back to you if I have some other questions. I started from Arduino and coming now to IDF and ADF in combination it's quite hard to get an overview.. Sometimes I miss these information in the official documentation or I didn't see it 😉..

texruska, 2023-05-29:
Yeah I've just gone through the same struggles as you, I'm working on a project using I2S on custom hardware

DerPicknicker, 2023-05-29:
@texruska .. 
Thank you for your kind words. I will try to test the custom board tomorrow. Hope this will fix my problem and I can start to focus on my main development 😉 .. Sometime I miss the easiness of Arduino ... 

I will give you an update if my custom board solves all problems.

TempoTian, 2023-05-30:
In ADF audio board is not optinal for that some components depend on it, if not provided it will not build success.
If you do not use ADF, there is no such option in menuconfig, you do not need to provide it.

DerPicknicker, 2023-05-30:
Issue can be closed. 
Thank you all of you. It was mainly caused by the missing custom board configuration. Now my code runs with some adjustments totally fine.

stupidSheep, 2023-06-04:
Hi @DerPicknicker Would it be possible to share your customized version of https://github.com/espressif/esp-adf/tree/master/examples/player/pipeline_http_mp3 showing how to use/configure a custom board? (Perhaps having an extra commit with all the changes between the original example and your fork?)
I'm in the same boat as you (I've a Freenove ESP32 starter kit which I'd like to use with ESP-ADF)

Googling ESP ADF + custom board has few results, with this thread being at the very top :) 
Your help is very much appreciated :)

DerPicknicker, 2023-06-04:
@stupidSheep .. 

I can understand that you're interested in my code. But my Code based on the Example you linked. So there is nothing special.. I thought the Code Espressif is not working - **but this is not the case**. The Problems are the custom Board config. If you doesn't include it - **it won't work with your board**. 

The custom Board Config contains required Information about your Board: I2C Pinout, I2S Pinout, Which Decoder are supported etc. 

So you need to create a folder inside your "component" folder: `my_board`.. Copy the content of this custom Board from here : https://github.com/espressif/esp-adf/tree/master/examples/get-started/play_mp3_control into the created folder. 

Now you can adapt the i2S config in the`board_pins_config.c` file. 

After these steps you can change the board-config inside idf.py menuconfig as you can see here: 

> [image]

Hope this Helps!

texruska, 2023-06-04:
Here is an example that uses a custom board. Tinker with it in order to get yours working, you don't have to implement everything that the example does: 

https://github.com/espressif/esp-adf/tree/master/examples/get-started/play_mp3_control

stupidSheep, 2023-06-04:
Hi @texruska and @DerPicknicker Thank you both for your prompt responses :) I've just noticed that esp-adf's esp-idf version is behind my esp-idf and thus some of my existing code isn't running (ADC related). I'll wait a bit with esp-adf and try again once its esp-idf version is newer

closed 2023-09-05 by jason-mao

tank104, 2024-10-15:
This thread has been super helpful for me getting a custom board working.  thanks!!

DerPicknicker, 2024-10-15:
@tank104 great to hear - it's a pity that the documentation is not really helpful. Maybe they can add those information at some point

TempoTian, 2024-10-17:
I have write a simple script to gen a new board based on json configuration. You can put then into `components/audio_board` folder. After json file configured finished, Run` perl gen_board.pl` in the same folder, it will auto generate `CMakeLists.txt' and board code for you. Then you can use menuconfig to select your newly added board. Hope it help you to fast customized a new board. 
[gen_board.zip](https://github.com/user-attachments/files/17405075/gen_board.zip)

DerPicknicker, 2024-10-18:
@TempoTian this is great please include it also in the ADF release.

Thank you for still replying even for closed issues.