espressif/esp-adf #999: afe_perferred typo in esp_afe_config.h and algorithm_stream.c

Gaai, 2023-05-22:
Declared in esp_afe_config.h
```
typedef struct {
...<cut>
    int afe_perferred_core;
    int afe_perferred_priority;
...<cut>
} afe_config_t;
```

and used in algorithm_stream.c:
```
    afe_config.afe_perferred_core = 1;
    afe_config.afe_perferred_priority = 21;
```

Solution:
Replace "perferred" with "preferred"

If that's all the references to this variable I can do a pull request.

Gaai, 2023-05-24:
Another mistake, this time in audio-element.c: line 1297
`The index of ringbuffer is gather than and equal to ringbuffer maximum (...)`
Not sure what that was meant to mean. Greater than and equal to ringbuffer maximum (...)? But that can't be right. Maybe: greater than or equal to ringbuffer maximum?

jason-mao, 2023-05-26:
@Gaai Thank you for your report. We will fix that.

feizi, 2023-05-26:
> Replace "perferred" with "preferred"

Thanks, I will fix it.

jason-mao, 2023-09-05:
This topic will be fixed on ESP-SR repo.

closed 2023-09-05 by jason-mao