espressif/esp-adf #1001: wifi_bt_ble_coex example doesn't build with audio/stack_on_psram_v3.3 release

mph070770, 2023-05-23:
## Environment

- Audio development kit:      ESP32-LyraT
- Audio kit version for ESP32-LyraT
- [Required] Module or chip used:  ESP32-WROVER-E
- [Required] IDF version (run ``git describe --tags`` in $IDF_PATH folder to find it):
    //v3.3.1-276-g606dd2b629
- [Required] ADF version (run ``git describe --tags`` in $ADF_PATH folder to find it):
    // v2.5-26-ga74a26f
- Build system:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` in your project folder to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
- Operating system:     Windows
- (Windows only) Environment type: ESP Command Prompt
- Using an IDE?: No
- Power supply:         external 5V

## Problem Description

I'm trying to build the demo wifi_bt_ble_coex

I downloaded esp_idf.  Then checked out audio/stack_on_pstam_v3.3

ran install.bat

```
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, esp32ulp-elf, cmake, openocd-esp32, mconf, ninja, idf-exe, ccache
Skipping xtensa-esp32-elf@1.22.0-80-g6c4433a5-5.2.0 (already installed)
Skipping esp32ulp-elf@2.28.51.20170517 (already installed)
Skipping cmake@3.13.4 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20190313 (already installed)
Skipping mconf@v4.6.0.0-idf-20190628 (already installed)
Skipping ninja@1.9.0 (already installed)
Skipping idf-exe@1.0.1 (already installed)
Skipping ccache@3.7 (already installed)
Setting up Python environment
Creating a new Python environment in C:\Users\marti\.espressif\python_env\idf3.3_py3.7_env
created virtual environment CPython3.7.3.final.0-64 in 437ms
  creator CPython3Windows(dest=C:\Users\marti\.espressif\python_env\idf3.3_py3.7_env, clear=False, global=False)
  seeder FromAppData(download=False, pip=latest, setuptools=latest, wheel=latest, via=copy, app_data_dir=C:\Users\marti\AppData\Local\pypa\virtualenv\seed-app-data\v1)
  activators BashActivator,BatchActivator,FishActivator,PowerShellActivator,PythonActivator,XonshActivator
Installing Python packages from C:\Espressif\frameworks\esp-idf\requirements.txt
Requirement already satisfied: setuptools in c:\users\marti\.espressif\python_env\idf3.3_py3.7_env\lib\site-packages (from -r C:\Espressif\frameworks\esp-idf\requirements.txt (line 4)) (46.0.0)
Collecting pyserial>=3.0
  Using cached pyserial-3.5-py2.py3-none-any.whl (90 kB)
Processing c:\users\marti\appdata\local\pip\cache\wheels\fa\cd\1f\c6b7b50b564983bf3011e8fc75d06047ddc50c07f6e3660b00\future-0.18.3-py3-none-any.whl
Collecting cryptography>=2.1.4
  Using cached cryptography-40.0.2-cp36-abi3-win_amd64.whl (2.6 MB)
Collecting pyparsing<2.4.0,>=2.0.3
  Using cached pyparsing-2.3.1-py2.py3-none-any.whl (61 kB)
Collecting cffi>=1.12
  Using cached cffi-1.15.1-cp37-cp37m-win_amd64.whl (179 kB)
Collecting pycparser
  Using cached pycparser-2.21-py2.py3-none-any.whl (118 kB)
Installing collected packages: pyserial, future, pycparser, cffi, cryptography, pyparsing
Successfully installed cffi-1.15.1 cryptography-40.0.2 future-0.18.3 pycparser-2.21 pyparsing-2.3.1 pyserial-3.5
WARNING: You are using pip version 20.0.2; however, version 23.1.2 is available.
You should consider upgrading via the 'C:\Users\marti\.espressif\python_env\idf3.3_py3.7_env\Scripts\python.exe -m pip install --upgrade pip' command.
```

ran export.bat

```
C:\Espressif\frameworks\esp-idf>export
Setting IDF_PATH: C:\Espressif\frameworks\esp-idf

Adding ESP-IDF tools to PATH...
    C:\Users\marti\.espressif\tools\xtensa-esp32-elf\1.22.0-80-g6c4433a5-5.2.0\xtensa-esp32-elf\bin
    C:\Users\marti\.espressif\tools\esp32ulp-elf\2.28.51.20170517\esp32ulp-elf-binutils\bin
    C:\Users\marti\.espressif\tools\cmake\3.13.4\bin
    C:\Users\marti\.espressif\tools\openocd-esp32\v0.10.0-esp32-20190313\openocd-esp32\bin
    C:\Users\marti\.espressif\tools\mconf\v4.6.0.0-idf-20190628\
    C:\Users\marti\.espressif\tools\ninja\1.9.0\
    C:\Users\marti\.espressif\tools\idf-exe\1.0.1\
    C:\Users\marti\.espressif\tools\ccache\3.7\
    C:\Users\marti\.espressif\python_env\idf3.3_py3.7_env\Scripts
    C:\Espressif\frameworks\esp-idf\tools

Checking if Python packages are up to date...
Python requirements from C:\Espressif\frameworks\esp-idf\requirements.txt are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build
```

then I tried to run menuconfig:

```
C:\Espressif\projects\wifi_bt_ble_coex>idf.py menuconfig
Note: You are using Python 3.7.3. Python 3 support is new, please report any problems you encounter. Search for 'Setting the Python Interpreter' in the ESP-IDF docs if you want to use Python 2.7.
Checking Python dependencies...
Python requirements from C:\Espressif\frameworks\esp-idf\requirements.txt are satisfied.
Running cmake in directory C:\Espressif\projects\wifi_bt_ble_coex\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized C:\Espressif\projects\wifi_bt_ble_coex"...
Warn about uninitialized values.
fatal: not a git repository (or any of the parent directories): .git
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.21.0.windows.1")
-- IDF_TARGET not set, using default target: esp32
-- Building for target esp32
-- ccache will be used for faster builds
-- The C compiler identification is GNU 5.2.0
-- The CXX compiler identification is GNU 5.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/marti/.espressif/tools/xtensa-esp32-elf/1.22.0-80-g6c4433a5-5.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/marti/.espressif/tools/xtensa-esp32-elf/1.22.0-80-g6c4433a5-5.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/marti/.espressif/tools/xtensa-esp32-elf/1.22.0-80-g6c4433a5-5.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Users/marti/.espressif/tools/xtensa-esp32-elf/1.22.0-80-g6c4433a5-5.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: C:/Users/marti/.espressif/tools/xtensa-esp32-elf/1.22.0-80-g6c4433a5-5.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at C:/Espressif/frameworks/esp-idf/tools/cmake/component_utils.cmake:20 (message):
  Required component esp_netif is not found in any of the provided
  COMPONENT_DIRS
Call Stack (most recent call first):
  C:/Espressif/frameworks/esp-idf/tools/cmake/scripts/expand_requirements.cmake:95 (find_component_path)
  C:/Espressif/frameworks/esp-idf/tools/cmake/scripts/expand_requirements.cmake:114 (expand_component_requirements)
  C:/Espressif/frameworks/esp-idf/tools/cmake/scripts/expand_requirements.cmake:114 (expand_component_requirements)
  C:/Espressif/frameworks/esp-idf/tools/cmake/scripts/expand_requirements.cmake:114 (expand_component_requirements)
  C:/Espressif/frameworks/esp-idf/tools/cmake/scripts/expand_requirements.cmake:114 (expand_component_requirements)
  C:/Espressif/frameworks/esp-idf/tools/cmake/scripts/expand_requirements.cmake:217 (expand_component_requirements)

CMake Error at C:/Espressif/frameworks/esp-idf/tools/cmake/idf_functions.cmake:338 (message):
  Failed to expand component requirements
Call Stack (most recent call first):
  C:/Espressif/frameworks/esp-idf/tools/cmake/project.cmake:173 (idf_import_components)
  CMakeLists.txt:8 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Espressif/projects/wifi_bt_ble_coex/build/CMakeFiles/CMakeOutput.log".
See also "C:/Espressif/projects/wifi_bt_ble_coex/build/CMakeFiles/CMakeError.log".
cmake failed with exit code 1

```
netif doesn't seem to exist in audio/stack_on_psram_v3.3 release but the readme instructs to use this branch.  Please help.

marcolinux46, 2023-05-24:
Hi @mph070770,

i'm interested too at wifi_bt_ble_coex.
How can i compile the project with audio/stack_on_psram_v3.3?

Thanks for your support

jason-mao, 2023-05-26:
@mph070770 @marcolinux46 Let we check and then reply.

mph070770, 2023-06-01:
@jason-mao do you have any update please? Thanks

JosephTang, 2023-06-02:
@mph070770 update IDF to v4.4 is more recommend.

marcolinux46, 2023-06-02:
@JosephTang i have IDF to 4.4.4 but there are building problems yet

JosephTang, 2023-06-02:
@marcolinux46 build with idf v4.4.4 (e8bdaf91) is OK of my test, so could you show me the error message of build this example with idf v4.4.4

mph070770, 2023-06-06:
Thanks @JosephTang. The instructions state use stack_on_pstam_v3.3 so could you please update the build instructions.  Thanks for your help.

closed 2023-06-06 by mph070770