espressif/esp-adf #1002: esp_codec_dev I2S PHILIPS MODE

tore-espressif, 2023-05-25:
## Environment

- Audio development kit:      ESP32-S3-Korvo-2-LCD
- Audio kit version v3.1
- [Required] Module or chip used:  ESP32-S3-WROOM-1
- [Required] IDF version v5.1
- [Required] ADF version esp_codec_dev v1.0.1
- Build system:         idf.py
- [Required] Running log:        Log is OK
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` in your project folder to find it):
    // 1.22.0-80-g6c4433a
- Operating system:     Windows
- (Windows only) Environment type: Power Shell
- Using an IDE?: No
- Power supply:         USB

## Problem Description

I can see that esp_codec_dev always reconfigures my I2S slot to `MSB` mode https://github.com/espressif/esp-adf/blob/master/components/esp_codec_dev/platform/audio_codec_data_i2s.c#L75 How do I configure it for `PHILIPS` mode?

TempoTian, 2023-05-29:
You can change the mentioned line to i2s_std_slot_config_t slot_cfg = I2S_STD_PHILIPS_SLOT_DEFAULT_CONFIG(fs->bits_per_sample, fs->channel); and have a try.
We will commit a new verison of esp_codec_dev it will add support for TDM mode, the default STD slot config will change to PHILIP mode.

tore-espressif, 2023-05-31:
Thank you for the quick response @TempoTian !

Just an FYI: I am attaching my changes that I had to do to get playback working on ES8311 with various sampling frequencies:
https://github.com/espressif/esp-adf/compare/master...tore-espressif:fix/esp_codec_dev

TempoTian, 2023-05-31:
Thanks! Yes, you are right, add codec format setting is necessary, if use PHILIP mode, need not change to left align inside codec also.

closed 2023-06-19 by espressif-bot