espressif/esp-adf #1004: esp-adf build warning with IDF v5.1

espradio, 2023-05-28:
```
*******************************************************************************
[1181/1247] Building C object esp-idf/audio_sal/CMakeFiles/__idf_audio_sal.dir/audio_thread.c.obj/export/iradio/esp-adf/components/audio_sal/audio_thread.c: In function 'audio_thread_create':
/export/iradio/esp-adf/components/audio_sal/audio_thread.c:78:64: warning: passing argument 2 of 'xTaskCreateRestrictedPinnedToCore' from incompatible pointer type [-Wincompatible-pointer-types]
   78 |         if (xTaskCreateRestrictedPinnedToCore(&xRegParameters, (TASK_HANDLE_T)p_handle, core_id) != pdPASS) {
      |                                                                ^~~~~~~~~~~~~~~~~~~~~~~
      |                                                                |
      |                                                                struct tskTaskControlBlock *
/export/iradio/esp-adf/components/audio_sal/audio_thread.c:43:130: note: expected 'struct tskTaskControlBlock **' but argument is of type 'struct tskTaskControlBlock *'
   43 | BaseType_t __attribute__((weak)) xTaskCreateRestrictedPinnedToCore(const TaskParameters_t *const pxTaskDefinition, TaskHandle_t *pxCreatedTask, const BaseType_t xCoreID)
      |                                                                                                                    ~~~~~~~~~~~~~~^~~~~~~~~~~~~
/export/iradio/esp-adf/components/audio_sal/audio_thread.c:86:72: warning: passing argument 6 of 'xTaskCreatePinnedToCore' from incompatible pointer type [-Wincompatible-pointer-types]
   86 |         if (xTaskCreatePinnedToCore(main_func, name, stack, arg, prio, (xTaskHandle)p_handle, core_id) != pdPASS) {
      |                                                                        ^~~~~~~~~~~~~~~~~~~~~
      |                                                                        |
      |                                                                        struct tskTaskControlBlock *
In file included from /export/iradio/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/queue.h:49,
                 from /export/iradio/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/semphr.h:42,
                 from /export/iradio/esp-adf/components/audio_sal/audio_thread.c:26:
/export/iradio/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/task.h:372:62: note: expected 'struct tskTaskControlBlock ** const' but argument is of type 'struct tskTaskControlBlock *'
  372 |                                         TaskHandle_t * const pvCreatedTask,
```

texruska, 2023-05-28:
I also had this issue. To fix it (temporarily, until esp-adf is fixed properly) you need to enable FreeRTOS backwards compatibility mode (in menuconfig)

texruska, 2023-05-29:
The argument is expecting a pointer to those task handles, so no wonder there's a warning. PR to fix

closed 2023-06-07 by espressif-bot