espressif/esp-adf #1005: Wrong hardcoded pin in headphone_detect.c ?

Gaai, 2023-05-28:
```
int headphone_status_get()
{
    return gpio_get_level(0);
}
```
I think this `0` should either be replaced by macro `HEADPHONE_DETECT`, or `get_headphone_detect_gpio()` which returns this macro value??
Another possibillity is using `(int)pvTimerGetTimerID(xTimer)` like it's done in `static void hp_timer_cb(TimerHandle_t xTimer)`. 
Although that seems convoluted if there is a MACRO declared already that is indirectly used anyway. 

Also I noticed:
```
static char *TAG = "HEADPHONE";
```
Which I think should be:
```
static const char *TAG = "HEADPHONE";
```
Let me know if I should create a pull request.

Gaai, 2023-05-28:
Not at all critical but might aswell replace typo in headphone_detect.h:
```
#ifndef _AUDIO_HEADPHONE_DETEC_H_
#define _AUDIO_HEADPHONE_DETEC_H_
```
With:
```
#ifndef _AUDIO_HEADPHONE_DETECT_H_
#define _AUDIO_HEADPHONE_DETECT_H_
```

texruska, 2023-05-29:
I've noticed a few hardcoded numbers/assumptions, I suggest make a PR to get attention on the issue. If there's a problem with it then it can be figured out there

jason-mao, 2023-06-08:
@Gaai Thank you for your report, let we fix it.

closed 2023-06-14 by espressif-bot