espressif/esp-adf #1012: VOIP example AEC not working

Nathanael29, 2023-05-29:
I'm trying out the VOIP example using esp-adf master and esp-idf v4.4. The AEC doesnt seem to be working as I can still hear myself when calling the esp32 lyrat mini v1.2 with Linphone. The microphone on the lyrat mini seems to have really high gain as well. How do I fix the echo cancelling and adjust microphone gain?

ahhfzhang, 2023-06-05:
Hi @Nathanael29 

You can use https://github.com/espressif/esp-adf/tree/master/examples/advanced_examples/algorithm example to debug AEC feature, but there are two things you can try : 

1. remove the agc in algo_mask
2. lower the SPK volume

Thanks.

Nathanael29, 2023-06-05:
I need the speaker to be at max volume, is there a way to reduce the microphone gain?

ahhfzhang, 2023-06-06:
1. For esp32 lyrat mini v1.2 board, you can set ES7243 mic gain here https://github.com/espressif/esp-adf/blob/f06049175127bf491c6a763abc1136715c5ac134/components/audio_hal/driver/es7243/es7243.c#L146

2. remove AGC here https://github.com/espressif/esp-adf/blob/f06049175127bf491c6a763abc1136715c5ac134/components/audio_stream/include/algorithm_stream.h#L132

Nathanael29, 2023-06-07:
I still find the mic gain too high. The code doesn't handle negative gain What value do I write in the register to attenuate the microphone?

switch (volume) {
        case 0 ... 12:
            ret |= es7243_write_reg(0x08, 0x11); // 1db
            break;