espressif/esp-adf #1015: does audio-player support spiffs stream?

IamFive, 2023-06-02:
does audio-player support spiffs stream? if so, what is the correct audio file path?
or it only supports tone? what is the different between tone and spiffs.

spiffs stream reader is added to audio_player as blew:
```
  spiffs_stream_cfg_t flash_cfg = SPIFFS_STREAM_CFG_DEFAULT();
  flash_cfg.type = AUDIO_STREAM_READER;
  spiffs_stream_reader = spiffs_stream_init(&flash_cfg);
  ESP_RETURN_IF(spiffs_stream_reader == NULL, ESP_FAIL, TAG,
                "failed to init spiffs audio stream reader.");
  esp_audio_input_stream_add(audio_player, spiffs_stream_reader);
```

but when player audio in spiffs, it can not find the input stream and returns `ESP_ERR_AUDIO_NO_INPUT_STREAM`

```
// all those 3 uri is not support, 
char *uri = "file://spiffs/xxxx.wav";
// char *uri = "/spiffs/xxxx.wav";
// char *uri = "file://storage/spiffs/xxxx.wav";
esp_audio_sync_play(audio_player, uri, 0);
```

partition info:
```
storage,  data, spiffs,  ,        2M,
```

spiffs registered through
```
spiffs_create_partition_image(storage ../spiffs FLASH_IN_PROJECT)
```

shootao, 2023-06-02:
Hi @IamFive
You can use the patch in the attachment. Currently,  you need to do is mount the SPIFFS file system without initializing `spiffs_stream_init`
[0001-fatfs-Remove-fatfs-stream-s-restriction-on-sdcard-us.patch](https://github.com/espressif/esp-adf/files/11631639/0001-fatfs-Remove-fatfs-stream-s-restriction-on-sdcard-us.patch)

IamFive, 2023-06-02:
> Hi @IamFive You can use the patch in the attachment. Currently, you need to do is mount the SPIFFS file system without initializing `spiffs_stream_init` [...]

what do you mean by ` mount the SPIFFS file system without initializing`, do u mean we can use fatfs stream directly? any code sample for how to play spiffs files directly. 

we have mount spiffs at `app_main`:
```
 esp_vfs_spiffs_conf_t conf = {
      .base_path = SPIFFS_STORAGE_PATH,
      .partition_label = SPIFFS_STORAGE_LABEL,
      .max_files = SPIFFS_STORAGE_MAX_FILE,
      .format_if_mount_failed = false,
  };
  ESP_ERROR_CHECK(srv_periph_mount_spiffs(&conf));
```

shootao, 2023-06-02:
Hi @IamFive 
You can use the follow patch to test console_example, the console commad is `play file://spfiis/adf_music.mp3`
Tips: the spiffs data from [spiff play example](https://github.com/espressif/esp-adf/tree/master/examples/player/pipeline_spiffs_mp3/tools)
```c
diff --git a/examples/cli/main/console_example.c b/examples/cli/main/console_example.c
index d55a6d6b..265d77bc 100644
--- a/examples/cli/main/console_example.c
+++ b/examples/cli/main/console_example.c
@@ -40,6 +40,7 @@
 #include "wav_encoder.h"
 #include "display_service.h"
 #include "led_bar_is31x.h"
+#include "periph_spiffs.h"
 
 #include "sdcard_list.h"
 #include "sdcard_scan.h"
@@ -656,6 +657,28 @@ static void cli_setup_player(void)
 }
 // Example of initializing esp_audio as an audio player -- END
 
+static esp_err_t mount_spiffs(esp_periph_set_handle_t set)
+{
+    // Initialize Spiffs peripheral
+    periph_spiffs_cfg_t spiffs_cfg = {
+        .root = "/spiffs",
+        .partition_label = NULL,
+        .max_files = 5,
+        .format_if_mount_failed = true
+    };
+    esp_periph_handle_t spiffs_handle = periph_spiffs_init(&spiffs_cfg);
+
+    // Start spiffs
+    esp_periph_start(set, spiffs_handle);
+
+
+    // Wait until spiffs is mounted
+    while (!periph_spiffs_is_mounted(spiffs_handle)) {
+        vTaskDelay(500 / portTICK_PERIOD_MS);
+    }
+    return ESP_OK;
+}
+
 void app_main(void)
 {
     esp_log_level_set("*", ESP_LOG_INFO);
@@ -669,6 +692,8 @@ void app_main(void)
     esp_periph_config_t periph_cfg = DEFAULT_ESP_PERIPH_SET_CONFIG();
     set = esp_periph_set_init(&periph_cfg);
 
+    mount_spiffs(set);
+
     audio_board_sdcard_init(set, SD_MODE_1_LINE);
     cli_setup_wifi();
     cli_setup_player();
diff --git a/examples/cli/partitions_console_example.csv b/examples/cli/partitions_console_example.csv
index 07eb06e8..8e43e5f2 100644
--- a/examples/cli/partitions_console_example.csv
+++ b/examples/cli/partitions_console_example.csv
@@ -3,3 +3,4 @@
 nvs,      data, nvs,     0x9000,  0x6000,
 phy_init, data, phy,     0xf000,  0x1000,
 factory,  app,  factory, 0x10000, 2M,
+storage,  data, spiffs,  0x210000,1M, 

```

shootao, 2023-06-05:
Hi @IamFive 
Can this modification meet your needs? If yes, can you close this `Issue`

IamFive, 2023-06-05:
sure, thanks.

closed 2023-06-05 by IamFive