espressif/esp-adf #1016: SOLVED: An ESP32-S3 1½cycle/tap fp32 FIR filter of arbitrary length and decimation

f4lc0n-asm, 2023-06-02:
Hello,

since ESP-DSP was not able to include this solution for 4 months, I am hereby offering it directly to you. Two versions are available:
• v3.x is less stringent on allocation and size of FIR coefficients a delay line, but it is slightly slower and its code bulkier than v4.x
• v4.x is just the opposite of v3.x :)

I am attaching the full ASM sources with validation routines in C (study them to see how to setup and use the FIR filters) and appropriate license. Extract them with 7-Zip (the actual format is 7-Zip with several times better compression than generic ZIP). Feel free to add them to your project!

Any question or tweaking request? Ask here! I am very busy so I did it this way…

Cheers!

f4lc0n

[FIR_fp32_1.5c_per_tap_v3.3.zip](https://github.com/espressif/esp-adf/files/11638390/FIR_fp32_1.5c_per_tap_v3.3.zip)
[FIR_fp32_1.5c_per_tap_v4.4.zip](https://github.com/espressif/esp-adf/files/11638391/FIR_fp32_1.5c_per_tap_v4.4.zip)

jason-mao, 2023-06-07:
@f4lc0n-asm Thank you for your contribution. We will merge to esp-dsp.

jason-mao, 2025-02-18:
Closing this due to a new similar issue (#1251).

closed 2025-02-18 by jason-mao