espressif/esp-adf #1017: ESP32-S3-Korvo-2 wwe example English Speech Commands Model has been unregistered

kostt, 2023-06-03:
Default example wwe Chinese Speech Commands Model
works well, but is it possible to change to English Speech Commands Model?

[image]

I (0) cpu_start: App cpu up.
I (701) spiram: SPI SRAM memory test OK
I (710) cpu_start: Pro cpu start user code
I (710) cpu_start: cpu freq: 240000000
I (710) cpu_start: Application information:
I (713) cpu_start: Project name:     example_wwe
I (718) cpu_start: App version:      v2.5-28-gf060491-dirty
I (724) cpu_start: Compile time:     Jun  3 2023 19:56:44
I (730) cpu_start: ELF file SHA256:  bde4cf59d1ed6c34...
I (736) cpu_start: ESP-IDF:          v4.4.4-dirty
I (742) heap_init: Initializing. RAM available for dynamic allocation:
I (749) heap_init: At 3FCA0A18 len 00048CF8 (291 KiB): D/IRAM
I (755) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (762) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (768) spiram: Adding pool of 8192K of external SPI memory to heap allocator
I (777) spi_flash: detected chip: gd
I (781) spi_flash: flash io: qio
I (785) sleep: Configure to isolate all GPIO pins in sleep state
I (791) sleep: Enable automatic switching of GPIO sleep configuration
I (798) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (819) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (829) DRV8311: ES8311 in Slave mode
I (839) gpio: GPIO[48]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (849) ES7210: ES7210 in Slave mode
I (859) ES7210: Enable ES7210_INPUT_MIC1
I (859) ES7210: Enable ES7210_INPUT_MIC2
I (859) ES7210: Enable ES7210_INPUT_MIC3
W (869) ES7210: Enable TDM mode. ES7210_SDP_INTERFACE2_REG12: 2
I (869) ES7210: config fmt 60
W (879) AUDIO_BOARD: The board has already been initialized!

----------------------------- ESP Audio Platform -----------------------------
|                                                                            |
|                       ESP_AUDIO-v1.7.2-20e6bd0-b92a149                     |
|                     Compile date: Nov 30 2022-07:50:12                     |
------------------------------------------------------------------------------
I (919) ESP32_S3_KORVO_2: I2S0, MCLK output by GPIO0
I (919) wwe_example: Func:setup_player, Line:142, MEM Total:8637979 Bytes, Inter:271691 Bytes, Dram:271691 Bytes

I (929) wwe_example: esp_audio instance is:0x3d800b10

E (939) I2S: register I2S object to platform failed
I (939) ESP32_S3_KORVO_2: I2S0, MCLK output by GPIO0
I (949) wwe_example: Recorder has been created
I (949) MODEL_LOADER: 
Initializing models from SPIFFS, partition label: model

I (1209) MODEL_LOADER: Partition size: total: 3900791, used: 2490422

I (1419) AFE_SR: afe interface for speech recognition

I (1419) AFE_SR: AFE version: SR_V220727

I (1419) AFE_SR: Initial auido front-end, total channel: 3, mic num: 2, ref num: 1

I (1429) AFE_SR: aec_init: 1, se_init: 1, vad_init: 1

I (1429) AFE_SR: wakenet_init: 1

model_name: wn9_hiesp model_data: /srmodel/wn9_hiesp/wn9_data
MC Quantized wakenet9: wakeNet9_v1h24_hiesp_3_0.63_0.635, tigger:v3, mode:2, p:0, (Nov  3 2022 11:49:16)
I (2179) AFE_SR: wake num: 3, mode: 0, (Nov  3 2022 11:49:26)

E (2179) RECORDER_SR: recorder_sr.c:548 (recorder_sr_create): Got NULL Pointer
I (2189) AFE_SR: afe_se_task quit

I (2189) AFE_SR: afe_mase task destroy finished

I (2199) AFE_SR: exit function: afe_destory

I (2199) MODEL_LOADER: model has been unregistered

BootsSiR, 2024-01-22:
did you manage to get this working?

robert-alfaro, 2024-01-29:
I just stumbled on this after having the same failures. Enabling english detection on adf v2.6 & master against idf v5.1 not working.

FionnOC, 2024-03-05:
E (1252) RECORDER_SR: recorder_sr.c:602 (recorder_sr_create): Got NULL Pointer
I (1262) AFE_SR: afe_se_task quit

I (1262) AFE_SR: afe_mase task destroy finished

I (1272) AFE_SR: exit function: afe_destory

I (1272) main_task: Returned from app_main()

Has anyone also received this error? We are still getting this issue as well. If anyone has any guidance that would be appreciated :)