espressif/esp-adf #1020: TLS for voip (sip protocol)

Gaai, 2023-06-14:
Looking through the protocol code and libraries I see TLS mentioned here and there. Is TLS already supported for the sip protocol? Encrypting voip data makes sense in almost any practical context right?
Would be great if you could offer some documentation on how to get it working with the project examples for sip and rtc.

ahhfzhang, 2023-06-30:
Hi @Gaai 

1. About tls config, you can refer to IDF http client example to configure it. ref : https://github.com/espressif/esp-idf/blob/master/examples/protocols/esp_http_client/main/esp_http_client_example.c 

2. Tls uri base on [this ](https://github.com/espressif/esp-adf-libs/blob/b0e908b0252d94025fb82274e5d23910bedf0d7c/esp_media_protocols/include/esp_rtc.h#L105).

Best Regards.

Gaai, 2023-06-30:
Will check it out soon as I get back. Thanks!

closed 2023-09-05 by jason-mao