espressif/esp-adf #1022: Compilation fails with ESP-IDF 5.1 and 4.4

parabuzzle, 2023-06-16:
I followed the guides for adding the ADF to my project. I'm currently running IDF 5.1 but I've tried this with 4.4 as well. 

I ran `git clone --recursive https://github.com/espressif/esp-adf.git`  into my ~/esp/ directory. 

I then added the variable to my `.vscode/settings.json` file:

```json
{
  "C_Cpp.intelliSenseEngine": "Tag Parser",
  "idf.adapterTargetName": "esp32",
  "idf.customExtraPaths": "c:\\Users\\parab\\.espressif\\tools\\xtensa-esp-elf-gdb\\12.1_20221002\\xtensa-esp-elf-gdb\\bin;c:\\Users\\parab\\.espressif\\tools\\riscv32-esp-elf-gdb\\12.1_20221002\\riscv32-esp-elf-gdb\\bin;c:\\Users\\parab\\.espressif\\tools\\xtensa-esp32-elf\\esp-12.2.0_20230208\\xtensa-esp32-elf\\bin;c:\\Users\\parab\\.espressif\\tools\\xtensa-esp32s2-elf\\esp-12.2.0_20230208\\xtensa-esp32s2-elf\\bin;c:\\Users\\parab\\.espressif\\tools\\xtensa-esp32s3-elf\\esp-12.2.0_20230208\\xtensa-esp32s3-elf\\bin;c:\\Users\\parab\\.espressif\\tools\\riscv32-esp-elf\\esp-12.2.0_20230208\\riscv32-esp-elf\\bin;c:\\Users\\parab\\.espressif\\tools\\esp32ulp-elf\\2.35_20220830\\esp32ulp-elf\\bin;c:\\Users\\parab\\.espressif\\tools\\cmake\\3.24.0\\bin;c:\\Users\\parab\\.espressif\\tools\\openocd-esp32\\v0.12.0-esp32-20230419\\openocd-esp32\\bin;c:\\Users\\parab\\.espressif\\tools\\ninja\\1.10.2;c:\\Users\\parab\\.espressif\\tools\\idf-exe\\1.0.3;c:\\Users\\parab\\.espressif\\tools\\ccache\\4.8\\ccache-4.8-windows-x86_64;c:\\Users\\parab\\.espressif\\tools\\dfu-util\\0.11\\dfu-util-0.11-win64;c:\\Users\\parab\\.espressif\\tools\\esp-rom-elfs\\20230320",
  "idf.customExtraVars": {
    "OPENOCD_SCRIPTS": "c:\\Users\\parab\\.espressif\\tools\\openocd-esp32\\v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts",
    "IDF_CCACHE_ENABLE": "1",
    "ESP_ROM_ELF_DIR": "c:\\Users\\parab\\.espressif\\tools\\esp-rom-elfs\\20230320/"
  },
  "idf.espIdfPathWin": "C:\\Users\\parab\\esp\\esp-idf",
  "idf.espAdfPathWin": "C:\\Users\\parab\\esp\\esp-adf",
  "idf.openOcdConfigs": [
    "board/esp32-wrover-kit-3.3v.cfg"
  ],
  "idf.portWin": "COM9",
  "idf.pythonBinPathWin": "c:\\Users\\parab\\.espressif\\python_env\\idf5.1_py3.8_env\\Scripts\\python.exe",
  "idf.toolsPathWin": "c:\\Users\\parab\\.espressif",
  "idf.flashType": "UART",
  "files.associations": {
    "*.tcc": "cpp",
    "compare": "cpp",
    "istream": "c",
    "i2s_std.h": "c",
    "cstddef": "c",
    "nonvol.h": "c"
  },
  "idf.gitPathWin": "c:\\Users\\parab\\.espressif\\tools\\idf-git\\2.30.1\\cmd\\git.exe"
}
```

Lastly I added to my `CMakeLists.txt` file:

```
# For more information about build system see
# https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/build-system.html
# The following five lines of boilerplate have to be in your project's
# CMakeLists in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.5)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
include($ENV{ADF_PATH}/CMakeLists.txt)

project(hello_world)
```

I did a clean and build and I get the following error:

```
[0/1] Re-running CMake...
Warning: in CMake variable EXTRA_COMPONENT_DIRS: Path component ' C:\Users\parab\esp\esp-adf/components' contains
  leading spaces

Note: In ESP-IDF v5.0 and later, COMPONENT_DIRS and EXTRA_COMPONENT_DIRS should be defined
      as CMake lists, not as space separated strings.

Examples:
    * set(EXTRA_COMPONENT_DIRS path/to/components path/to/more/components)
      # Correct, EXTRA_COMPONENT_DIRS is defined as a CMake list, with two paths added

    * list(APPEND EXTRA_COMPONENT_DIRS path/to/component)
      list(APPEND EXTRA_COMPONENT_DIRS path/to/more/components)
      # Correct, use when building EXTRA_COMPONENT_DIRS incrementally

[... 454 lines cut ...]
      |                  ~~~~~~^~~~~
C:/Users/parab/esp/esp-adf/components/esp_event_cast/esp_event_cast.c: In function 'esp_event_cast_get_count':
C:/Users/parab/esp/esp-adf/components/esp_event_cast/esp_event_cast.c:162:22: warning: passing argument 1 of 'mutex_lock' makes pointer from integer without a cast [-Wint-conversion]
  162 |     mutex_lock(handle->_mux);
      |                ~~~~~~^~~~~~
      |                      |
      |                      int
C:/Users/parab/esp/esp-adf/components/audio_sal/include/audio_mutex.h:60:22: note: expected 'void *' but argument is of type 'int'
   60 | int mutex_lock(void *mutex);
      |                ~~~~~~^~~~~
C:/Users/parab/esp/esp-adf/components/esp_event_cast/esp_event_cast.c:168:24: warning: passing argument 1 of 'mutex_unlock' makes pointer from integer without a cast [-Wint-conversion]
  168 |     mutex_unlock(handle->_mux);
      |                  ~~~~~~^~~~~~
      |                        |
      |                        int
C:/Users/parab/esp/esp-adf/components/audio_sal/include/audio_mutex.h:69:24: note: expected 'void *' but argument is of type 'int'
   69 | int mutex_unlock(void *mutex);
      |                  ~~~~~~^~~~~
cc1.exe: some warnings being treated as errors
[793/1100] Building C object esp-idf/protocomm/CMakeFiles/__idf_protocomm.dir/src/security/security1.c.obj
ninja: build stopped: subcommand failed.

 *  The terminal process "C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command ninja " terminated with exit code: 1. 

```

parabuzzle, 2023-06-16:
I'm building for an ESP32-WROVER-E and here's my sdconfig:

```
#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_SOC_BROWNOUT_RESET_SUPPORTED="Not determined"
CONFIG_SOC_TWAI_BRP_DIV_SUPPORTED="Not determined"
CONFIG_SOC_DPORT_WORKAROUND="Not determined"
CONFIG_SOC_CAPS_ECO_VER_MAX=301
CONFIG_SOC_ADC_SUPPORTED=y
CONFIG_SOC_DAC_SUPPORTED=y
CONFIG_SOC_UART_SUPPORTED=y
CONFIG_SOC_MCPWM_SUPPORTED=y
CONFIG_SOC_GPTIMER_SUPPORTED=y
CONFIG_SOC_SDMMC_HOST_SUPPORTED=y
CONFIG_SOC_BT_SUPPORTED=y
CONFIG_SOC_PCNT_SUPPORTED=y
CONFIG_SOC_WIFI_SUPPORTED=y
CONFIG_SOC_SDIO_SLAVE_SUPPORTED=y
CONFIG_SOC_TWAI_SUPPORTED=y
CONFIG_SOC_EMAC_SUPPORTED=y
CONFIG_SOC_ULP_SUPPORTED=y
CONFIG_SOC_CCOMP_TIMER_SUPPORTED=y
CONFIG_SOC_RTC_FAST_MEM_SUPPORTED=y
CONFIG_SOC_RTC_SLOW_MEM_SUPPORTED=y
CONFIG_SOC_RTC_MEM_SUPPORTED=y
CONFIG_SOC_I2S_SUPPORTED=y
CONFIG_SOC_RMT_SUPPORTED=y
CONFIG_SOC_SDM_SUPPORTED=y
CONFIG_SOC_GPSPI_SUPPORTED=y
CONFIG_SOC_LEDC_SUPPORTED=y
CONFIG_SOC_I2C_SUPPORTED=y
CONFIG_SOC_SUPPORT_COEXISTENCE=y
CONFIG_SOC_AES_SUPPORTED=y
CONFIG_SOC_MPI_SUPPORTED=y
CONFIG_SOC_SHA_SUPPORTED=y
CONFIG_SOC_FLASH_ENC_SUPPORTED=y
CONFIG_SOC_SECURE_BOOT_SUPPORTED=y
CONFIG_SOC_TOUCH_SENSOR_SUPPORTED=y
CONFIG_SOC_BOD_SUPPORTED=y
CONFIG_SOC_ULP_FSM_SUPPORTED=y
CONFIG_SOC_DPORT_WORKAROUND_DIS_INTERRUPT_LVL=5
CONFIG_SOC_XTAL_SUPPORT_26M=y
CONFIG_SOC_XTAL_SUPPORT_40M=y
CONFIG_SOC_XTAL_SUPPORT_AUTO_DETECT=y
CONFIG_SOC_ADC_RTC_CTRL_SUPPORTED=y
CONFIG_SOC_ADC_DIG_CTRL_SUPPORTED=y
CONFIG_SOC_ADC_DMA_SUPPORTED=y
CONFIG_SOC_ADC_PERIPH_NUM=2
CONFIG_SOC_ADC_MAX_CHANNEL_NUM=10
CONFIG_SOC_ADC_ATTEN_NUM=4
CONFIG_SOC_ADC_DIGI_CONTROLLER_NUM=2
CONFIG_SOC_ADC_PATT_LEN_MAX=16
CONFIG_SOC_ADC_DIGI_MIN_BITWIDTH=9
CONFIG_SOC_ADC_DIGI_MAX_BITWIDTH=12
CONFIG_SOC_ADC_DIGI_RESULT_BYTES=2
CONFIG_SOC_ADC_DIGI_DATA_BYTES_PER_CONV=4
CONFIG_SOC_ADC_SAMPLE_FREQ_THRES_HIGH=2
CONFIG_SOC_ADC_SAMPLE_FREQ_THRES_LOW=20
CONFIG_SOC_ADC_RTC_MIN_BITWIDTH=9
CONFIG_SOC_ADC_RTC_MAX_BITWIDTH=12
CONFIG_SOC_SHARED_IDCACHE_SUPPORTED=y
CONFIG_SOC_CPU_CORES_NUM=2
CONFIG_SOC_CPU_INTR_NUM=32
CONFIG_SOC_CPU_HAS_FPU=y
CONFIG_SOC_CPU_BREAKPOINTS_NUM=2
CONFIG_SOC_CPU_WATCHPOINTS_NUM=2
CONFIG_SOC_CPU_WATCHPOINT_SIZE=64
[... 1885 lines cut ...]
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_TIME_SYSCALL_USE_RTC_HRT=y
CONFIG_ESP32_TIME_SYSCALL_USE_RTC_FRC1=y
# CONFIG_ESP32_TIME_SYSCALL_USE_RTC is not set
# CONFIG_ESP32_TIME_SYSCALL_USE_HRT is not set
# CONFIG_ESP32_TIME_SYSCALL_USE_FRC1 is not set
# CONFIG_ESP32_TIME_SYSCALL_USE_NONE is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
# CONFIG_ESP32_ULP_COPROC_ENABLED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
# End of deprecated options
```

TempoTian, 2023-06-19:
You can enable this config `CONFIG_FREERTOS_ENABLE_BACKWARD_COMPATIBILITY=y`.

closed 2023-09-05 by jason-mao