espressif/esp-adf #1027: DLNA Example XML Parsing Error

marcolinux46, 2023-06-25:
## Environment

- Audio development kit:      [ESP32-LyraT-Mini]
- Audio kit version (ESP32-LyraT-Mini]
- [Required] Module or chip used:  [ESP32-WROVER-E]
- [Required] IDF version v4.4.5-104-g8b94183c9c
- [Required] ADF version v2.5-46-g4b60d81
- Build system:         [CMake]
- [Required] Running log: 
     E (1817932) HTTP_CLIENT: Error parse url uuid:5378d5a8-004e-87a6-c714-4c598d93d725
     E (1817942) HTTP_CLIENT: Error parse url uuid:71dfed43-977b-bbbe-4b90-666d97169589
     E (1817942) HTTP_CLIENT: Error parse url uuid:2b449456-8624-e644-32df-6db505597d8d
- Operating system:     [Windows]
- (Windows only) Environment type: [Powershell]
- Using an IDE?: [Yes, VSCode with ESP extension v.1.6.3]
- Power supply:         [USB]

## Problem Description

Hi,

I have successfully compiled and installed the DLNA sample on my LyraT-mini card.
I try to setup the device on my Home Assistant server but it doesn't work.
In the server logs I see many errors on "UpnpXmlParseError('not well formed (invalid token): row 3, column 100')"

Below a screenshot of my browser when i try to reach the device via web.
The problem occurs with all three main XML files (AVTransport, ConnectionManager, RenderingControl)

Thanks for your support

// If possible, attach a picture of your setup/wiring here.

[image]

jason-mao, 2023-09-05:
@marcolinux46 Would you provide the XML context at here?

marcolinux46, 2023-09-05:
Hi @jason-mao,

the XMLs are the files provided by the example under github dlna/components/esp_dlna/xml

jason-mao, 2023-09-05:
@jason-mao Thanks for your feedback, let we check and fix.

Kaiserdragon2, 2024-10-01:
I have an similar issue with the ESP-IDF v5.3.1 where the xml files are served with an additional NUL character at the end.
This lets some DLNA Servers run into an XMLParserError. I have tested this with jellyfin.

Files that are served with the additional NUL character:

- AVTransport.xml
- ConnectionManager.xml
- RenderingControl.xml

The files in the project are correct, but when building the NUL character gets appended. And when flashing the project and running it the file are served with the additional NUL character.

When browsing to such a file it looks like this: 
[image]

The error in Jellyfin reads:
System.Xml.XmlException: '.', hexadecimal value 0x00, is an invalid character. Line 515, position 8.