espressif/esp-adf #1031: Allow to disable audio forge in the ESP AUDIO library

masterxq, 2023-06-30:
**Is your feature request related to a problem? Please describe.**

Hello *,
I would love to be able to deactivate audio forge in the esp audio library completely.
In my case there is no post-processing needed. As the Hardware can handle all needed functionality.
For be honest I do not completely know what audio forge else is doing for the esp audio library. But from user view, my pipeline does not need this.
If a special format is needed that not is reflected by the decoded audio, I choose to implement it to the output module where the samples are touched in every case and I can ensure the quality lose of the samples is not higher than needed by the conversion. What of cause depends on the output module, but in many cases the target device supports multiply formats.

I want to avoid that a by use case useless component eats Memory and CPU or even lover the quality of the samples.

**Describe the solution you'd like**
```
#define ESP_AUDIO_COMPONENT_SELECT_DEFAULT    0x00      /*!< Default selected */
#define ESP_AUDIO_COMPONENT_SELECT_ALC        0x04      /*!< ALC selected */
#define ESP_AUDIO_COMPONENT_SELECT_EQUALIZER  0x08      /*!< Equalizer selected */
#define ESP_AUDIO_COMPONENT_SELECT_NONE  0x10   /*!< Disabled - No post-processing */
```
Or even better if this not breaks compatibility (Let default selection overwrite other flags):
```
#define ESP_AUDIO_COMPONENT_SELECT_NONE    0x00      /*!< Disabled - No post-processing */
#define ESP_AUDIO_COMPONENT_SELECT_ALC        0x04      /*!< ALC selected */
#define ESP_AUDIO_COMPONENT_SELECT_EQUALIZER  0x08      /*!< Equalizer selected */
#define ESP_AUDIO_COMPONENT_SELECT_DEFAULT  0x80   /*!< Default selected */
```
**Describe alternatives you've considered**

Maybe there is a way to disable audio forge, or maybe this ticket is useless because I have no knowledge about the esp audio library.

Depending on what is true I see this alternative ways:
- Update the docs and add more text about what the ADF components does and how to manipulate them
- Release the source code and let user modify it
- Release the source code or part of it and let user read it to understand the behavior, so it is possible to write better tickets

Of course, I can use a custom pipeline too, and I already did, and it also works, but it is really hard to write all the features that esp audio library already did implement, and it feels wrong because I only have problems with details of the esp audio library.

In every case, thanks for your work on this, even it still has some small issues it's a really impressive library.

TempoTian, 2023-07-04:
It is hard to decouple `audio_forge` from `esp-audio`.
If memory is not cared, you can bypass audio forge so that it does not affect the output quality and consume CPU.
Audio forge element name is "Audio_forge"
you can get its element handle by `audio_pipeline_get_el_by_tag` and call `audio_forge_sonic_set_component_select` after `esp_audio_play`.

masterxq, 2023-07-05:
Many thanks!

> It is hard to decouple `audio_forge` from `esp-audio`.
I see, this is what I expected.

I gave up for now and completed my implementation component similar to esp_audio. For now, it works fine, but I already thought before, and then there were some edge cases that needed to be fixed ^^
So lets see how long it comes.

> you can get its element handle by `audio_pipeline_get_el_by_tag` and call `audio_forge_sonic_set_component_select` after `esp_audio_play`.

If I want to go back to esp_audio, can I just call this to disable it:
```C
audio_forge_sonic_set_component_select(el_audio_forge, 0);
```
?

Thanks again and have a nice day :)

jason-mao, 2023-09-05:
@masterxq  audio_forge_sonic_set_component_select(el_audio_forge, 0); can be diable the process, the input data bypass to output.

closed 2023-09-05 by jason-mao