espressif/esp-adf #1034: VoIP例程，APP端网络波动时，设备端解析收到的数据包长度错误

shangpeng01, 2023-07-04:
SDK版本: esp-adf-v2.5+esp-idf-v4.4.4
硬件: ESP32-S3-Korvo-2L
SIP服务器：FreeSwitch
手机APP: 安卓手机Linphone
问题描述：
语音通话过程中，手机网络波动，比如WiFi切换4G或者4G关闭重连等，
手机端网络恢复后，设备端收到的RTP数据包解析出错，会导致崩溃重启，添加打印发现解析出来的音频数据长度是1342177285。
[image]

网络抓包发现在上述场景网络恢复后，设备收到的RTP包的长度从之前的214变成了374，设备端解析这种长度为374的数据包会出错
附件为网络抓包文件
[sip_test.zip](https://github.com/espressif/esp-adf/files/11948847/sip_test.zip)

这种长度为374的RTP包用WireShark工具可以解析播放
[image]
[image]

ahhfzhang, 2023-07-07:
Hi @shangpeng01 ,

我使用了Android和Iphone都测试了一下，这里切回网络的时候，还是按照之前的包发送的并没有去改变RTP包的大小；

理论上按照协议规则，在协商完成之后，后面发送的RTP包都应该是同一个规则和同样的ptime的，如果后面修改了必须要进行重新协商，所以这个看起来是不合理的，后面发送的包改变了ptime。

另外我使用的是ADF master分支，这一部分我无法复现，所以也不好确定为什么这里的size产生了异常行为。

br.

shangpeng01, 2023-07-07:
用ADF master分支也复现了，设备端表现为不会处理这种长度为374的RTP包
我这边用的Linphone版本为5.0.6， LinphoneSDK版本为5.2.23，用LinphoneSDK5.2.62也能复现

比较容易复现的方式为手机用4G联网，语音通话过程中断开4G网络约10秒钟左右，然后恢复4G网络连接。
刚恢复4G网络连接时，设备端收到的RTP包长度是214，可以正常解析播放，约3秒后，设备端收到的RTP包长度变成374，设备端开始不处理。
请问下您用的Linphone的版本是多少，这个异常是不是可能和Linphone的版本有关系？

附日志截图
[image]
[image]
[image]

和抓包文件
[sip_test_2.zip](https://github.com/espressif/esp-adf/files/11982316/sip_test_2.zip)

shangpeng01, 2023-07-10:
手机端，Linphone音频设置里“自适应速率控制”选项关掉，网络波动时APP端ptime会固定不变，可以规避掉这个问题。

closed 2023-09-04 by jason-mao