espressif/esp-adf #1041: pipeline_loop_playback_without_gap example is not fully continuous

alanmosely, 2023-07-21:
- Module or chip used: ESP32-S3-WROOM-1
- IDF version: v4.4.4
- ADF version: v2.5-1-g6046621 
- Build system: idf.py
- Compiler version: 8.4.0
- Operating system: Windows
- Environment type: PowerShell
- Using an IDE?: VS Code
- Power supply: USB

## Problem Description

I am running https://github.com/espressif/esp-adf/tree/master/examples/player/pipeline_loop_playback_without_gap on my custom board, playing ogg and mp3 files from an external flash. There are no errors and the example works as expected except for a very small gap when updating the ring buffer and running the pipeline. Would it be possible to have no gap at all?

### Expected Behavior

On an audio file that has been made to loop seamlessly, support completely seamless looping with no noticeable pause in audio at all

### Actual Behavior

A very small pause occurs

### Steps to Reproduce

Use example with a continuous, looping sound, for example white noise

## Debug Logs

`I (2758) audio_player:        - Create fatfs_stream_reader[0] to read data from sdcard
I (2778) audio_player:        - Create music_decoder[0]
I (2778) AUDIO_PIPELINE: link el->rb, el:0x3de182fc, tag:file, rb:0x3de185d4
I (2778) audio_player:        - Create fatfs_stream_reader[1] to read data from sdcard
I (2798) audio_player:        - Create music_decoder[1]
I (2798) AUDIO_PIPELINE: link el->rb, el:0x3de1a640, tag:file, rb:0x3de1a918
I (2818) audio_player:        - Create i2s_stream_writer to write data to codec chip
W (2818) I2S: APLL not supported on current chip, use I2S_CLK_D2CLK as default clock source
I (2838) I2S: DMA Malloc info, datalen=blocksize=1200, dma_buf_count=3
I (2838) I2S: DMA Malloc info, datalen=blocksize=1200, dma_buf_count=3
I (2838) audio_player: [ 4 ] Set up event listener
I (2838) audio_player:        - Listening event from all elements of pipeline
I (2868) audio_player: [ 5 ] Create the ringbuf and connect decoder to i2s
I (2868) audio_player: [ 6 ] Start pipeline_input and i2s_stream_writer
I (2888) audio_player: volume set to: 43
I (2898) AUDIO_ELEMENT: [file-0x3de182fc] Element task created
I (2898) AUDIO_THREAD: The dec task allocate stack on external memory
I (2908) AUDIO_ELEMENT: [dec-0x3de18460] Element task created
I (2908) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:359, MEM Total:2063455 Bytes, Inter:104615 Bytes, Dram:104615 Bytes
I (2928) AUDIO_ELEMENT: [file] AEL_MSG_CMD_RESUME,state:1
I (2928) FATFS_STREAM: File size: 296371 byte, file position: 0
I (2938) AUDIO_ELEMENT: [dec] AEL_MSG_CMD_RESUME,state:1
I (2948) VORBIS_DECODER: VORBIS codec
I (2948) CODEC_ELEMENT_HELPER: The element is 0x3de18460. The reserve data 2 is 0x0.
I (2958) VORBIS_DECODER: a new song playing
I (2958) VORBIS_DECODER: a new songI (3038) AUDIO_ELEMENT: [iis-0x3de1ca48] Element task created
I (3038) AUDIO_ELEMENT: [iis] AEL_MSG_CMD_RESUME,state:1
I (3048) I2S_STREAM: AUDIO_STREAM_WRITER
I (3068) audio_player: [ 7 ] Listen for all pipeline events
I (3068) audio_player: [ * ] Receive music info from music_decoder[0], sample_rates=44100, bits=16, ch=1
I (3118) AUDIO_ELEMENT: [iis] AEL_MSG_CMD_PAUSE
W (3118) I2S: APLL not supported on current chip, use I2S_CLK_D2CLK as default clock source
W (3118) AUDIO_ELEMENT: [iis-0x3de1ca48] RESUME timeout
W (3118) AUDIO_ELEMENT: [iis-0x3de1ca48] RESUME timeout
I (3138) I2S_STREAM: AUDIO_STREAM_WRITER
W (12558) FATFS_STREAM: No more data, ret:0
I (12558) AUDIO_ELEMENT: IN-[file] AEL_IO_DONE,0
I (12718) AUDIO_ELEMENT: IN-[dec] AEL_IO_DONE,-2
I (12938) VORBIS_DECODER: Closed
I (12938) audio_player: [ * ] The Pipeline[0] has been finished. Switch to another one
I (12938) AUDIO_ELEMENT: [file-0x3de1a640] Element task created
I (12948) AUDIO_THREAD: The dec task allocate stack on external memory
I (12948) AUDIO_ELEMENT: [dec-0x3de1a7a4] Element task created
I (12948) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:359, MEM Total:2039427 Bytes, Inter:98503 Bytes, Dram:98503 Bytes
I (12968) AUDIO_ELEMENT: [file] AEL_MSG_CMD_RESUME,state:1
I (12978) FATFS_STREAM: File size: 296371 byte, file position: 0
I (12988) AUDIO_ELEMENT: [dec] AEL_MSG_CMD_RESUME,state:1
I) AUDIO) VORBIS_DECODER: VORBIS codec
I) AUDIO) VORBIS_DECODER: VORBIS codec
I (12988) CODEC_ELEMENT_HELPER: The element is 0x3de1a7a4. The reserve data 2 is 0x0.
W (12998) AUDIO_ELEMENT: [file] Element already stopped
I (12998) VORBIS_DECODER: a new song playing
W (13008) AUDIO_ELEMENT: [dec] Element already stopped`

jason-mao, 2023-08-04:
@alanmosely How long is the small gap?  Can you provide an audio chart or a spectrum chart?

alanmosely, 2023-08-07:
Sorry for wasting your time, it was an issue on my end.

closed 2023-08-07 by alanmosely

vcjackchen, 2025-09-04:
Hello, I encountered the same issue when using the pipeline_loop_playback_without_gap routine in this way. There are gaps. Have you solved this problem yet? I still have no idea how to solve this problem. Could you please tell me how it was done?