espressif/esp-adf #1042: S3 jpeg编码速度如何

weilian1977, 2023-07-24:
VGA 分辨率   灰度图像

jason-mao, 2023-08-04:
@weilian1977 
VGA 灰度图编码大概 18 FPS

weilian1977, 2023-08-05:
基于什么条件测试的？我编译https://github.com/espressif/esp-adf/tree/master/examples/display/lcd_camera 好像是白屏无显示

jason-mao, 2023-08-07:
测试数据来自于文件读取独立VGA 灰度图进行编码。[lcd_camera](https://github.com/espressif/esp-adf/tree/master/examples/display/lcd_camera) 中ensor 输出默认是yuv的格式，不能直接使用。

weilian1977, 2023-08-08:
> 测试数据来自于文件读取独立VGA 灰度图进行编码。[lcd_camera](https://github.com/espressif/esp-adf/tree/master/examples/display/lcd_camera) 中ensor 输出默认是RGB的格式，不能直接使用，需要把Camera 改为 YUV422，JPEG 的Subsampling 配置为JPEG_SUB_SAMPLE_Y。

OK！非常感谢！

weilian1977, 2023-12-13:
> 测试数据来自于文件读取独立VGA 灰度图进行编码。[lcd_camera](https://github.com/espressif/esp-adf/tree/master/examples/display/lcd_camera) 中sensor 输出默认是RGB的格式，不能直接使用，需要把Camera 改为 YUV422，JPEG 的Subsampling 配置为JPEG_SUB_SAMPLE_Y。

> @weilian1977 VGA 灰度图编码大概 18 FPS

使用https://github.com/espressif/esp-adf/examples/display/lcd_jpeg/main/image.jpg文件（调整分辨率为640*480），使用/esp-adf/examples/display/lcd_jpeg/程序，先解码为RGB565格式，然后计算转为gray格式，再调用esp-adf-libs/esp_codec/include/codec中的esp_jpeg_enc进行压缩，输入格式JPEG_RAW_TYPE_GRAY，Subsampling 配置为JPEG_SUB_SAMPLE_Y，CPU频率240M，SRAM频率80M，计算压缩前后的时间差，结果如下：
质量设置为40，耗时120ms；质量设置为40，耗时140ms；质量设置为40，耗时160ms；
这个速度和18帧的速度似乎差距较大，请问是哪里的原因呢？