espressif/esp-adf #1043: How to use adf with Ethernet ?

ArvinHou, 2023-07-27:
I'm using IDF V4.4.5 and ADF V2.5.select lyrat_v4.3 board.
When I debug the network alone，it's OK.
When I debug the es8388 alone，it's OK too .
When I debug together，something wrong.
the es8388 works OK,but the ethernet will link down.
I think it's a problem with the pins,I have changed a lot of pins。

Pins used of my board:

LAN
LAN_RXD0  --->  IO25
LAN_RXD1  --->  IO26
LAN_CRS_DV  --->  IO27
LAN_MDC  --->  IO23
LAN_MDIO  --->  IO18
LAN_TXD1  --->  IO22
LAN_TXD0  --->  IO19
LAN_TX_EN  --->  IO21
LAN_RESET  --->  IO5
LAN_REF_CLK  --->  IO0

es8388
ASDOUT  --->  IO35
DSDIN  --->  IO33
LRCK  --->  IO14
SCLK  --->  IO2
MCLK  --->  IO3
SDA  --->  IO15
SCL  --->  IO13

There are some buttons, PA, and button in the ADF library,I also changed.

///////////////////////////////////////////////////////
the board_pins_config.h of lyrat_v4.3 :

i2c_config->sda_io_num = GPIO_NUM_15;
i2c_config->scl_io_num = GPIO_NUM_13;

i2s_config->bck_io_num = GPIO_NUM_2;	//SCLK
i2s_config->ws_io_num = GPIO_NUM_14;	//LRCK
i2s_config->data_out_num = GPIO_NUM_33;	//DSDIN
i2s_config->data_in_num = GPIO_NUM_35;	//DSDOUT

///////////////////////////////////////////////////////

the i2s_stream.c :

i2s_mclk_gpio_select(i2s->config.i2s_port, GPIO_NUM_3);

///////////////////////////////////////////////////////
the board_def.h of lyrat_v4.3 :

#define FUNC_SDCARD_EN            (0)
#define SDCARD_OPEN_FILE_NUM_MAX  5
#define SDCARD_INTR_GPIO          GPIO_NUM_4//GPIO_NUM_34

#define FUNC_SYS_LEN_EN           (0)
#define GREEN_LED_GPIO            GPIO_NUM_4//GPIO_NUM_22

#define FUNC_AUDIO_CODEC_EN       (1)
#define AUXIN_DETECT_GPIO         GPIO_NUM_4//GPIO_NUM_12
#define HEADPHONE_DETECT          GPIO_NUM_4//GPIO_NUM_19
#define PA_ENABLE_GPIO            GPIO_NUM_4//GPIO_NUM_21

#define FUNC_BUTTON_EN            (0)
#define INPUT_KEY_NUM             6
//hou
#define BUTTON_REC_ID             GPIO_NUM_4//GPIO_NUM_36
#define BUTTON_MODE_ID            GPIO_NUM_4//GPIO_NUM_39
#define BUTTON_SET_ID             GPIO_NUM_4//TOUCH_PAD_NUM9
#define BUTTON_PLAY_ID            GPIO_NUM_4//TOUCH_PAD_NUM8
#define BUTTON_VOLUP_ID           GPIO_NUM_4//TOUCH_PAD_NUM7
#define BUTTON_VOLDOWN_ID         GPIO_NUM_4//TOUCH_PAD_NUM4

the log bellow

```
I (0) cpu_start: App cpu up.
I (846) spiram: SPI SRAM memory test OK
I (853) cpu_start: Pro cpu start user code
I (853) cpu_start: cpu freq: 240000000
I (853) cpu_start: Application information:
I (856) cpu_start: Project name:     esp32
I (861) cpu_start: App version:      f3e59b9-dirty
I (866) cpu_start: Compile time:     Jul 27 2023 15:55:47
I (872) cpu_start: ELF file SHA256:  939f8ba93b237586...
I (878) cpu_start: ESP-IDF:          v4.4.5-dirty
I (884) cpu_start: Min chip rev:     v0.0
I (889) cpu_start: Max chip rev:     v3.99 
I (893) cpu_start: Chip rev:         v1.0
I (898) heap_init: Initializing. RAM available for dynamic allocation:
I (905) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (911) heap_init: At 3FFB5448 len 0002ABB8 (170 KiB): DRAM
I (918) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (924) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (930) heap_init: At 40093178 len 0000CE88 (51 KiB): IRAM
I (937) spiram: Adding pool of 4095K of external SPI memory to heap allocator
I (945) spi_flash: detected chip: generic
I (949) spi_flash: flash io: dio
I (954) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (964) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (974) esp32: [APP] Startup..
I (974) esp32: [APP] Free memory: 4445467 bytes
I (984) esp32: [APP] IDF version: v4.4.5-dirty
I (984) esp32: [APP] MAC: 98CDACB11B74
I (1004) system_api: Base MAC address is not set
I (1004) system_api: read default base MAC address from EFUSE
I (1024) esp_eth.netif.netif_glue: 98:cd:ac:b1:1b:77
I (1024) esp_eth.netif.netif_glue: ethernet attached to netif
I (3524) eth: Ethernet Started
I (3524) eth: Ethernet init
I (3524) eth: Ethernet Link Up
I (3524) eth: Ethernet HW Addr 98:cd:ac:b1:1b:77
I (3524) TCP SERVER: Socket created, serv_sockfd=54
I (3524) audio: [ 1 ] Start codec chip
I (3524) UDP SERVER: Socket created
I (3524) SERIAL: start serial rx task
I (3534) TCP SERVER: Socket binded
I (3544) UDP SERVER: Socket bound, port 1234
I (3544) ES8388_DRIVER: init,out:02, in:00
I (3554) UDP SERVER: UDP waiting for data
I (3554) TCP SERVER: Socket listening
I (3564) AUDIO_HAL: Codec mode is 2, Ctrl:1
I (3574) audio: [2.0] Create audio pipeline for playback
I (3574) audio: [2.1] Create tone stream to read data from flash
I (3584) audio: [2.2] Create i2s stream to write data to codec chip
I (3594) I2S: APLL expected frequency is 22579200 Hz, real frequency is 22579193 Hz
I (3604) I2S: DMA Malloc info, datalen=blocksize=1200, dma_buf_count=3
I (3604) I2S: DMA Malloc info, datalen=blocksize=1200, dma_buf_count=3
I (3614) LYRAT_V4_3: I2S0, MCLK output by GPIO3
I (3614) audio: [2.3] Create mp3 decoder to decode mp3 file
I (3624) MP3_DECODER: MP3 init
I (3624) audio: [2.4] Register all elements to audio pipeline
I (3634) audio: [2.5] Link it together [flash]-->tone_stream-->mp3_decoder-->i2s_stream-->[codec_chip]
I (3644) AUDIO_PIPELINE: link el->rb, el:0x3f805d04, tag:tone, rb:0x3f806188
I (3654) AUDIO_PIPELINE: link el->rb, el:0x3f806018, tag:mp3, rb:0x3f8069c8
I (3664) audio: [2.6] Set up  uri (file as tone_stream, mp3 as mp3 decoder, and default output is i2s)
I (3674) audio: [ 3 ] Set up event listener
I (3674) audio: [3.1] Listening event from all elements of pipeline
I (3684) audio: [ 4 ] Start audio_pipeline
I (3684) AUDIO_ELEMENT: [tone-0x3f805d04] Element task created
I (3694) AUDIO_ELEMENT: [mp3-0x3f806018] Element task created
I (3704) AUDIO_ELEMENT: [i2s-0x3f805e9c] Element task created
I (3704) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:359, MEM Total:4335031 Bytes, Inter:250135 Bytes, Dram:198067 Bytes

I (3714) AUDIO_ELEMENT: [tone] AEL_MSG_CMD_RESUME,state:1
I (3724) AUDIO_ELEMENT: [mp3] AEL_MSG_CMD_RESUME,state:1
I (3734) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_RESUME,state:1
I (3734) AUDIO_PIPELINE: Pipeline started
I (3734) I2S_STREAM: AUDIO_STREAM_WRITER
I (3744) audio: [ 4 ] Listen for all pipeline events
I (3734) MP3_DECODER: MP3 opened
I (3744) TONE_PARTITION: tone partition format 0, total 21
I (3764) TONE_STREAM: Tone offset:00005410, Tone length:5400, pos:1

I (3774) audio: [ * ] Receive music info from mp3 decoder, sample_rates=16000, bits=16, ch=1
I (3804) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_PAUSE
I (3804) I2S: APLL expected frequency is 8192000 Hz, real frequency is 8191999 Hz
I (3804) I2S: DMA Malloc info, datalen=blocksize=600, dma_buf_count=3
I (3814) I2S: DMA Malloc info, datalen=blocksize=600, dma_buf_count=3
W (3824) AUDIO_ELEMENT: [i2s-0x3f805e9c] RESUME timeout
I (3824) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_RESUME,state:4
I (3834) I2S_STREAM: AUDIO_STREAM_WRITER
W (3844) TONE_STREAM: No more data,ret:0 ,info.byte_pos:5400
I (3844) AUDIO_ELEMENT: IN-[tone] AEL_IO_DONE,0
I (4474) AUDIO_ELEMENT: IN-[mp3] AEL_IO_DONE,-2
I (4524) esp_netif_handlers: eth ip: 192.168.3.137, mask: 255.255.255.0, gw: 192.168.3.1
I (4524) eth: Ethernet Got IP Address
I (4524) eth: ~~~~~~~~~~~
I (4524) eth: ETHIP:192.168.3.137
I (4534) eth: ETHMASK:255.255.255.0
I (4534) eth: ETHGW:192.168.3.1
I (4534) eth: ~~~~~~~~~~~
I (5244) MP3_DECODER: Closed
I (5524) eth: Ethernet Link Down
I (5564) AUDIO_ELEMENT: IN-[i2s] AEL_IO_DONE,-2
W (5784) audio: [ * ] Stop event received
I (5784) audio: [ 5 ] Stop audio_pipeline
E (5784) AUDIO_ELEMENT: [tone] Element already stopped
E (5794) AUDIO_ELEMENT: [mp3] Element already stopped
E (5794) AUDIO_ELEMENT: [i2s] Element already stopped
W (5804) AUDIO_PIPELINE: There are no listener registered
I (5804) AUDIO_PIPELINE: audio_pipeline_unlinked
W (5814) AUDIO_ELEMENT: [tone] Element has not create when AUDIO_ELEMENT_TERMINATE
W (5824) AUDIO_ELEMENT: [i2s] Element has not create when AUDIO_ELEMENT_TERMINATE
I (5834) I2S: DMA queue destroyed
I (5834) I2S: DMA queue destroyed
W (5834) AUDIO_ELEMENT: [mp3] Element has not create when AUDIO_ELEMENT_TERMINATE
```

I really don't know where is the problem ，please  help  me .thanks

ArvinHou, 2023-07-27:
I find the ethernet link down when the  code run.  Does the clock of i2s conflict with the clock of Ethernet ?

/* Step 7: Set I2S clocks and start. No need to give parameters since configurations has been set in 'i2s_driver_init' */
    ESP_GOTO_ON_ERROR(i2s_set_clk(i2s_num, 0, 0, 0), err, TAG, "I2S set clock failed");

TempoTian, 2023-07-28:
I think the issue is caused by ethernet use GPIO 0
LAN_REF_CLK ---> IO0

For IO0 is used as I2S mclk PIN， if possible move it to another PIN and retry.

ArvinHou, 2023-07-28:
> I think the issue is caused by ethernet use GPIO 0 LAN_REF_CLK ---> IO0
> 
> For IO0 is used as I2S mclk PIN， if possible move it to another PIN and retry.
I2S mclk PIN has chanded 
MCLK ---> IO3

someone reply me:
There's only one APLL in ESP32. I2S can use APLL as a clock source and Ethernet can also use APLL to generate the RMII reference clock.If both of them want to be used together, then APLL must be set to 50MHz on the I2S side.

Set use_apll=false.But it doesn't work.Ethernet link down.

i2s_cfg.i2s_config.use_apll = false;
i2s_cfg.i2s_config.fixed_mclk = 0;

Set code bellow .Ethernet link down too.

i2s_cfg.i2s_config.use_apll = true;
i2s_cfg.i2s_config.fixed_mclk = 50000000;

TempoTian, 2023-07-28:
You can check
 https://esp32.com/viewtopic.php?t=10674
https://www.esp32.com/viewtopic.php?t=24084
 to see whether it helps.

ArvinHou, 2023-08-06:
use IO0 for ethernet and IO3 for i2s is OK. 
IO0 is input a 50M clock.
I2S is output clock.

closed 2023-08-06 by ArvinHou