espressif/esp-adf #1049: how to record and play at same time

ArvinHou, 2023-08-06:
I want to use   codec->i2s->raw  and   raw->i2s->codec   at same  time.
I have no idea.
1.test   codec->i2s-raw  OK
Raw data is sent to the computer by UDP and played without any issues
2.test raw->i2s-codec OK
Raw data is receive from computer by UDP and played without any issues
3.use at same  time  only  raw->i2s-codec  works OK
I create 2 pipeline
'''
audio_pipeline_register(pipeline, i2s_stream_reader, "i2s");
audio_pipeline_register(pipeline, raw_read, "raw");

ESP_LOGI(TAG, "[3.4] Link it together [codec_chip]-->i2s_stream->raw stream");
const char *link_tag[2] = {"i2s","raw"};
audio_pipeline_link(pipeline, &link_tag[0], 2);

audio_pipeline_register(pipeline_play, i2s_stream_writer, "i2s2");
audio_pipeline_register(pipeline_play, raw_write, "raw2");
ESP_LOGI(TAG, "[4.4] Link it together raw_stream->i2s stream-->[codec_chip]");
const char *link_tag2[2] = {"raw2","i2s2"};
audio_pipeline_link(pipeline_play, &link_tag2[0], 2);

'''
start both of them.only pipeline_play is OK.

how to do it?

TempoTian, 2023-08-11:
Can you show your full code, I think the issue maybe caused by during i2s read or write you have changed i2s settings such as call install or set clock API.

jason-mao, 2024-08-01:
@ArvinHou Do you have any progress of this issue?

closed 2025-02-18 by jason-mao