espressif/esp-adf #1052: error on memory stack while running

be-inexplicable, 2023-08-08:
i am facing this error when initial

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7432
load:0x40078000,len:16564
load:0x40080400,len:4
load:0x40080404,len:4284
entry 0x40080668
I (29) boot: ESP-IDF v5.1-dirty 2nd stage bootloader
I (29) boot: compile time Aug  8 2023 20:21:05
I (29) boot: Multicore bootloader
I (34) boot: chip revision: v3.0
I (38) qio_mode: Enabling default flash chip QIO
I (43) boot.esp32: SPI Speed      : 80MHz
I (48) boot.esp32: SPI Mode       : QIO
I (52) boot.esp32: SPI Flash Size : 4MB
I (57) boot: Enabling RNG early entropy source...
I (62) boot: Partition Table:
I (66) boot: ## Label            Usage          Type ST Offset   Length
I (73) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (80) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (88) boot:  2 factory          factory app      00 00 00010000 00300000
I (95) boot: End of partition table
I (100) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=63e9ch (409244) map
I (221) esp_image: segment 1: paddr=00073ec4 vaddr=3ffb0000 size=03b54h ( 15188) load
I (226) esp_image: segment 2: paddr=00077a20 vaddr=40080000 size=085f8h ( 34296) load
I (238) esp_image: segment 3: paddr=00080020 vaddr=400d0020 size=cfddch (851420) map
I (472) esp_image: segment 4: paddr=0014fe04 vaddr=400885f8 size=16b9ch ( 93084) load
I (518) boot: Loaded app from partition at offset 0x10000
I (518) boot: Disabling RNG early entropy source...
I (529) cpu_start: Multicore app
I (530) quad_psram: This chip is ESP32-D0WD
I (530) esp_psram: Found 8MB PSRAM device
I (532) esp_psram: Speed: 80MHz
I (536) esp_psram: PSRAM initialized, cache is in low/high (2-core) mode.
W (543) esp_psram: Virtual address not enough for PSRAM, map as much as we can. 4MB is mapped
I (552) cpu_start: Pro cpu up.
I (556) cpu_start: Starting app cpu, entry point is 0x40081968
I (0) cpu_start: App cpu up.
I (1076) esp_psram: SPI SRAM memory test OK
I (1084) cpu_start: Pro cpu start user code
I (1084) cpu_start: cpu freq: 240000000 Hz
I (1084) cpu_start: Application information:
I (1087) cpu_start: Project name:     toy1
I (1092) cpu_start: App version:      1
I (1097) cpu_start: Compile time:     Aug  8 2023 20:19:50
I (1103) cpu_start: ELF file SHA256:  14a52be0b5d29f5b...
I (1109) cpu_start: ESP-IDF:          v5.1-dirty
I (1114) cpu_start: Min chip rev:     v0.0
I (1119) cpu_start: Max chip rev:     v3.99 
I (1124) cpu_start: Chip rev:         v3.0
I (1129) heap_init: Initializing. RAM available for dynamic allocation:
I (1136) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1142) heap_init: At 3FFB8590 len 00027A70 (158 KiB): DRAM
I (1149) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1155) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1162) heap_init: At 4009F194 len 00000E6C (3 KiB): IRAM
I (1168) esp_psram: Adding pool of 4096K of PSRAM memory to heap allocator
I (1176) spi_flash: detected chip: gd
I (1180) spi_flash: flash io: qio
W (1184) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
W (1197) ADC: legacy driver is deprecated, please migrate to `esp_adc/adc_oneshot.h`
W (1205) i2s(legacy): legacy i2s driver is deprecated, please migrate to use driver/i2s_std.h, driver/i2s_pdm.h or driver/i2s_tdm.h
I (1218) app_start: Starting scheduler on CPU0
I (1223) app_start: Starting scheduler on CPU1
I (1223) main_task: Started on CPU0
I (1233) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1241) main_task: Calling app_main()
I (1271) MODE_SELECTION: [ 2 ] Start codec chip
E (1271) gpio: gpio_install_isr_service(498): GPIO isr service already installed
E (1292) gpio: gpio_install_isr_service(498): GPIO isr service already installed
I (1293) MODE_SELECTION: Waiting for wifi connected
W (3799) PERIPH_WIFI: Wi-Fi disconnected from SSID Pixel_2261, auto-reconnect enabled, reconnect after 1000 ms
W (7205) PERIPH_WIFI: Wi-Fi disconnected from SSID Pixel_2261, auto-reconnect enabled, reconnect after 1000 ms
W (10611) PERIPH_WIFI: Wi-Fi disconnected from SSID Pixel_2261, auto-reconnect enabled, reconnect after 1000 ms
---- Closed the serial port COM5 ----
---- Opened the serial port COM5 ----
W (71994) AUDIO_EVT: There is no space in external queue
W (71995) PERIPH_WIFI: Wi-Fi disconnected from SSID Pixel_2261, auto-reconnect enabled, reconnect after 1000 ms
W (75405) AUDIO_EVT: There is no space in external queue
W (75406) PERIPH_WIFI: Wi-Fi disconnected from SSID Pixel_2261, auto-reconnect enabled, reconnect after 1000 ms
W (78816) AUDIO_EVT: There is no space in external queue
W (78817) PERIPH_WIFI: Wi-Fi disconnected from SSID Pixel_2261, auto-reconnect enabled, reconnect after 1000 ms
W (82227) AUDIO_EVT: There is no space in external queue
W (82228) PERIPH_WIFI: Wi-Fi disconnected from SSID Pixel_2261, auto-reconnect enabled, reconnect after 1000 ms
W (85638) AUDIO_EVT: There is no space in external queue
W (85639) PERIPH_WIFI: Wi-Fi disconnected 

> `from` SSID Pixel_2261, auto-reconnect enabled, reconnect after 1000 ms

W (89049) AUDIO_EVT: There is no space in external queue
W (89050) PERIPH_WIFI: Wi-Fi disconnected from SSID Pixel_2261, auto-reconnect enabled, reconnect after 1000 ms
W (92460) AUDIO_EVT: There is no space in external queue
W (92461) PERIPH_WIFI: Wi-Fi disconnected from SSID Pixel_2261, auto-reconnect enabled, reconnect after 1000 ms
W (95871) AUDIO_EVT: There is no space in external queue
W (95872) PERIPH_WIFI: Wi-Fi disconnected from SSID Pixel_2261, auto-reconnect enabled, reconnect after 1000 ms
W (99282) AUDIO_EVT: There is no space in external queue
W (99283) PERIPH_WIFI: Wi-Fi disconnected from SSID Pixel_2261, auto-reconnect enabled, reconnect after 1000 ms
I (101395) MODE_SELECTION: Wifi connect fail, please check your settings
I (101395) MODE_SELECTION: [ 3 ] Create and start input key service
I (101398) MODE_SELECTION: [3.0] Create audio pipeline for recording

**_Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
**Core  0 register dump:
PC      : 0x400de332  PS      : 0x00060b30  A0      : 0x800de42c  A1      : 0x3ffba790  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3f408b30  A5      : 0x3ffba7c0  
A6      : 0x3ffba7a0  A7      : 0x0000000c  A8      : 0x800f3c54  A9      : 0x3ffba780  
A10     : 0x3f80f3c0  A11     : 0x00000000  A12     : 0x00000020  A13     : 0x3f80f3e0  
A14     : 0x00000c00  A15     : 0x00000002  SAR     : 0x00000001  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x000000d5  LBEG    : 0x4008d49e  LEND    : 0x4008d4a9  LCOUNT  : 0x00000000  

Backtrace: 0x400de32f:0x3ffba790 0x400de429:0x3ffba7c0 0x400dbe27:0x3ffba7e0 0x4019df5f:0x3ffba890 0x4009402d:0x3ffba8c0**

ELF file SHA256: 14a52be0b5d29f5b_**

jason-mao, 2023-08-10:
@be-inexplicable Could parse the backtrace address by ` xtensa-esp32-elf-addr2line -pfia -e YOUR.elf YOUR Backtrace address `?

be-inexplicable, 2023-08-16:
> @be-inexplicable Could parse the backtrace address by ` xtensa-esp32-elf-addr2line -pfia -e YOUR.elf YOUR Backtrace address `?

i have raised another issue with backtrace isue

closed 2023-08-17 by be-inexplicable