espressif/esp-adf #1054: voip invite request has no ack,and cpu 0&1 is crashed due to no data received.

Bruce213, 2023-08-12:
idf version:4.4.5
adf version:adf-v2.6
board: esp32-lyrt

l can call esp32 from phone, but when i call esp_rtc_answer(esp_sip)，_audio_dec() task will crash because no data recv from rtp.
below is releated log :
thanks in advance！

I (3752) ESP_RTC_CORE: esp_rtc ver: 1.1.0 compile date: Jun 8 2023-12:29:28
I (3752) SIP: Conecting...
I (3762) ESP_RTC_CORE: esp_rtc_init done
W (3762) SIP: CHANGE STATE FROM 0, TO 1, :func: sip_connect:1834
I (3772) SIP: [1970-01-01/00:00:01]=======WRITE 0592 bytes>>
I (3772) SIP:

REGISTER sip:192.168.2.218:5060 SIP/2.0
Via: SIP/2.0/UDP 192.168.2.16:17730;branch=z9hG4bK-2001598745;rport
From: sip:107@192.168.2.218:5060;tag=-2008977675
To: sip:107@192.168.2.218:5060
Contact: sip:107@192.168.2.16:17730;transport=UDP
Max-Forwards: 70
Call-ID: 1C27471EC69B321EA4314587C03A2A3BC075920DECB2
CSeq: 1 REGISTER
Expires: 3600
User-Agent: ESP32 SIP/2.0
Content-Length: 0
Allow: INVITE, ACK, CANCEL, BYE, UPDATE, REFER, MESSAGE, OPTIONS, INFO, SUBSCRIBE
Supported: replaces, norefersub, extended-refer, timer, X-cisco-serviceuri
Allow-Events: presence, kpml

I (3832) SIP: [1970-01-01/00:00:01]=======================>>
I (3852) SIP: [1970-01-01/00:00:01]<<=====READ 0397 bytes==
I (3852) SIP:

SIP/2.0 401 Unauthorized
Via: SIP/2.0/UDP 192.168.2.16:17730;rport;branch=z9hG4bK-2001598745
To: sip:107@192.168.2.218:5060;tag=d7b81d5c
From: sip:107@192.168.2.218:5060;tag=-2008977675
Call-ID: 1C27471EC69B321EA4314587C03A2A3BC075920DECB2
CSeq: 1 REGISTER
WWW-Authenticate: Digest algorithm=MD5,nonce="2D51C62E38FAFA5F76995E2F104A62524E81B42E",realm="newrocktech"
Content-Length: 0

I (3882) SIP: [1970-01-01/00:00:01]<<======================
I (3892) SIP: Required authentication
I (3892) SIP: [1970-01-01/00:00:01]=======WRITE 0790 bytes>>
I (3902) SIP:

REGISTER sip:192.168.2.218:5060 SIP/2.0
Via: SIP/2.0/UDP 192.168.2.16:17730;branch=z9hG4bK--1138121571;rport
From: sip:107@192.168.2.218:5060;tag=115674838
To: sip:107@192.168.2.218:5060
Contact: sip:107@192.168.2.16:17730;transport=UDP
Max-Forwards: 70
Call-ID: 1C27471EC69B321EA4314587C03A2A3BC075920DECB2
CSeq: 2 REGISTER
Expires: 3600
User-Agent: ESP32 SIP/2.0
Content-Length: 0
Allow: INVITE, ACK, CANCEL, BYE, UPDATE, REFER, MESSAGE, OPTIONS, INFO, SUBSCRIBE
Supported: replaces, norefersub, extended-refer, timer, X-cisco-serviceuri
Allow-Events: presence, kpml
Authorization: Digest username="107", realm="newrocktech", nonce="2D51C62E38FAFA5F76995E2F104A62524E81B42E", uri="sip:192.168.2.218:5060", response="1800ea413e06c5ae168cf4434562f4ed", algorithm=MD5

I (3972) SIP: [1970-01-01/00:00:01]=======================>>
I (3992) SIP: [1970-01-01/00:00:01]<<=====READ 0411 bytes==
I (3992) SIP:

SIP/2.0 200 OK
Via: SIP/2.0/UDP 192.168.2.16:17730;rport;branch=z9hG4bK--1138121571
[... 219 lines cut ...]

I (57932) AFE_VC: afe task destroy finished

I (57942) AFE_VC: exit function: afe_destory

W (57942) AUDIO_PIPELINE: There are no listener registered
I (57942) AUDIO_PIPELINE: audio_pipeline_unlinked
W (57952) AUDIO_ELEMENT: [algo] Element has not create when AUDIO_ELEMENT_TERMINATE
W (57962) AUDIO_ELEMENT: [raw] Element has not create when AUDIO_ELEMENT_TERMINATE
I (57972) AV_STREAM: _audio_dec task stoped
I (57972) SIP_SERVICE: ESP_RTC_EVENT_HANGUP
W (57982) SIP: CHANGE STATE FROM 32, TO 2, :func: sip_uas_process_req:1041
I (57982) SIP: [1970-01-01/00:00:28]=======WRITE 0421 bytes>>
I (57992) SIP:

SIP/2.0 200 OK
Via: SIP/2.0/UDP 192.168.2.218:5060;rport;branch=z9hG4bK-63735993616897831741689442079
Contact: sip:107@192.168.2.16:17730
From: sip:200@localhost:5060;tag=16897831441689442076-1
To: sip:107@localhost;tag=819454335;tag=-1809214522
Call-ID: 1689783144503237589-0@192.168.2.218
CSeq: 101141 BYE
Server: ESP32 SIP/2.0
Allow: ACK, INVITE, BYE, UPDATE, CANCEL, OPTIONS, INFO
Content-Length: 0

jason-mao, 2023-08-23:
@Bruce213 I saw some watchdog errors. I think you need to make the two-core loading balance. It means that move some tasks to core0.

Bruce213, 2023-08-23:
> @Bruce213 I saw some watchdog errors. I think you need to make the two-core loading balance. It means that move some tasks to core0.

i don't think this is the root cause , what i saw is :
esp32 device answer the session , but the peer sip phone did not receive the ack from esp32 device.
when i use IDF4.3.4+ADF2.4.1, it is ok to call in or out. but IDF4.4.5+ADF2.6.0 is not OK。
is it any difference between ADF2.4.1 and ADF2.6.0 ?  
thanks in advance。

ahhfzhang, 2023-09-06:
Same issue with https://github.com/espressif/esp-adf/issues/1037.

Bruce213, 2023-09-06:
> Same issue with #1037.

below is  the call log from sip server , i use some client like linphone with the sip server,it is ok , but esp32 device is timeout .

[23/09/07 02:39:32.767]LINE-200(1) offhook
[23/09/07 02:39:37.373]SIP:send[192.168.2.218:5060] rc=843 msg=
INVITE sip:107@localhost SIP/2.0

Via: SIP/2.0/UDP 192.168.2.218:5060;rport;branch=z9hG4bK-63735993616940255771693234064

To: <sip:107@localhost>;frte=FXS;local=c021;rte=FXS17

From: <sip:200@localhost:5060>;tag=16940255771693234063-1

Call-ID: 16940255771929349806-0@192.168.2.218

CSeq: 100660 INVITE

Max-Forwards: 70

Expires: 60000

Contact: <sip:200@192.168.2.218:5060>

Allow: INVITE,ACK,CANCEL,BYE,OPTIONS,INFO,REFER,PRACK,UPDATE

User-Agent: NewRockTech OM20G-2S/2 VER.2.1.5.197/UA 2.0

Content-Type: application/sdp

Content-Length: 293

v=0

o=- 1694025577 367760 IN IP4 192.168.2.218

s=-

c=IN IP4 192.168.2.218

t=0 0

m=audio 3065 RTP/AVP 0 8 18 104 9

a=ptime:20

a=rtpmap:0 PCMU/8000

a=rtpmap:8 PCMA/8000

a=rtpmap:18 G729/8000

a=rtpmap:104 AMR/8000

a=fmtp:104 mode-change-capability=2;max-red=220

a=rtpmap:9 G722/8000

[23/09/07 02:39:37.382]SIP: recv[192.168.2.218:5060],len=[843],msg=
INVITE sip:107@localhost SIP/2.0

Via: SIP/2.0/UDP 192.168.2.218:5060;rport;branch=z9hG4bK-63735993616940255771693234064

To: <sip:107@localhost>;frte=FXS;local=c021;rte=FXS17

From: <sip:200@localhost:5060>;tag=16940255771693234063-1

Call-ID: 16940255771929349806-0@192.168.2.218

CSeq: 100660 INVITE
[... 339 lines cut ...]

Max-Forwards: 70

Content-Length: 0

[23/09/07 02:40:27.622]SIP: recv[192.168.2.30:16260],len=[436],msg=
SIP/2.0 200 OK

Via: SIP/2.0/UDP 192.168.2.218:5060;rport;branch=z9hG4bK-63735993616940256271693234070

Contact: <sip:107@192.168.2.30:16260;transport=UDP>

From: <sip:200@localhost:5060>;tag=16940255771693234067-1

To: <sip:107@localhost>;tag=-359928859;tag=1308089307

Call-ID: 16940255771625200313-0@192.168.2.218

CSeq: 100701 BYE

Server: ESP32 SIP/2.0

Allow: ACK, INVITE, BYE, UPDATE, CANCEL, OPTIONS, INFO

Content-Length: 0