espressif/esp-adf #1059: Lost in Configuration? ESP-ADF Voice Recognition with SPIRAM Setup.

Whiteliondnd8, 2023-08-22:
I ran the code related to voice recognition using ESP-ADF and I'm utilizing the ESP LyraT mini v1.2 development board.

Upon flashing the code, I encountered the following warning:

"W (531) AUDIO_THREAD: Please ensure that you have selected the `CONFIG_SPIRAM_BOOT_INIT` and `CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY` options through the `make menuconfig` command."

Upon investigating this warning, it suggests that I must choose the `CONFIG_SPIRAM_BOOT_INIT` and `CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY` options within the menuconfig. However, I'm having difficulty locating these options.

While navigating through the menuconfig, I discovered an option labeled "HI Lexin." When I attempt to trigger a response by saying "HI Lexin," the device remains unresponsive.

As I'm new to this, I would greatly appreciate any assistance provided.

```
entry 0x40080698
I (27) boot: ESP-IDF v4.4.5-300-ga49e0180ee-dirty 2nd stage bootloader
I (27) boot: compile time 15:40:52
I (28) boot: chip revision: v3.0
I (32) boot.esp32: SPI Speed      : 40MHz
I (37) boot.esp32: SPI Mode       : DIO
I (41) boot.esp32: SPI Flash Size : 2MB
I (46) boot: Enabling RNG early entropy source...
I (51) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (70) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (77) boot:  2 factory          factory app      00 00 00010000 00100000
I (85) boot: End of partition table
I (89) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1fe18h (130584) map
I (145) esp_image: segment 1: paddr=0002fe40 vaddr=3ffb0000 size=001d8h (   472) load
I (145) esp_image: segment 2: paddr=00030020 vaddr=400d0020 size=50dcch (331212) map
I (270) esp_image: segment 3: paddr=00080df4 vaddr=3ffb01d8 size=01b4ch (  6988) load
I (273) esp_image: segment 4: paddr=00082948 vaddr=40080000 size=0defch ( 57084) load
I (306) boot: Loaded app from partition at offset 0x10000
I (306) boot: Disabling RNG early entropy source...
I (318) cpu_start: Pro cpu up.
I (318) cpu_start: Starting app cpu, entry point is 0x4008137c
0x4008137c: call_start_cpu1 at E:/esp/components/esp_system/port/cpu_start.c:147

I (0) cpu_start: App cpu up.
I (332) cpu_start: Pro cpu start user code
I (332) cpu_start: cpu freq: 160000000
I (332) cpu_start: Application information:
I (337) cpu_start: Project name:     example_vad
I (342) cpu_start: App version:      v2.6
I (347) cpu_start: Compile time:     Aug 22 2023 15:39:21
I (353) cpu_start: ELF file SHA256:  e6d18b2f9fb6b601...
I (359) cpu_start: ESP-IDF:          v4.4.5-300-ga49e0180ee-dirty
I (366) cpu_start: Min chip rev:     v0.0
I (370) cpu_start: Max chip rev:     v3.99
I (375) cpu_start: Chip rev:         v3.0
I (380) heap_init: Initializing. RAM available for dynamic allocation:
I (387) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (393) heap_init: At 3FFB3898 len 0002C768 (177 KiB): DRAM
I (399) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (406) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (412) heap_init: At 4008DEFC len 00012104 (72 KiB): IRAM
I (420) spi_flash: detected chip: generic
I (423) spi_flash: flash io: dio
W (427) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (441) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (451) EXAMPLE-VAD: [ 1 ] Start codec chip
W (481) I2C_BUS: I2C bus has been already created, [port:0]
I (491) EXAMPLE-VAD: [ 2 ] Create audio pipeline for recording
I (491) EXAMPLE-VAD: [2.1] Create i2s stream to read audio data from codec chip
I (501) EXAMPLE-VAD: [2.2] Create filter to resample audio data
I (501) EXAMPLE-VAD: [2.3] Create raw to receive data
I (511) EXAMPLE-VAD: [ 3 ] Register all elements to audio pipeline
I (521) EXAMPLE-VAD: [ 4 ] Link elements together [codec_chip]-->i2s_stream-->filter-->raw-->[VAD]
I (531) EXAMPLE-VAD: [ 5 ] Start audio_pipeline
W (531) AUDIO_THREAD: Make sure selected the `CONFIG_SPIRAM_BOOT_INIT` and `CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY` by `make menuconfig`
I (551) EXAMPLE-VAD: [ 6 ] Initialize VAD handle

```

jason-mao, 2023-08-23:
@Whiteliondnd8 Please check the "ESP32S3_SPIRAM_SUPPORT" on IDF v4.4.5

JosephTang, 2023-08-23:
Hi @Whiteliondnd8 
If you can not find configuration such as `CONFIG_SPIRAM_BOOT_INIT` within the menuconfig, you can just type `/` within menuconfig and than type the keywords you want to search. you will find the configuration and its location.
So you can enable those configurations as it suggested, and try again.
Moreover, those configuration is enabled as default in example `wwe`, so if you are using this example, delete the `sdkconfig` and than rebuild the project is anther option.
Thanks.

closed 2024-08-01 by jason-mao