espressif/esp-adf #1060: cannot build with IDF 5.0 CMD

0x0fe, 2023-08-22:
So, i try to build one of the example project of last ADF 2.6, here are the steps i followed 
-download ADF 2.6 and unzip 
-set ADF_PATH from IDF 5.0 CMD with **set ADF_PATH=C:\Users\xxxx\Downloads\esp-adf-2.6**
-copy one example as work directory : **xcopy /e /i %ADF_PATH%\examples\get-started\play_mp3_control play_mp3_control**
-cd to the work directory : **cd C:\Users\xxxx\Downloads\play_mp3_control\**
-set the target : **idf.py set-target esp32**
-attempt to build : **idf.py build**

unfortunately, it cannot build, with the following errors : 

```
C:\Users\xxxx\Downloads\play_mp3_control>idf.py build
Executing action: all (aliases: build)
Running cmake in directory c:\users\xxxx\downloads\play_mp3_control\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=1 c:\users\xxxx\downloads\play_mp3_control"...
fatal: not a git repository (or any of the parent directories): .git
-- IDF_TARGET not set, using default target: esp32
-- Found Git: C:/Espressif/tools/idf-git/2.39.2/cmd/git.exe (found version "2.39.2.windows.1")
-- Component directory C:/Users/xxxx/Downloads/esp-adf-2.6/components/esp-adf-libs does not contain a CMakeLists.txt file. No component will be added
-- Component directory C:/Users/xxxxx/Downloads/esp-adf-2.6/components/esp-sr does not contain a CMakeLists.txt file. No component will be added
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
Processing 2 dependencies:
[1/2] espressif/jsmn (1.1.0)
[2/2] idf (5.0.1)
CMake Error at C:/Espressif/frameworks/esp-idf-v5.0.1/tools/cmake/build.cmake:245 (message):
  Failed to resolve component 'esp-adf-libs'.
Call Stack (most recent call first):
  C:/Espressif/frameworks/esp-idf-v5.0.1/tools/cmake/build.cmake:281 (__build_resolve_and_add_req)
  C:/Espressif/frameworks/esp-idf-v5.0.1/tools/cmake/build.cmake:282 (__build_expand_requirements)
  C:/Espressif/frameworks/esp-idf-v5.0.1/tools/cmake/build.cmake:288 (__build_expand_requirements)
  C:/Espressif/frameworks/esp-idf-v5.0.1/tools/cmake/build.cmake:574 (__build_expand_requirements)
  C:/Espressif/frameworks/esp-idf-v5.0.1/tools/cmake/project.cmake:440 (idf_build_process)
  CMakeLists.txt:10 (project)
ho
-- Configuring incomplete, errors occurred!
See also "C:/Users/xxxx/Downloads/play_mp3_control/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1, output of the command is in the c:\users\xxxx\downloads\play_mp3_control\build\log\idf_py_stderr_output_52132 and c:\users\xxxx\downloads\play_mp3_control\build\log\idf_py_stdout_output_52132
```

But IDF 5.0 CMD works fine for IDF projects

ftab, 2023-08-22:
Downloading the source zip from the GitHub release does not include the submodules. This is an unfortunate drawback for complex projects like esp-idf and esp-adf which make heavy use of submodules.

Here are the components for v2.6. Note those listed in blue through GitHub.

https://github.com/espressif/esp-adf/tree/v2.6/components

They have to be downloaded separately and placed at that location so that it can find them.

Alternatively, you can use git commands to get esp-adf and all submodules recursively, as described at the bottom of the release notes.

0x0fe, 2023-08-23:
@ftab yes indeed, after a recursive clone i was able to get all submodules and to build correctly.

closed 2023-08-23 by 0x0fe