espressif/esp-adf #1061: VOIP 运行过程中esp_rtc库报错

shangpeng01, 2023-08-23:
esp-adf-v2.6+esp-idf-v4.4.5
VoIP应用运行过程中，esp_rtc库报错，打印信息如下
E (6408319) TRANS_TCP: tcp_poll_read select error 0, errno = Success, fd = 55
E (6408327) TRANS_TCP: tcp_poll_read select error 0, errno = Success, fd = 55
E (6408335) TRANS_TCP: tcp_poll_read select error 0, errno = Success, fd = 55
E (6408343) TRANS_TCP: tcp_poll_read select error 0, errno = Success, fd = 55
E (6408351) TRANS_TCP: tcp_poll_read select error 0, errno = Success, fd = 55
E (6408358) TRANS_TCP: tcp_poll_read select error 0, errno = Success, fd = 55
E (6408366) TRANS_TCP: tcp_poll_read select error 0, errno = Success, fd = 55
E (6408374) TRANS_TCP: tcp_poll_read select error 0, errno = Success, fd = 55
E (6408382) TRANS_TCP: tcp_poll_read select error 0, errno = Success, fd = 55
E (6408390) TRANS_TCP: tcp_poll_read select error 0, errno = Success, fd = 55
E (6408397) TRANS_TCP: tcp_poll_read select error 0, errno = Success, fd = 55
E (6408405) TRANS_TCP: tcp_poll_read select error 0, errno = Success, fd = 55
E (6408413) TRANS_TCP: tcp_poll_read select error 0, errno = Success, fd = 55
E (6408421) TRANS_TCP: tcp_poll_read select error 0, errno = Success, fd = 55
E (6408429) TRANS_TCP: tcp_poll_read select error 0, errno = Success, fd = 55
E (6408436) TRANS_TCP: tcp_poll_read select error 0, errno = Success, fd = 55
E (6408444) TRANS_TCP: tcp_poll_read select error 0, errno = Success, fd = 55
E (6408452) TRANS_TCP: tcp_poll_read select error 0, errno = Success, fd = 55
E (6408460) TRANS_TCP: tcp_poll_read select error 0, errno = Success, fd = 55
E (6408468) TRANS_TCP: tcp_poll_read select error 0, errno = Success, fd = 55
E (6408476) TRANS_TCP: tcp_poll_read select error 0, errno = Success, fd = 55
E (6408483) TRANS_TCP: tcp_poll_read select error 0, errno = Success, fd = 55
E (6408491) TRANS_TCP: tcp_poll_read select error 0, errno = Success, fd = 55
E (6408499) TRANS_TCP: tcp_poll_read select error 0, errno = Success, fd = 55
E (6408507) TRANS_TCP: tcp_poll_read select error 0, errno = Success, fd = 55
E (6408515) TRANS_TCP: tcp_poll_read select error 0, errno = Success, fd = 55
E (6408522) TRANS_TCP: tcp_poll_read select error 0, errno = Success, fd = 55
E (6408530) TRANS_TCP: tcp_poll_read seleE (6408534) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (6408534) task_wdt:  - IDLE (CPU 0)
E (6408534) task_wdt: Tasks currently running:
E (6408534) task_wdt: CPU 0: sip_task
E (6408534) task_wdt: CPU 1: IDLE
E (6408534) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x420B9B06:0x3FC9B4C0 0x4037741D:0x3FC9B4E0 0x42006191:0x3D837DE0 0x42006C69:0x3D837E00 0x42005C49:0x3D837E20 0x420054A1:0x3D837E40 0x421420F9:0x3D837E60 0x4213FF8A:0x3D837E80 0x42140012:0x3D837EA0 0x42140D85:0x3D837ED0 0x42156291:0x3D837F00 0x4214D42F:0x3D837F20 0x4214D5A5:0x3D838230 0x42168A91:0x3D838260 0x40386D29:0x3D838290 0x420AC207:0x3D8382E0 0x4216256B:0x3D838330 0x420AC276:0x3D838350 0x42162537:0x3D838370 0x420A6489:0x3D838390 0x420A6F81:0x3D8383C0
0x420b9b06: task_wdt_isr at C:/Espressif/frameworks/esp-idf-v4.4.5/components/esp_system/task_wdt.c:162 (discriminator 1)

0x4037741d: _xt_lowint1 at C:/Espressif/frameworks/esp-idf-v4.4.5/components/freertos/port/xtensa/xtensa_vectors.S:1114

0x42006191: usb_serial_jtag_tcgetattr at C:/Espressif/frameworks/esp-idf-v4.4.5/components/vfs/vfs_usb_serial_jtag.c:326

0x42006c69: uart_write at C:/Espressif/frameworks/esp-idf-v4.4.5/components/vfs/vfs_uart.c:200

0x42005c49: console_open at C:/Espressif/frameworks/esp-idf-v4.4.5/components/vfs/vfs_console.c:65

0x420054a1: esp_vfs_write at C:/Espressif/frameworks/esp-idf-v4.4.5/components/vfs/vfs.c:431 (discriminator 1)

0x421420f9: __sread at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdio/stdio.c:59

0x4213ff8a: __sflush_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdio/fflush.c:196

0x42140012: _fflush_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdio/fflush.c:272 (discriminator 5)

0x42140d85: __sfvwrite_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdio/fvwrite.c:235

0x42156291: get_arg at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdio/vfprintf.c:2241

0x4214d42f: _vfprintf_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1773 (discriminator 1)

0x4214d5a5: __sbprintf at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdio/vfprintf.c:483

0x42168a91: esp_log_level_set at C:/Espressif/frameworks/esp-idf-v4.4.5/components/log/log.c:136

0x40386d29: esp_log_write at C:/Espressif/frameworks/esp-idf-v4.4.5/components/log/log.c:210

0x420ac207: tcp_poll_read at /builds/adf/esp-adf-libs-source/esp_media_protocols/esp_rtc/esp_rtc_media/transport/rtc_transport_tcp.c:167 (discriminator 2)

0x4216256b: rtc_transport_connect at ??:?

0x420ac276: tcp_close at /builds/adf/esp-adf-libs-source/esp_media_protocols/esp_rtc/esp_rtc_media/transport/rtc_transport_tcp.c:205

0x42162537: rtc_transport_get_default_parent at rtc_transport.c:?

0x420a6489: _sip_parse at /builds/adf/esp-adf-libs-source/esp_media_protocols/esp_rtc/esp_rtc_core/esp_rtc_sip/esp_rtc_sip.c:510

0x420a6f81: sip_register at /builds/adf/esp-adf-libs-source/esp_media_protocols/esp_rtc/esp_rtc_core/esp_rtc_sip/esp_rtc_sip.c:1853
 (inlined by) _sip_task at /builds/adf/esp-adf-libs-source/esp_media_protocols/esp_rtc/esp_rtc_core/esp_rtc_sip/esp_rtc_sip.c:1992

E (6408534) task_wdt: Print CPU 1 backtrace

Backtrace: 0x4037C7C9:0x3FC9BAC0 0x4037741D:0x3FC9BAE0 0x400559DD:0x3FCF4680 |<-CORRUPTED
0x4037c7c9: esp_crosscore_isr at C:/Espressif/frameworks/esp-idf-v4.4.5/components/esp_system/crosscore_int.c:92

0x4037741d: _xt_lowint1 at C:/Espressif/frameworks/esp-idf-v4.4.5/components/freertos/port/xtensa/xtensa_vectors.S:1114

ct error 0, errno = Success, fd = 55
E (6408627) TRANS_TCP: tcp_poll_read select error 0, errno = Success, fd = 55
E (6408635) TRANS_TCP: tcp_poll_read select error 0, errno = Success, fd = 55
E (6408643) TRANS_TCP: tcp_poll_read select error 0, errno = Success, fd = 55
E (6408651) TRANS_TCP: tcp_poll_read select error 0, errno = Success, fd = 55
E (6408658) TRANS_TCP: tcp_poll_read select error 0, errno = Success, fd = 55

backtrace信息分析结果如下
task_wdt_isr
C:/Espressif/frameworks/esp-idf-v4.4.5/components/esp_system/task_wdt.c:259 (discriminator 2)
0x4037741d
_xt_lowint1
C:/Espressif/frameworks/esp-idf-v4.4.5/components/freertos/port/xtensa/xtensa_vectors.S:1114
0x42006191
usb_serial_jtag_tcgetattr
C:/Espressif/frameworks/esp-idf-v4.4.5/components/vfs/vfs_usb_serial_jtag.c:361
0x42006c69
uart_write
C:/Espressif/frameworks/esp-idf-v4.4.5/components/vfs/vfs_uart.c:200
0x42005c49
console_open
C:/Espressif/frameworks/esp-idf-v4.4.5/components/vfs/vfs_console.c:65
0x420054a1
esp_vfs_write
C:/Espressif/frameworks/esp-idf-v4.4.5/components/vfs/vfs.c:1223 (discriminator 4)
0x421420f9
__sread
/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdio/stdio.c:59
0x4213ff8a
__sflush_r
/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdio/fflush.c:196
0x42140012
_fflush_r
/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdio/fflush.c:272 (discriminator 5)
0x42140d85
__sfvwrite_r
/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdio/fvwrite.c:235
0x42156291
get_arg
/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdio/vfprintf.c:2241
0x4214d42f
_vfprintf_r
/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1773 (discriminator 1)
0x4214d5a5
__sbprintf
/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdio/vfprintf.c:483
0x42168a91
esp_log_level_set
C:/Espressif/frameworks/esp-idf-v4.4.5/components/log/log.c:171
0x40386d29
esp_log_write
C:/Espressif/frameworks/esp-idf-v4.4.5/components/log/log.c:171
0x420ac207
tcp_poll_read
/builds/adf/esp-adf-libs-source/esp_media_protocols/esp_rtc/esp_rtc_media/transport/rtc_transport_tcp.c:167 (discriminator 2)
0x4216256b
rtc_transport_connect
??:?
0x420ac276
tcp_close
/builds/adf/esp-adf-libs-source/esp_media_protocols/esp_rtc/esp_rtc_media/transport/rtc_transport_tcp.c:205
0x42162537
rtc_transport_get_default_parent
rtc_transport.c:?
0x420a6489
_sip_parse
/builds/adf/esp-adf-libs-source/esp_media_protocols/esp_rtc/esp_rtc_core/esp_rtc_sip/esp_rtc_sip.c:2324
0x420a6f81
sip_register

jason-mao, 2023-08-23:
@shangpeng01 Thanks for your report; let me check and fix it.

shangpeng01, 2023-08-29:
@jason-mao 这个问题在网络不稳定时会比较容易复现，比如运行过程中把设备连接的WiFi热点关掉

closed 2023-09-04 by espressif-bot

shangpeng01, 2023-09-04:
@jason-mao 请问有验证复现到吗？

jason-mao, 2023-09-04:
是的，你可以基于这个版本进行测试。