espressif/esp-adf #1069: Audio overlap best practice for sound notification over playback

0x0fe, 2023-08-31:
So, in most audio applications, especially for device with limited user interface (1 or few button, no display), it is a common practice to be able to play notification sounds over the current playback. For practical and storage reasons these nottification will almost always be in MP3 format.
I wonder wether there is any "good practice" in Adf o achieve this, maybe with the audio pipeline system? and if not it would be a good idea to implement it.

basically it would mean, being able to temporaraly instanciate a second stream reader (from  sd_mmc for example), MP3 decoder and merge the output to the existing audio pîpeline, with some basic merging of the two signals.

I can see how the merging could be, with a module similar to resampling module for example, which we can insert in the audio pipeline and which takes and input buffer and provides an output buffer, but i fails to see how we can instanciate a second streamreader and mp3 decoder at the same time as the already existing one.

If ever this is achievable with the current IDF apis, any pointer will be very welcome, if not maybe such option should be considered in the future because it is very useful if not necessary for many audio devices, be it for button press feedback, low battery warning, or any other UI notifications.

crispmfc, 2023-08-31:
Hello, there is an example that allows to play multiple audios simultaneously. :)
https://github.com/espressif/esp-adf/tree/master/examples/advanced_examples/downmix_pipeline

0x0fe, 2023-08-31:
@crispmfc nice, i missed this one

crispmfc, 2023-09-01:
Hello, could you please tell me which lines of code and where did you put them to define the sdio pin of sdcard? I've been trying to set the pins for days without success. Not only that, it seems that now I have managed to hardcode the I2S pins and can no longer define them manually from my main.c but hey, it would help me to know which is the correct path at least with sdcard. With i2s I don't have many problems. Thanks in advance.

0x0fe, 2023-09-01:
@0x0fe you have to setup the board related files, and the best way to do this is to create a custom board, there is a topic about that in the documentation. In fact a lot of things are defined in this set of files, including your codec settings, so it is very important to do it properly. Here is a screenshot of the custom board definition files for this specific project:

[image]
[image]
[image]

crispmfc, 2023-09-02:
Thank you, I really appreciate it! I will follow your instructions, since I have been trying solutions for many hours for about 1 week and precisely the custom board is something that I need to do.

0x0fe, 2023-09-02:
@crispmfc OK, you can use this as a base https://github.com/espressif/esp-adf/tree/master/examples/get-started/play_mp3_control/components/my_board

jason-mao, 2023-09-07:
@0x0fe  Has this issue been solved?

0x0fe, 2023-09-08:
hello @jason-mao 
No, this issue has not been solved and is very blocking on our side.

majingjing123, 2023-09-11:
Hi, could you please provide the two audio files you tested? I have already test your example with ADF v2.6 and IDF v5.0.2, but this issue did not recur.

0x0fe, 2023-09-11:
@majingjing123 please find the files attached, CP01.mp3 is the playback asset, Success.mp3 is the "tone".
[CP01.zip](https://github.com/espressif/esp-adf/files/12576298/CP01.zip)
[Success.zip](https://github.com/espressif/esp-adf/files/12576315/Success.zip)

majingjing123, 2023-09-12:
@0x0fe  Hi, i have already use your test music and sdkconfig to test example on ESP32 with board ESP32_LyraT, but unfortunately, i still fail to reproduce this issue. 
Here is my test bin.  You can test it.
[downmix.zip](https://github.com/espressif/esp-adf/files/12581841/downmix.zip)
And could you please provide your test bin to help me reproduce the issue.

0x0fe, 2023-09-12:
@majingjing123 
I cannot test your BIN because the baord has a different codec. No problem to provide a bin to you, but you also wont have the same codec, so i am not sure how it will be useful.
My codec is NAU88C22

majingjing123, 2023-09-12:
@0x0fe 
Hi, our LyraT codec can play all the sample rate.
Do you have our official development board， such as 'ESP32_LyraT' or 'ESP32-S3-Korvo'.
You can use our development board to test the issues. If can't recur this issue, you could check for hardware issues.

0x0fe, 2023-09-12:
@majingjing123 
Oh ok i see, no i dont have official dev board, only the custom design.
I dont see how it could come from the hardware, since it plays fine in all scenario (sd_mmc, https, a2dp sink etc.
In fact on the downmix example when both sources are working together it plays fine, the problem occurs when the playback assets plays alone.

majingjing123, 2023-09-13:
@0x0fe
I would like to confirm that the ADF branch is [https://github.com/espressif/esp-adf/releases](url), and the IDF branch is 
`commit 5181de8ac5ec5e18f04f634da8ce173b7ef5ab73 (HEAD, tag: v5.0.2)
Author: Ivan Grokhotkov <ivan@espressif.com>
Date:   Wed May 10 09:59:18 2023 +0200

    versions: Update version to 5.0.2
`

0x0fe, 2023-09-13:
@majingjing123 
Hi, could the issue be related to mclk? my codec uses (and needs) mclk, i use GPIO3 for this purpose. My board_pin_config.c is correctly set, and the I2S works perfectly fine in all other scenarios i tested, but i wonder weether the downmix could interfere in some way with the mclk.
I ran more tests and i can confirm the problam alwys occurs when both pipeline are setup and only one is running. 
As soon as both are running at the same time ,everythign is perfect, which excludes an issue with sd_mmc for example.
I tried to enable fixed_mclk and disable apll, but it did not change anything. 
By the way is there a way to ensure the default UART port (the logging port) is set to simplex mode (TX only) because as told, i use GPIO3 for mclk and so i would like that the log uart is cleanly set to TX only at the setup. I

majingjing123, 2023-09-14:
@0x0fe 
Hi,i have played your issue music, but the sample rate is 48kHz. The music you created for the project should be at 44.1Khz. Please check it. You'd better to write the downmix data to file.

0x0fe, 2023-09-14:
@majingjing123 
There must be a confusion, the sample i sent was not destined to be played on the board, it was destined to be played on a PC or mobile to hear the issue, so the samplerate is not relevant.
OK i will change the `downmix_set_input_rb_timeout` and see if it helps.

0x0fe, 2023-09-14:
@majingjing123 
Well, setting `downmix_set_input_rb_timeout` solved the issue.

majingjing123, 2023-09-14:
@0x0fe 
Our player currently does not support downmix. You need to integrate this feature into your project by yourself. This example only demonstrate how downmix work.

0x0fe, 2023-09-14:
@majingjing123 
Unfortunately there is now a new problem, while the base playback works fine, and it also works fine when both the base playback and newcome are playing together, as soon as the newcomes finises, the base gets messed up again. I will take a small recording.
I tried to set the downmix_set_input_rb_timeout to 0, 10 and 50, it does not help.

[image]

majingjing123, 2023-09-14:
@0x0fe 
You can comment `downmix_set_input_rb_timeout(downmixer, 0, INDEX_BASE_STREAM);` out， because base stream is still playing, but you set its timeout to 0.

0x0fe, 2023-09-14:
@majingjing123 
No, it doesnt work.
Commenting `downmix_set_input_rb_timeout(downmixer, 0, INDEX_BASE_STREAM);` totally cuts all audio when the newcome is finished
commenting both 
```
downmix_set_input_rb_timeout(downmixer, 0, INDEX_BASE_STREAM);
downmix_set_input_rb_timeout(downmixer, 0, INDEX_NEWCOME_STREAM);
```
leads to massive samplerate issue, similar to before but 10x worst.
I also tried with timeout at 10 and 50 before, it did not sole the issue.

majingjing123, 2023-09-14:
@0x0fe 
Try this code.

`#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_log.h"
#include "audio_element.h"
#include "audio_pipeline.h"
#include "audio_event_iface.h"
#include "i2s_stream.h"
#include "mp3_decoder.h"
#include "fatfs_stream.h"
#include "downmix.h"
#include "filter_resample.h"
#include "raw_stream.h"
#include "board.h"
#include "periph_sdcard.h"
#include "periph_button.h"
#include "input_key_service.h"
#include "periph_adc_button.h"

static const char *TAG = "DOWNMIX_PIPELINE_EXAMPLE";

#define INDEX_BASE_STREAM 0
#define INDEX_NEWCOME_STREAM 1
#define SAMPLERATE 44100
#define NUM_INPUT_CHANNEL 2
#define TRANSMITTIME 500
#define MUSIC_GAIN_DB 0
#define PLAY_STATUS ESP_DOWNMIX_OUTPUT_TYPE_TWO_CHANNEL //ESP_DOWNMIX_OUTPUT_TYPE_ONE_CHANNEL
#define NUMBER_SOURCE_FILE 2

void app_main(void)
{
    audio_element_handle_t base_fatfs_reader_el = NULL;
    audio_element_handle_t base_mp3_decoder_el = NULL;
    //audio_element_handle_t base_rsp_filter_el = NULL;
    audio_element_handle_t base_raw_write_el = NULL;

    audio_element_handle_t newcome_fatfs_reader_el = NULL;
    audio_element_handle_t newcome_mp3_decoder_el = NULL;
    //audio_element_handle_t newcome_rsp_filter_el = NULL;
    audio_element_handle_t newcome_raw_write_el = NULL;

    esp_log_level_set("*", ESP_LOG_WARN);
    esp_log_level_set(TAG, ESP_LOG_INFO);

    audio_board_handle_t board_handle = audio_board_init();
    audio_hal_ctrl_codec(board_handle->audio_hal, AUDIO_HAL_CODEC_MODE_DECODE, AUDIO_HAL_CTRL_START);

    esp_periph_config_t periph_cfg = DEFAULT_ESP_PERIPH_SET_CONFIG();
    esp_periph_set_handle_t set = esp_periph_set_init(&periph_cfg);
    audio_board_sdcard_init(set, SD_MODE_1_LINE);

    audio_board_key_init(set);
    
    input_key_service_info_t input_key_info[] = INPUT_KEY_DEFAULT_INFO();
    input_key_service_cfg_t input_cfg = INPUT_KEY_SERVICE_DEFAULT_CONFIG();
    input_cfg.handle = set;
    periph_service_handle_t input_ser = input_key_service_create(&input_cfg);
    input_key_service_add_key(input_ser, input_key_info, INPUT_KEY_NUM);
    //periph_service_set_callback(input_ser, idle_adkey_cb, NULL);

    ESP_LOGI(TAG, "[3.0] Create pipeline_mix pipeline");
    audio_pipeline_cfg_t pipeline_cfg = DEFAULT_AUDIO_PIPELINE_CONFIG();
    audio_pipeline_handle_t pipeline_mix = audio_pipeline_init(&pipeline_cfg);

    ESP_LOGI(TAG, "[3.1] Create down-mixer element");
[... 206 lines cut ...]
    audio_element_deinit(newcome_mp3_decoder_el);
    //audio_element_deinit(newcome_rsp_filter_el);
    audio_element_deinit(newcome_raw_write_el);

    /* Stop mixer stream pipeline, Release resources */
    audio_pipeline_stop(pipeline_mix);
    audio_pipeline_wait_for_stop(pipeline_mix);
    audio_pipeline_terminate(pipeline_mix);
    audio_pipeline_unregister_more(pipeline_mix, downmixer, i2s_writer, NULL);
    audio_pipeline_remove_listener(pipeline_mix);

    /* Stop all peripherals before removing the listener */
    esp_periph_set_stop_all(set);
    audio_event_iface_remove_listener(esp_periph_set_get_event_iface(set), evt);

    /* Make sure audio_pipeline_remove_listener & audio_event_iface_remove_listener are called before destroying event_iface */
    audio_event_iface_destroy(evt);

    /* Release resources */
    audio_pipeline_deinit(pipeline_mix);
    audio_element_deinit(downmixer);
    audio_element_deinit(i2s_writer);
    esp_periph_set_destroy(set);
}
`

0x0fe, 2023-09-14:
@majingjing123 
Yes, it works now, thank you very much for your help debugging the issue. I guess the example should be updated to reflect these corrections.

closed 2023-09-14 by 0x0fe

Jack-agilian, 2024-02-29:
I'm having the similar issue, with new_come_stream alone. My solution was to increase the 'max_sample' in 'downmix_cfg'. The root cause is that stream processing on a node is too slow.