espressif/esp-adf #1070: OTA example cannot build, not fit in region 'iram_0_0_seg'

0x0fe, 2023-08-31:
So, when testing the OTA example from last ADF (2.6), compiled alone (all other sources are disabled by define) it cannot build and the IRAM is overflown by 664 byte. 
There must bea bug somewhere, furthermore that we use ESP32D0QDR2 which has 2 MB internal PSRAM, and it is enabled in sdkconfig, along with all optimisations related to SPIRAM.

If i clean, and then build, it can build (once), but as soon as i try to upload, it fails with the IRAM overflow message. Which doesnt make much sense.

**after upload** or **upload and monitor**

```
Linking .pio\build\esp32dev\firmware.elf
c:/users/xxxx/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: missing --end-group; added as last command line option
c:/users/xxxx/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio/build/esp32dev/firmware.elf section `.iram0.text' will
 not fit in region `iram0_0_seg'
c:/users/xxxx/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: IRAM0 segment data does not fit.
c:/users/xxxx/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: region `iram0_0_seg' overflowed by 664 bytes
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32dev\firmware.elf] Error 1
```

after **clean**, then **build** (but doing upload after that, fails with IRAM overflow).

```
c:/users/xxxx/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: missin
g --end-group; added as last command line option
Retrieving maximum program size .pio\build\esp32dev\firmware.elf
Checking size .pio\build\esp32dev\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   8.6% (used 28248 bytes from 327680 bytes)
Flash: [====      ]  37.3% (used 1245365 bytes from 3342336 bytes)
Building .pio\build\esp32dev\firmware.bin
```

jason-mao, 2023-09-05:
@0x0fe I'm not repeating your problem. I think you can disable all the IRAM optimization option in sdkconfig, such as ESP32_WIFI_IRAM_OPT, ESP32_WIFI_RX_IRAM_OPT, etc.

jason-mao, 2023-09-05:
Same issue as https://github.com/espressif/esp-adf/issues/917

0x0fe, 2023-09-10:
@jason-mao all IRAM optimisations were disabled already, the only option which allowed to build was to set 
`CONFIG_FREERTOS_PLACE_FUNCTIONS_INTO_FLASH`
Which was advised by yourself in another issue ticket (i cannot find it at the moment).