espressif/esp-adf #1071: WWE示例程序运行报错I2S端口被占用，使用ESP32-LyraTD-MSC V2.2开发板

ZaynQiu, 2023-09-01:
## Environment

- Audio development kit:      [ESP32-LyraTD-MSC V2.2]
- Audio kit version (for ESP32-LyraT): [v2]
-  Module or chip used:  [ESP32-WROVER-E]
-  IDF version: 5.1.1
- ADF version: v2.6
- Build system: idf.py
- Operating system:     [Linux]
- Using an IDE?: [No]
- Power supply: [USB]

## Problem Description

运行例程后，提示I2S端口已经被占用

### Expected Behavior
正常运行示例程序，不报错
### Actual Behavior
示例程序报错，程序不能正常运行
### Steps to Reproduce

1. 复制示例程序至工作文件夹
2. 根据README文件中所示，更改zl38063.c中函数
3. 选择ESP32-LyraTD-MSC开发板`idf.py menuconfig`
4. 构建项目`idf.py build`
5. 烧录程序`idf.py -p /dev/ttyUSB0 flash monitor` (更改38063函数后，程序烧录时一种重启，此时已经更换为初始的函数代码)

## Debug Logs

```
rst:0x1 (POWERON_RESET),boot:0x1e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7432
load:0x40078000,len:16576
load:0x40080400,len:4
0x40080400: _init at ??:?

ho 8 tail 4 room 4
load:0x40080404,len:4284
entry 0x40080668
I (31) boot: ESP-IDF v5.1.1-dirty 2nd stage bootloader
I (31) boot: compile time Sep  1 2023 22:44:29
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v1.0
I (39) qio_mode: Enabling default flash chip QIO
I (45) boot.esp32: SPI Speed      : 80MHz
I (49) boot.esp32: SPI Mode       : QIO
I (54) boot.esp32: SPI Flash Size : 4MB
I (58) boot: Enabling RNG early entropy source...
I (64) boot: Partition Table:
I (67) boot: ## Label            Usage          Type ST Offset   Length
I (75) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (82) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (89) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (97) boot:  3 ota_0            OTA app          00 10 00010000 00300000
I (104) boot:  4 flash_tone       Unknown data     01 27 00310000 00032000
I (112) boot: End of partition table
I (116) boot: No factory image, trying OTA 0
I (121) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1979c8h (1669576) map
I (588) esp_image: segment 1: paddr=001a79f0 vaddr=3ffb0000 size=02934h ( 10548) load
I (592) esp_image: segment 2: paddr=001aa32c vaddr=40080000 size=05cech ( 23788) load
I (602) esp_image: segment 3: paddr=001b0020 vaddr=400d0020 size=7a3d0h (500688) map
I (740) esp_image: segment 4: paddr=0022a3f8 vaddr=40085cec size=0e71ch ( 59164) load
I (770) boot: Loaded app from partition at offset 0x10000
I (811) boot: Set actual ota_seq=1 in otadata[0]
I (811) boot: Disabling RNG early entropy source...
I (821) cpu_start: Multicore app
[... 595 lines cut ...]
# CONFIG_RINGBUF_PLACE_FUNCTIONS_INTO_FLASH is not set
# end of ESP Ringbuf

#
# ESP System Settings
#
# CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ_80 is not set
# CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ_160 is not set
CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ_240=y
CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ=240

#
# Memory
#
# CONFIG_ESP32_USE_FIXED_STATIC_RAM_SIZE is not set

#
# Non-backward compatible options
#
# CONFIG_ESP_SYSTEM_ESP32_SRAM1_REGION_AS_IRAM is not set
# end of Non-backward compatible options
# end of Memory

# End of deprecated options
```

jason-mao, 2023-09-05:
@ZaynQiu Thanks for your report, let we check and fix.

JosephTang, 2023-09-08:
@ZaynQiu 
1. log中的提示的i2s 被占用，是因为idf 5.1 增加的提示，但不影响使用。
2. 我这边使用默认代码，不做任何修改，配合idf 5.1 配合 esp-sr 的master 分支是没有问题的。建议使用这样的配置。后续ADF会更新esp-sr。

谢谢。

ZaynQiu, 2023-09-10:
@JosephTang 感谢解答！

closed 2023-09-22 by espressif-bot