espressif/esp-adf #1072: Reproducible builds

ftab, 2023-09-01:
**Is your feature request related to a problem? Please describe.**

CONFIG_APP_REPRODUCIBLE_BUILD is not handled by esp-adf, so some objects have local paths embedded in uses of the `__FILE__` macro. There are other differences in the .elf files between two different executions of the same or similar build environments, and thus it is difficult/impossible to reproduce a build precisely. I am not sure how much of this is esp-adf's fault or if there is still some non-reproducibility in esp-idf. My app uses esp-adf extensively and I cannot figure out how to recreate the same builds.

I used the same Docker image on my machine in two different ways and it produced different binaries (one through GitLab CI/CD, one through `docker run ...` at the command line). Another machine produced yet another different binary with the same Docker image and `docker run` command. This then makes the build unknown to stuff like [ESP Insights](https://insights.espressif.com/) unless each developer either uploads their specific copies of the build as well, or only ever flashes boards using OTA updates / binaries downloaded from CI/CD.

**Describe the solution you'd like**

Building the same application from the same git checkout should produce the same binaries if:
* CONFIG_APP_REPRODUCIBLE_BUILD is enabled
* we are using the same version of esp-adf and esp-idf on the two machines and 
* the two machines are similar in build environment (e.g. both Ubuntu 20.04, latest cmake & ninja installed with pip instead of apt, or both using the same Docker image)

Differences like the path to adf or esp tools ideally should not create differences in the binaries, and they should produce the same .elf file since the SHA256 of this is embedded in the app.

For example, CMakeLists.txt should have something similar to this (or whatever the equivalent may be that makes it work alongside ESP-IDF's counterparts)

```cmake
    if(CONFIG_COMPILER_HIDE_PATHS_MACROS)
        list(APPEND compile_options "-fmacro-prefix-map=${ADF_PATH}=/ADF")
    endif()

    if(CONFIG_APP_REPRODUCIBLE_BUILD)
        list(APPEND compile_options "-fdebug-prefix-map=${ADF_PATH}=/ADF")
    endif()
```

There may be more tweaks needed to get reproducible that I'm not currently aware of.

**Describe alternatives you've considered**

I am considering whether we have to rely solely on CI/CD builds / OTA updates for flashing, since we can't reproduce the builds on demand. Sounds a little painful.

jason-mao, 2023-09-05:
@ftab Thank you for bringing this up, it will be fixed on next major version.

jgeudens, 2025-08-25:
Is there any progress on this topic?