espressif/esp-adf #1074: recorder_engine:

stintel, 2023-09-06:
## Environment

- Audio development kit:      ESP32-S3-Box-Lite
- Module or chip used:  ESP32-S3-WROOM-1
- IDF version: v4.4.5
- ADF version: v2.6-4-g5c7d67ec
- Build system:         idf.py
- Running log:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x2a (SPI_FAST_FLASH_BOOT)
Saved PC:0x40048d00
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x162c
load:0x403c9700,len:0xbcc
load:0x403cc700,len:0x3024
entry 0x403c9954
I (24) boot: ESP-IDF v4.4.5 2nd stage bootloader
I (25) boot: compile time 16:42:10
I (25) boot: chip revision: v0.1
I (26) boot.esp32s3: Boot SPI Speed : 80MHz
I (31) boot.esp32s3: SPI Mode       : DIO
I (36) boot.esp32s3: SPI Flash Size : 2MB
I (41) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00100000
I (79) boot: End of partition table
I (84) esp_image: segment 0: paddr=00010020 vaddr=3c060020 size=28330h (164656) map
I (122) esp_image: segment 1: paddr=00038358 vaddr=3fc91a00 size=025c0h (  9664) load
I (124) esp_image: segment 2: paddr=0003a920 vaddr=40374000 size=056f8h ( 22264) load
I (132) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=59aa0h (367264) map
I (202) esp_image: segment 4: paddr=00099ac8 vaddr=403796f8 size=08300h ( 33536) load
I (215) boot: Loaded app from partition at offset 0x10000
I (215) boot: Disabling RNG early entropy source...
I (226) cpu_start: Pro cpu up.
I (226) cpu_start: Starting app cpu, entry point is 0x403752ac
0x403752ac: call_start_cpu1 at /opt/esp/idf/components/esp_system/port/cpu_start.c:147

I (0) cpu_start: App cpu up.
I (240) cpu_start: Pro cpu start user code
I (241) cpu_start: cpu freq: 160000000
I (241) cpu_start: Application information:
I (243) cpu_start: Project name:     esp_adf_skel
I (249) cpu_start: App version:      b35972e
I (254) cpu_start: Compile time:     Sep  6 2023 16:42:05
I (260) cpu_start: ELF file SHA256:  b3ae7fd0e41ec34b...
I (266) cpu_start: ESP-IDF:          v4.4.5
I (271) cpu_start: Min chip rev:     v0.0
I (275) cpu_start: Max chip rev:     v0.99
I (280) cpu_start: Chip rev:         v0.1
I (285) heap_init: Initializing. RAM available for dynamic allocation:
I (292) heap_init: At 3FC95C60 len 00053AB0 (334 KiB): D/IRAM
I (298) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DIRAM
I (305) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (311) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (318) spi_flash: detected chip: gd
I (322) spi_flash: flash io: dio
W (326) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (340) sleep: Configure to isolate all GPIO pins in sleep state
I (346) sleep: Enable automatic switching of GPIO sleep configuration
I (353) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (374) ESP_ADF_SKEL: starting
E (374) REC_ENG: Invalid parameters
E (374) ESP_ADF_SKEL: failed to create recorder engine
I (384) ESP_ADF_SKEL: running main loop
```
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch5) 8.4.0
- Operating system:     Linux
- Using an IDE?: No
- Power supply:         USB

## Problem Description

We're investigating [recorder_engine](https://espressif-docs.readthedocs-hosted.com/projects/esp-adf/en/latest/api-reference/speech-recognition/recorder_engine.html) for use in [Willow](https://github.com/toverainc/willow). It is unclear from the documentation how to properly use it, and there are no examples using it.

### Expected Behavior

Recorder engine initializes.

### Actual Behavior

Recorder engine fails to initialize. Error message is not useful:

```
E (373) REC_ENG: Invalid parameters
```

### Steps to Reproduce

See https://github.com/toverainc/esp-skel/commit/6c89e0e438133d4fdbdeea720e36cf113e88dd7e

### Code to Reproduce This Issue

See https://github.com/toverainc/esp-skel/tree/bugreport/rec_engine/adf.

## Debug Logs

```
Debug log goes here. It should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plaintext here for us to search the error log. Or attach the complete logs and leave the main part here if the log is *too* long.
```

## Other Items If Possible

- [ ] sdkconfig file (Attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**Note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

jason-mao, 2023-09-07:
@stintel  An open source, compatible and extensible component `audio_recorder` is recommended instead of `recorder_engine`.

stintel, 2023-09-11:
Hi @jason-mao. Thanks for getting back to me. We're currently using audio_recorder in Willow. If use of audio_recorder is recommended instead of recorder_engine, it would be a good idea to mention that in the recorder_engine docs.

jason-mao, 2023-09-13:
@stintel Thanks  for your advice, I will fixed it.

closed 2023-09-22 by espressif-bot