espressif/esp-adf #1075: esp_player and A2DP (aadp) issue with sink mode, and source mode

0x0fe, 2023-09-11:
ADF version 2.6
IDF version 5.0.2
Chip : ESP32D0WDR2 V3.1
IDE : VSCode / PIO

The device is an audio player providing 3 sources : sd_mmc, http and a2dp-sink and 2 outputs : i2s and a2dp-source. 

I managed to get all these working with the low level APIs (pipelines and a ton of callback/handlers) but this is already very akward to use with a single source, and pretty unmanageable with multiple source / outputs. 

I discovered the esp_player and esp_player_wrapper APIs, which make things a lot cleaner and easier to use, 
so far i managed to get sd_mmc and http to i2s working fine, however i am facing troubles with a2dp.

So, in the player setup, i added a stream for a2dp, and tried my best to intilize the bluetooth service : 

```cpp
void *player_setup(void *cb, void *ctx){

    esp_audio_cfg_t cfg = DEFAULT_ESP_AUDIO_CONFIG();
    audio_board_handle_t board_handle = audio_board_init();
    audio_hal_ctrl_codec(board_handle->audio_hal, AUDIO_HAL_CODEC_MODE_BOTH, AUDIO_HAL_CTRL_START);

    cfg.vol_handle = board_handle->audio_hal;
    cfg.vol_set = (audio_volume_set)audio_hal_set_volume;
    cfg.vol_get = (audio_volume_get)audio_hal_get_volume;
    cfg.resample_rate = 48000;
    cfg.prefer_type = ESP_AUDIO_PREFER_MEM;
    cfg.cb_func = cb;
    cfg.cb_ctx = ctx;
    handle = esp_audio_create(&cfg);

    // Create readers and add to esp_audio
    fatfs_stream_cfg_t fs_reader = FATFS_STREAM_CFG_DEFAULT();
    fs_reader.type = AUDIO_STREAM_READER;

    // FatFS
    esp_audio_input_stream_add(handle, fatfs_stream_init(&fs_reader));

    // HTTP
    http_stream_cfg_t http_cfg = HTTP_STREAM_CFG_DEFAULT();
    http_cfg.event_handle = _http_stream_event_handle;
    http_cfg.type = AUDIO_STREAM_READER;
    http_cfg.enable_playlist_parser = true;
    audio_element_handle_t http_stream_reader = http_stream_init(&http_cfg);
    esp_audio_input_stream_add(handle, http_stream_reader);

    //http_stream_reader = http_stream_init(&http_cfg); // why twice???? i assume it is an error in the example
    //esp_audio_input_stream_add(handle, http_stream_reader); // why twice???? i assume it is an error in the example

    // A2DP SINK
    bluetooth_service_cfg_t bt_cfg = {
        .device_name = "FABA",
        .mode = BLUETOOTH_A2DP_SINK,
        .remote_name = "",
    };
    bluetooth_service_start(&bt_cfg);
    esp_audio_input_stream_add(handle,bluetooth_service_create_stream());
    //bt_periph = bluetooth_service_create_periph();//should we init manually?
    //esp_periph_start(set, bt_periph);//should we start manually?
    esp_bt_gap_set_scan_mode(ESP_BT_CONNECTABLE, ESP_BT_GENERAL_DISCOVERABLE);// shoud we set these manually?
    // where is AVRC play cmd issued? in the esp_player_music_play function? 

    // Add decoders and encoders to esp_audio
    mp3_decoder_cfg_t mp3_dec_cfg = DEFAULT_MP3_DECODER_CONFIG();
    mp3_dec_cfg.task_core = 1;
    esp_audio_codec_lib_add(handle, AUDIO_CODEC_TYPE_DECODER, mp3_decoder_init(&mp3_dec_cfg));

    // Create writers and add to esp_audio
    i2s_stream_cfg_t i2s_writer = I2S_STREAM_CFG_DEFAULT();
    i2s_writer.type = AUDIO_STREAM_WRITER;
    i2s_writer.i2s_config.sample_rate = 48000;
[... 130 lines cut ...]
        .remote_name = "",
    };
    bluetooth_service_start(&bt_cfg);
    //A2DP SOURCE
    bluetooth_service_cfg_t bt_cfg = {
        .device_name = "FABA",
        .mode = BLUETOOTH_A2DP_SOURCE,
        .remote_name = "",
    };
    bluetooth_service_start(&bt_cfg);
```

are we supposed to merge the two enums? (i doubt)

```
    // A2DP SINK + SOURCE
    bluetooth_service_cfg_t bt_cfg = {
        .device_name = "FABA",
        .mode = BLUETOOTH_A2DP_SINK | BLUETOOTH_A2DP_SOURCE,
        .remote_name = "",
    };
    bluetooth_service_start(&bt_cfg);
```

Thanks in advance and clear example of simple function blocks are really lacking in ADF, we would need clear examples with esp_player for each and every source, at least, plus various other things like OTA without tone partition, BLE PROV (which is not even implemented), etc.

jason-mao, 2023-09-12:
@0x0fe  I think you need the component as  https://github.com/espressif/esp-adf/tree/master/examples/korvo_du1906/components/audio_player. One thing to note is that sink and source cannot be enabled at the same time.

0x0fe, 2023-09-12:
@jason-mao Oh, i just noticed it yesterday and wondered wether it was better than the wrapper i use. Of course sink and source are only used exclusively, we will either use sink to I2S, or sd_mmc/http to source.
I am going to test it and yesterday i noticed the downmix is used inside, may i know if it allows us to play sone tone over the playback? If so i hope it may work here because i still have this problem with the downmix. https://github.com/espressif/esp-adf/issues/1069

0x0fe, 2023-09-14:
@jason-mao 
Sadly, the player component you pointed out does not work correctly, it is quite outdated and when i tried to run the untiy tests i found it was failing many of them, in particular the bluetooth doesnt work.