espressif/esp-adf #1077: 关于双麦克风回声消除（aec）的疑惑

ZorkyZorky, 2023-09-16:
非常不理解四通道的回声消除（aec）是如何实现的。

四通道的encode芯片一般是ES7210，查看了ES7210 user guide一般有两种方式读取四通道。
（1）I2S飞利浦标准
这个需要ES7210的SDOUT1和SDOUT2分别连接引脚，但是查看了esp-box和esp-korvo-2的原理图，SDOUT2引脚均没有连接（NC就是断开了）
[image]
（2）TDM模式
符合只接SDOUT1的只有TDM模式，但是查看了esp-adf、esp-box、esp-skainet的I2S配置，均没有发现使用TDM模式
[image]

（1）被困住很久了，麻烦解答一下双通道的回声消除有没有被实现？
（2）如果是使用的I2S飞利浦标准实现的，需要使用两个I2S通道吗？两个I2S通道的时钟引脚可以共用吗？
（3）如果是使用TDM模式实现的，四通道下decode播放音乐通道该如何处理？

jason-mao, 2023-09-18:
@ZorkyZorky esp-box和esp-korvo-2 使用的是32bit 位宽的一路全双工I2S，不是TDM模式。 这种情况下可以接收4通道16bit的 ADC 数据，2路mic，1路AEC 回采，空余1路；DAC的数据是将解码过后的数据转换为32bit 送给I2S。

ZorkyZorky, 2023-09-20:
但是ES8311的DAC好像最多支持24bit
[image]

jason-mao, 2023-09-20:
扩展的 DAC 数据只有16位有效位，剩余16位是0。I2S 的接收端只要能接收到这有效的 16bit 就可以了。

ZorkyZorky, 2023-09-21:
[image]
好的，还有一个疑惑是ES7210的user guide上说MIC3和MIC4在标准I2S下是从SDOUT2输出，所以为啥在标准I2S下也可以从SDOUT1一起输出？

jason-mao, 2023-09-25:
@ZorkyZorky ES7210 工作的是TDM 模式，它不是在I2S 模式下，S3 实在I2S 模式下。16 bit 4 channel 的TDM 模式数据格式和32bit 2channel I2S 的数据格式是相同的。

closed 2024-07-31 by jason-mao