espressif/esp-adf #1080: ADF VoIP exmaple does not configure dual mic

ukfasthands, 2023-09-23:
## Environment

- Audio development kit: ESP32-S3-Korvo-2
- Audio kit version: v3
- [Required] Module or chip used: ESP32-S3-WROOM-1
- [Required] IDF version (run ``git describe --tags`` in $IDF_PATH folder to find it): v5.1.1-393-g7698a413bc
- [Required] ADF version (run ``git describe --tags`` in $ADF_PATH folder to find it): v2.6-14-g5cfa0ee
- Build system:         idf.py
- [Required] Running log:  
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` in your project folder to find it): crosstool-NG esp-12.2.0_20230208
- Operating system:    Windows
- (Windows only) Environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell]
- Using an IDE?: VSCode
- Power supply:  USB

## Problem Description

The ESP32-S3-Korvo-2 board has 2 microphone inputs and 1 referenace input into the ADC.  The AFE documentation states that the audio input format configuration should be total_ch_num=3, mic_num=2, ref_num=1

The VoIP exmaple in esp-adf\examples\protocols\voip seems to incorrectly configure the AFE.  In esp-adf\examples\protocols\components\av_stream\av_stream.c, it uses the macro ALGORITHM_STREAM_CFG_DEFAULT to configure algorithm_stream_cfg_t algo_config structure in av_audio_enc_start().  This defaults the algo_config.mic_ch to 1.

This results in the AFE been configured as 1 mic and 1 aec referance channel.  Is this correct?  
My assumption would be that the mic_ch should be 2 to make use of the dual microphone on the ESP32-S3-Korvo-2 board?
Does the AEC algorithums using the second microphone to imporve performace of the AEC?

ahhfzhang, 2023-09-25:
Hi @ukfasthands,

Yes, our AEC uses 1 mic and 1 reference channel, AFE uses 2 mic for better voice wake-up performance, but 2 mic won't help the AEC algorithm.

br.

EdHoldsworth, 2023-09-25:
Hi @ahhfzhang,

Thankyou for the quick reply.  Will enabling both microphones improve noise cancellation, or is the only real benifit for the wake-up performance?

Regards

ahhfzhang, 2023-09-25:
Hi @ukfasthands,

We don't recommend enabling both microphones for noise cancellation and yes it will only benifit for the wake-up performance.

br.

closed 2023-09-25 by ukfasthands