espressif/esp-adf #1081: visualcode 1.82.2 error compile

Nexet, 2023-09-23:
hi guys, using windows 11 with the latest version of vlc, I can't compile everything and get the error:
I don't know if it's related to the new versions, but I updated everything, reinstalled everything and I can't find the error.
thanks 

[https://nopaste.net/zWSwti8U5x](url)

jason-mao, 2023-09-25:
@Nexet Which  ADF and IDF versions are used? And did you select your board before building?

Nexet, 2023-09-25:
esp-idf 5.1.1 release version ,the adf version is the latest, download visualcode via the command.
yes I selected esp32 as the board.
the problem exists after copying the audio kit files, before editing I can compile, if I copy these I have the problem. https://github.com/Ai-Thinker-Open/ESP32-A1S-AudioKit

PumpkinLin314, 2023-10-09:
Hi @Nexet 
ESP-IDF v5.1.1 currently requires ESP-ADF 2.6 or master, but ESP32-A1S-AudioKit is not yet adapted to the latest ESP-ADF and ESP-IDF 5.x.

I modified `board_def.h` and `board_pins_config.c` and they compile correctly. Here are the two files below. You also need to change other configurations such as SDcard PINs according to the board design to make ESP-ADF and ESP-IDF work properly.

You can also refer to `$ADF_PATH/components/audio_board/` to modify the board support files.

---
board_def.h
```c
/*
 * @Author: your name
 * @Date: 2020-01-26 10:13:06
 * @LastEditTime : 2020-01-28 19:24:42
 * @LastEditors  : Please set LastEditors
 * @Description: In User Settings Edit
 * @FilePath: \esp-adf\components\audio_board\aithinker\board_def.h
 */
/*
 * ESPRESSIF MIT License
 *
 * Copyright (c) 2019 <ESPRESSIF SYSTEMS (SHANGHAI) CO., LTD>
 *
 * Permission is hereby granted for use on all ESPRESSIF SYSTEMS products, in which case,
 * it is free of charge, to any person obtaining a copy of this software and associated
 * documentation files (the "Software"), to deal in the Software without restriction, including
 * without limitation the rights to use, copy, modify, merge, publish, distribute, sublicense,
 * and/or sell copies of the Software, and to permit persons to whom the Software is furnished
 * to do so, subject to the following conditions:
 *
 * The above copyright notice and this permission notice shall be included in all copies or
 * substantial portions of the Software.
 *
 * THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
 * IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY, FITNESS
 * FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE AUTHORS OR
 * COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER LIABILITY, WHETHER
 * IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM, OUT OF OR IN
 * CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE SOFTWARE.
 *
 */

#ifndef _AUDIO_BOARD_DEFINITION_H_
#define _AUDIO_BOARD_DEFINITION_H_

/* SD card related */
#define SD_CARD_INTR_GPIO GPIO_NUM_34
#define SD_CARD_INTR_SEL GPIO_SEL_34
#define SD_CARD_OPEN_FILE_NUM_MAX 5

#define ESP_SD_PIN_CLK              -1  // Please configure to your settings
#define ESP_SD_PIN_CMD              -1  // Please configure to your settings
#define ESP_SD_PIN_D0               -1  // Please configure to your settings
#define ESP_SD_PIN_D3               -1  // Please configure to your settings

#define HEADPHONE_DETECT GPIO_NUM_5
#define PA_ENABLE_GPIO GPIO_NUM_21

#define GREEN_LED_GPIO GPIO_NUM_22
#define BLUE_LED_GPIO GPIO_NUM_19

#define BUTTON_REC_ID GPIO_NUM_36
#define BUTTON_MODE_ID GPIO_NUM_13

/* Touch pad related */
#define TOUCH_SEL_SET GPIO_NUM_19  // Please configure to your settings
#define TOUCH_SEL_PLAY GPIO_NUM_23  // Please configure to your settings
#define TOUCH_SEL_VOLUP GPIO_NUM_18  // Please configure to your settings
#define TOUCH_SEL_VOLDWN GPIO_NUM_5  // Please configure to your settings

[... 194 lines cut ...]
int8_t get_input_play_id(void)
{
    return TOUCH_PLAY;
}

int8_t get_input_volup_id(void)
{
    return TOUCH_VOLUP;
}

int8_t get_input_voldown_id(void)
{
    return TOUCH_VOLDWN;
}

// sdcard
int8_t get_sdcard_intr_gpio(void){
    return SD_CARD_INTR_GPIO;
}

int8_t get_sdcard_open_file_num_max(void)
{
    return SD_CARD_OPEN_FILE_NUM_MAX;
}
```

Nexet, 2023-10-09:
> Hi @Nexet ESP-IDF v5.1.1 currently requires ESP-ADF 2.6 or master, but ESP32-A1S-AudioKit is not yet adapted to the latest ESP-ADF and ESP-IDF 5.x.
>
> I modified `board_def.h` and [...]

thanks for the help, were you able to complete on the new version? because I have the same problem with your files too.

PumpkinLin314, 2023-10-11:
Hi @Nexet 
To solve the problem of adapting to the new version, the recommended method is to copy the board support files that are already supported by ESP-ADF and then modify them.

You can copy `$ADF_PATH/components/audio_board/lyrat_v4_3/` to `audio_board` folder and rename it. Then follow the module tutorial to add your board to ESP-ADF.

After that, modify the PINs and codec configuration according to the hardware design. You can also refer to the old version file or existing files in `$ADF_PATH/components/audio_board/`.

closed 2024-07-31 by jason-mao