espressif/esp-adf #1085: Support AW88298 in esp_codec_dev

espzav, 2023-10-02:
Please, is there any plan to add support for AW88298 into `esp_codec_dev`?

It is inside M5Stack CoreS3 https://docs.m5stack.com/en/core/CoreS3 and I would like to add it into BSP.

Thank you.

jason-mao, 2023-10-08:
@espzav  I think we can plan to support this chip.

espzav, 2023-10-09:
@jason-mao Thank you, it will be great, if you can add it into plan :-)

kristiankielhofner, 2023-10-23:
With Willow we've had quite a few users request support for the CoreS3. We've started work on an initial version but support for the AW88298 is definitely necessary to achieve this.

espzav, 2023-10-24:
Hi @kristiankielhofner, thank you, it is great! I have the first draft od BSP for CoreS3 here: https://github.com/espressif/esp-bsp/tree/bsp/m5stack_s3 I will add audio, when it will be available in `esp_codec_dev`. Thank you very much for your help! 
Please, let me know, when it will be available.

closed 2024-02-23 by espressif-bot