espressif/esp-adf #1086: webserver

CYCHEN97, 2023-10-03:
使用ADF时，请问怎么嵌入html，我移植了以前IDF下的webserver到ADF的工程中，嵌入的html好像不起作用，访问ESP32的webserver时没法进入，麻烦帮忙看一下，谢谢

CYCHEN97, 2023-10-04:
@ESP-YJM

CYCHEN97, 2023-10-07:
@ESP-YJM

ESP-YJM, 2023-10-07:
ADF 工程你问下对应负责人吧，你可以加点日志定位下，看看是 html 页面没有编译到固件里还是没有读成功。

CYCHEN97, 2023-10-07:
@jason-mao

CYCHEN97, 2023-10-08:
@jason-mao @krzychb @HengYongChao @tuanpmt @espressif-zhanghu @ahhfzhang
 @ahhfzhang

jason-mao, 2023-10-08:
> ADF 工程你问下对应负责人吧，你可以加点日志定位下，看看是 html 页面没有编译到固件里还是没有读成功。
@CYCHEN97  上述的排查项有啥反馈不？另外单独跑 IDF 对应 example是否成功 ？当前失败的情况下有哪些出错信息？还需要您提供更多的信息以便我们分析问题，谢谢。

CYCHEN97, 2023-10-13:
@jason-mao 
webserver中跟以前idf一样获取嵌入的html文件
```
//发送HTTP响应和运行时生成的html
static esp_err_t http_resp_dir_html(httpd_req_t *req)
{
    /* 获取嵌入式文件 */
    extern const unsigned char WIFI_Config_Script_start[] asm("_binary_WIFI_Config_Script_html_start");
    extern const unsigned char WIFI_Config_Script_end[]   asm("_binary_WIFI_Config_Script_html_end");
    const size_t WIFI_Config_Script_size = (WIFI_Config_Script_end - WIFI_Config_Script_start);

    /* 添加文件上传表单和脚本，执行时发送POST请求到/upload */
    httpd_resp_send_chunk(req, (const char *)WIFI_Config_Script_start, WIFI_Config_Script_size);
    
    httpd_resp_sendstr_chunk(req, NULL);
    return ESP_OK;
}
```
在adf工程的CMakeLists.txt中添加
```
set(COMPONENT_EMBED_FILES   ./wifi/favicon.ico
                            ./wifi/WIFI_Config_Script.html)
```
编译成功了，但是浏览器访问webserver时，进不去html页面，也没有任何报错信息和log输出

之前idf工程的CMakeLists.txt中添加的是
```
idf_component_register(SRCS "main.c"
                    INCLUDE_DIRS "."
                    EMBED_FILES "./wifi/favicon.ico" 
                                "./wifi/WIFI_Config_Script.html")
```
不知道是不是CMakeLists.txt这里面错误导致获取不到html页面，我以前跑idf的工程是没问题的，现在只是想移植过来添加音频功能，麻烦再帮忙看下哈，谢谢大佬。

CYCHEN97, 2023-10-17:
@jason-mao 麻烦帮忙看一下，谢谢大佬

CYCHEN97, 2023-10-17:
@jason-mao

jason-mao, 2023-10-17:
@CYCHEN97 你代码中把WIFI_Config_Script_start 的字符打印出来，可以检验CMakeLists是否编入了。同时可以在httpd_* 的API 中加一些打印，看是否能正确获取到html 的字符。

CYCHEN97, 2023-10-17:
@jason-mao 大佬你好，昨天那个webserver的已经搞定了，是因为设置的socket数量太少了导致服务器没有成功注册

closed 2025-02-18 by jason-mao