espressif/esp-adf #1088: Full duplex I2S

mike-2020, 2023-10-08:
I use ES8388 with ESP32S3. Use ES8388 for both DAC (tts stream) and ADC (speech recognition).

How can I create an I2S emplement supports both reader and writer?

    i2s_stream_cfg_t i2s_cfg = I2S_STREAM_CFG_DEFAULT();
    i2s_cfg.type = AUDIO_STREAM_READER;
    i2s_stream_reader = i2s_stream_init(&i2s_cfg);

    i2s_cfg = I2S_STREAM_CFG_DEFAULT();
    i2s_cfg.type = AUDIO_STREAM_WRITER;
    i2s_stream_writer = i2s_stream_init(&i2s_cfg);

code like above will produce errors:

E (919) i2s(legacy): i2s_check_cfg_validity(926): this i2s port is in use
E (919) i2s(legacy): i2s_driver_install(1582): I2S configuration is invalid

jason-mao, 2023-10-12:
The error may is warning, is it working properly?

mike-2020, 2023-10-13:
Yes, the message confused me, but it is working...

nilesfromm, 2023-10-23:
Hi @mike-2020 I'm working on a project also using a custom board with ESP32S3 + ES8388 + NS4150, I've been trying to set up a minimal project using esp-adf for the past two days and haven't been able to get any audio out on my board. I haven't been able to find if it's a problem with hw or sw yet, any chance you could share the project you have working?

mike-2020, 2023-10-30:
@nilesfromm 
I cannot share my project.
For software, you need create a custom board, and make sure its refereced by ADF. You can clone an existing board definition files, and modify those PINs.

For hardware, you may want to check: (1) ESP32S3's I2S Data out PIN should connect to ES8388's Data in PIN; (2) Give NS4150's Enable PIN a high valtage. If you have oscillograph, you may want to verify if clock/data output/input PINs work as expected.

closed 2024-07-31 by jason-mao