espressif/esp-adf #1089: pipeline_raw_http example doesn't work

patrickkeenan, 2023-10-09:
## Environment

- Audio development kit:      ESP32-LyraT-Mini
- Audio kit version (for ESP32-LyraT/ESP32-LyraT-Mini/ESP32-S3-Korvo-2): v1.2
- [Required] Module or chip used: ESP32-WROVER-E
- [Required] IDF version (run ``git describe --tags`` in $IDF_PATH folder to find it):
    v5.1.1-439-gcb174b0fe1
- [Required] ADF version (run ``git describe --tags`` in $ADF_PATH folder to find it):
    v2.6-23-gef058da
- Build system:         idf.py
- [Required] Running log:         All logs from power-on to problem recurrence
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` in your project folder to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-12.2.0_20230208) 12.2.0
- Operating system:     Mac OS
- Using an IDE?: VSCode
- Power supply:         USB

## Problem Description

Building the pipeline_raw_http example doesn't work. Things I've tried
1. Applying the patch for xtask
2. Setting the memory variables SPIRAM_BOOT_INIT and the other one mentioned
3. Trying a different board (I have 2)
4. Trying with and without headphones

Perhaps this is an issue with the out dated I2S library, but it's quite hard to know. I would assume these examples should work with the latest builds.

### Expected Behavior

The wav file should have audio in it.

### Actual Behavior

Python server receives the http request and its the right length, but its just silence.

### Steps to Reproduce

1. Get the same IDF and ADF versions
2. Use the Lyra T Mini 1.2
3. Try to build the pipeline_raw_http example

## Debug Logs

```
=a60e0h (680160) map[0m
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7100
ho 0 tail 12 room 4
load:0x40078000,len:15604
load:0x40080400,len:4
load:0x40080404,len:3876
entry 0x4008064c
[0;32mI (31) boot: ESP-IDF v5.1.1-439-gcb174b0fe1-dirty 2nd stage bootloader[0m
[0;32mI (31) boot: compile time Oct  9 2023 00:46:34[0m
[0;32mI (33) boot: Multicore bootloader[0m
[0;32mI (37) boot: chip revision: v3.0[0m
[0;32mI (41) boot.esp32: SPI Speed      : 80MHz[0m
[0;32mI (46) boot.esp32: SPI Mode       : DIO[0m
[0;32mI (50) boot.esp32: SPI Flash Size : 4MB[0m
[0;32mI (55) boot: Enabling RNG early entropy source...[0m
[0;32mI (60) boot: Partition Table:[0m
[0;32mI (64) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (86) boot:  2 factory          factory app      00 00 00010000 00200000[0m
[... 113 lines cut ...]
[A[2K
Total bytes written: 32768
[A[2K
Total bytes written: 36864
[A[2K
Total bytes written: 40960
[A[2K
Total bytes written: 45056
[A[2K
Total bytes written: 49152
[A[2K
Total bytes written: 53248
[A[2K
Total bytes written: 57344
[0;31mE (22582) REC_RAW_HTTP: [ * ] [Rec] key released, stop pipeline ...[0m
[A[2K
Total bytes written: 61200
[A[2K
Total bytes written: 64800
[0;32mI (22642) REC_RAW_HTTP: [ + ] HTTP client HTTP_STREAM_POST_REQUEST, write end chunked marker[0m
[0;32mI (22662) REC_RAW_HTTP: [ + ] HTTP client HTTP_STREAM_FINISH_REQUEST[0m
[0;33mW (22662) HTTP_CLIENT: esp_transport_read returned:-1 and errno:128 [0m
[0;32mI (22662) REC_RAW_HTTP: Got HTTP Response = File 20231009T044951Z_16000_16_1.wav was written, size 64800[0m
[0;31mE (78282) REC_R
```

patrickkeenan, 2023-10-09:
Yes, I've ensured both of those things, also, the file is written on the server, so its connecting. Here are the server logs:

Serving HTTP on 192.168.86.22 port 8000
Do Post......
Audio information, sample rates: 16000, bits: 16, channel(s): 1
Total bytes received: 82800
192.168.86.59 - - [09/Oct/2023 09:46:07] "POST /upload HTTP/1.1" 200 -

shootao, 2023-10-20:
Hi @patrickkeenan 
I'm using the same ADF and IDF for testing, and indeed, there is a “**_HTTP_CLIENT: esp_transport_read returned:-1 and errno:128_**” log out in the IDF. The correct approach in IDF should be as follows, but this is just an erroneous log. The server-side does have the correct audio recording data.
```c
diff --git a/components/esp_http_client/esp_http_client.c b/components/esp_http_client/esp_http_client.c
index 59273915ec..b4e8a74c96 100644
--- a/components/esp_http_client/esp_http_client.c
+++ b/components/esp_http_client/esp_http_client.c
@@ -1160,8 +1160,9 @@ int esp_http_client_read(esp_http_client_handle_t client, char *buffer, int len)
                     http_parser_execute(client->parser, client->parser_settings, res_buffer->data, 0);
                     /* ...and lowering the message severity, as closed connection from server side is expected in chunked transport */
                     sev = ESP_LOG_DEBUG;
+                } else if (rlen != ERR_TCP_TRANSPORT_CONNECTION_CLOSED_BY_FIN) {
+                    ESP_LOG_LEVEL(sev, TAG, "esp_transport_read returned:%d and errno:%d ", rlen, errno);
                 }
-                ESP_LOG_LEVEL(sev, TAG, "esp_transport_read returned:%d and errno:%d ", rlen, errno);
             }

             if (rlen == ERR_TCP_TRANSPORT_CONNECTION_TIMEOUT) {
```

PrathamG, 2024-02-02:
Hi! I'm having the same issue with the same board. I'm not sure but it might be something to do with the board itself? Even the google speech example returns an empty response, so it could be that the i2s_read is simply not sending proper audio data for LyraT-Mini-1.2

PrathamG, 2024-02-02:
Okay I was able to solve the issue for the google cloud speech example. I'm fairly certain it's the same issue in this case too. Has to do with LyraT-mini's I2S configuration being different.

See here (https://github.com/espressif/esp-adf/issues/934#issuecomment-1923532342)

closed 2024-07-31 by jason-mao