espressif/esp-adf #1090: Can we use 24bit/192?

vwwwvwww, 2023-10-09:
I found that the new version of ESP-ADF supports 24-bit audio 

and I would like to know if it supports outputting 24-bit/192k (SDCard) audio through I2S ?
and receive & play 24-bit Bluetooth music ?

Another question is, does ESP-ADF only support predefined development boards?

 If I want to use a DIY board, are there any examples available?

jason-mao, 2023-10-12:
Refer to https://github.com/espressif/esp-adf/tree/master/examples/get-started/play_mp3_control/components/my_board to define your board.

I think it can be work on 24bit-192kHz from sdcard. And, the A2DP can't support 24bit music, If I remember correctly.

vwwwvwww, 2023-10-18:
Could you provide a template for using a third-party decoder, similar to the template for "my_board"?

For example, how can I use the following code to decode a WAV file?
                      
                      uint32_t recvSize = 0;
                      do
                      {
                        recvSize =  file.read((void*)buffer, 1024);
                      
                        for(uint32_t i = 0;i<recvSize/2;i++)
                        {
                          partWavData[2*i] = buffer[i];   
                          partWavData[2*i+1] = buffer[i]; 
                        }
                      
                        mi.Write((char*)partWavData, recvSize*2); //i2s_write
                      
                          // startPos += recvSize;
                          // file.seek(startPos);
                      }while(recvSize>0);
                      file.close();

jason-mao, 2024-08-01:
@vwwwvwww Do you have any progress to share？