espressif/esp-adf #1098: Could provide a template for using a third-party decoder, similar to the template for "my_board"?

vwwwvwww, 2023-10-26:
**Is your feature request related to a problem? Please describe.**

When using the default library, we do it like this:

                  //   mp3_decoder_cfg_t mp3_cfg = DEFAULT_MP3_DECODER_CONFIG();
                  //   audio_element_handle_t mp3_decoder = mp3_decoder_init(&mp3_cfg);
                  //   esp_audio_codec_lib_add(player, AUDIO_CODEC_TYPE_DECODER, mp3_decoder);
                  //  //    audio_pipeline_register(pipeline, mp3_decoder , "dec");

When I want to use a third-party decoder:

For example, how can I use the following code to decode a WAV file?

                  uint32_t recvSize = 0;
                  do
                  {
                    recvSize =  file.read((void*)buffer, 1024);
                  
                    for(uint32_t i = 0;i<recvSize/2;i++)
                    {
                      partWavData[2*i] = buffer[i];   
                      partWavData[2*i+1] = buffer[i]; 
                    }
                  
                    mi.Write((char*)partWavData, recvSize*2); //i2s_write
                  
                      // startPos += recvSize;
                      // file.seek(startPos);
                  }while(recvSize>0);
                  file.close();

jason-mao, 2023-10-27:
That's a good idea. I think we can provide an example like this.

vwwwvwww, 2023-10-27:
That's great! I hope to be able to use it soon.