espressif/esp-adf #1100: 运行esp-adf-v2.6\examples\advanced_examples\algorithm  失败

Jackyliszsh, 2023-10-26:
board  esp32s3-box-lite,   idf version 5.1.1 ,adf versionv 2.6
运行此例程，打印如下：
I (1225) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:359, MEM Total:8589532 Bytes, Inter:302663 Bytes, Dram:302663 Bytes

I (1237) AUDIO_ELEMENT: [algo] AEL_MSG_CMD_RESUME,state:1
I (1243) AFE_VC: afe interface for voice communication

I (1249) AFE_VC: AFE version: VC_V220727

I (1254) AFE_VC: Initial auido front-end, total channel: 2, mic num: 1, ref num: 1

I (1262) AFE_VC: aec_init: 1, se_init: 1, vad_init: 0

I (1268) AFE_VC: wakenet_init: 0, voice_communication_agc_init: 0

I (1281) AUDIO_ELEMENT: [wav_encoder] AEL_MSG_CMD_RESUME,state:1
I (1282) AUDIO_PIPELINE: Pipeline started
I (1286) ALGORITHM_EXAMPLES: [7.0] Listen for all pipeline events
I (1291) AFE_VC: mode: 0, (Nov  3 2022 11:49:25)

I (1299) AUDIO_THREAD: The algo_fetch task allocate stack on external memory
I (1361) ALGORITHM_EXAMPLES: [ * ] Receive music info from mp3 decoder, sample_rates=16000, bits=16, ch=1
I (1472) AUDIO_ELEMENT: OUT-[wav_encoder] AEL_IO_DONE,0
W (2013) AFE_VC: ERROR! rb_out slow!!!

W (2029) AFE_VC: ERROR! rb_out slow!!!

W (2031) AFE_VC: ERROR! rb_out slow!!!

一直报这个错误，运行异常。

jason-mao, 2023-10-27:
@Jackyliszsh Same as https://github.com/espressif/esp-adf/issues/873

Jackyliszsh, 2023-10-27:
#873打印的错误跟这个是一样的。但是这是在esp32s3_box_lite的DEMO板上跑的example下的例程也出现这个错误。请指导一下该如何修改才能避免2个CPU之间的负载不均衡？

Jackyliszsh, 2023-10-28:
sorry,  这个错误可能是因为在esp32s3_box_lite上没有SD卡，去掉了fatfs write，pipeline里没取走数据引起的。

closed 2023-10-28 by Jackyliszsh