espressif/esp-adf #1104: components/esp_codec_dev. do not support ac101 anymore ,could you support it again

luyongfugx, 2023-11-04:
components/esp_codec_dev. do not support ac101 anymore ,could you support it again  

Playback | Record
-- | --
ES8311 | Y | Y
ES8388 | Y | Y
ES8374 | Y | Y
ZL38063 | Y | Y
TAS6805M | Y | N
ES7210 | N | Y
ES7243 | N | Y
ES7243E | N | Y
ES8156 | N | Y

jason-mao, 2023-11-21:
@luyongfugx  We implemented AC101 with some limitations, such as no hardware testing. I recommend adding not-supported chips to the [ESP component registry](https://components.espressif.com/) individually.

closed 2023-11-21 by jason-mao

luyongfugx, 2023-11-21:
> @luyongfugx We implemented AC101 with some limitations, such as no hardware testing. I recommend adding not-supported chips to the [ESP component registry](https://components.espressif.com/) individually.

thanks, i have done