espressif/esp-adf #1105: micropython_adf 的korvo2v3 编程出错

tongxincw, 2023-11-08:
- Audio development kit:  ESP32-S3-Korvo-2v3
-IDF version: 🏷️  v5.0.2
- ADF version :  master

使用micropython_adf 按要求配置korvo2v3 编程出错，而lyrat43可正常编译

`/mnt/g/SDK/espressif-SDK/esp-idf/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Executing action: all (aliases: build)
Running cmake in directory /mnt/g/SDK/micropython-SDK/micropython/ports/esp32/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DMICROPY_BOARD_DIR=/mnt/g/SDK/espressif-SDK/esp-adf/micropython_adf/boards/korvo2v3 -DUSER_C_MODULES=/mnt/g/SDK/espressif-SDK/esp-adf/micropython_adf/mod/micropython.cmake -DCCACHE_ENABLE=0 /mnt/g/SDK/micropython-SDK/micropython/ports/esp32"...
-- Found Git: /usr/bin/git (found version "2.25.1") 
-- Component directory /mnt/g/SDK/espressif-SDK/esp-idf/components/fatfs does not contain a CMakeLists.txt file. No component will be added
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/dahanzimin/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/dahanzimin/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/dahanzimin/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
Processing 6 dependencies:
[1/6] espressif/esp-dsp (1.4.8)
[2/6] espressif/esp_tinyusb (1.0.3)
[3/6] espressif/mdns (1.1.0)
[4/6] espressif/nghttp (1.52.0~1)
[5/6] espressif/tinyusb (0.15.0~3)
[6/6] idf (5.0.2)
-- Project sdkconfig file /mnt/g/SDK/micropython-SDK/micropython/ports/esp32/build/sdkconfig
Loading defaults file /mnt/g/SDK/micropython-SDK/micropython/ports/esp32/build/sdkconfig.combined...
/tmp/confgen_tmpf0ll_t80:110 line was updated to CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ_40=n
/tmp/confgen_tmpf0ll_t80:111 line was updated to CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ_80=n
/tmp/confgen_tmpf0ll_t80:112 line was updated to CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ_160=n
/tmp/confgen_tmpf0ll_t80:130 line was updated to CONFIG_SPIRAM_MODE_QUAD=n
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of time_t
-- Check size of time_t - done
-- Found Python3: /home/dahanzimin/.espressif/python_env/idf5.0_py3.8_env/bin/python (found version "3.8.10") found components: Interpreter 
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "micropython" version: v1.21.0-dirty
-- Adding linker script /mnt/g/SDK/micropython-SDK/micropython/ports/esp32/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /mnt/g/SDK/espressif-SDK/esp-idf/components/esp_system/ld/esp32s3/sections.ld.in
-- Adding linker script /mnt/g/SDK/espressif-SDK/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script /mnt/g/SDK/espressif-SDK/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script /mnt/g/SDK/espressif-SDK/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script /mnt/g/SDK/espressif-SDK/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script /mnt/g/SDK/espressif-SDK/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld
-- Adding linker script /mnt/g/SDK/espressif-SDK/esp-idf/components/soc/esp32s3/ld/esp32s3.peripherals.ld
-- Chip is esp32s3, /mnt/g/SDK/espressif-SDK/esp-adf/components/esp-adf-libs
Including User C Module(s) from /mnt/g/SDK/espressif-SDK/esp-adf/micropython_adf/mod/micropython.cmake
Found User C Module(s): adf_module
-- Components: app_trace app_update audio_pipeline audio_sal boards bootloader bootloader_support bt cmock console cxx driver efuse esp-adf-libs esp-sr esp-tls esp_adc esp_app_format esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_netif esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump espressif__esp-dsp espressif__esp_tinyusb espressif__mdns espressif__nghttp espressif__tinyusb esptool_py freertos hal heap http_parser idf_test ieee802154 json log lwip main_esp32s3 mbedtls mqtt newlib nvs_flash openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport touch_element ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa        
-- Component paths: /mnt/g/SDK/espressif-SDK/esp-idf/components/app_trace /mnt/g/SDK/espressif-SDK/esp-idf/components/app_update /mnt/g/SDK/espressif-SDK/esp-adf/components/audio_pipeline /mnt/g/SDK/espressif-SDK/esp-adf/components/audio_sal /mnt/g/SDK/espressif-SDK/esp-adf/micropython_adf/boards /mnt/g/SDK/espressif-SDK/esp-idf/components/bootloader /mnt/g/SDK/espressif-SDK/esp-idf/components/bootloader_support /mnt/g/SDK/espressif-SDK/esp-idf/components/bt /mnt/g/SDK/espressif-SDK/esp-idf/components/cmock /mnt/g/SDK/espressif-SDK/esp-idf/components/console /mnt/g/SDK/espressif-SDK/esp-idf/components/cxx /mnt/g/SDK/espressif-SDK/esp-idf/components/driver /mnt/g/SDK/espressif-SDK/esp-idf/components/efuse /mnt/g/SDK/espressif-SDK/esp-adf/components/esp-adf-libs /mnt/g/SDK/espressif-SDK/esp-adf/components/esp-sr /mnt/g/SDK/espressif-SDK/esp-idf/components/esp-tls /mnt/g/SDK/espressif-SDK/esp-idf/components/esp_adc /mnt/g/SDK/espressif-SDK/esp-idf/components/esp_app_format /mnt/g/SDK/espressif-SDK/esp-idf/components/esp_common /mnt/g/SDK/espressif-SDK/esp-idf/components/esp_eth /mnt/g/SDK/espressif-SDK/esp-idf/components/esp_event /mnt/g/SDK/espressif-SDK/esp-idf/components/esp_gdbstub /mnt/g/SDK/espressif-SDK/esp-idf/components/esp_hid /mnt/g/SDK/espressif-SDK/esp-idf/components/esp_http_client /mnt/g/SDK/espressif-SDK/esp-idf/components/esp_http_server /mnt/g/SDK/espressif-SDK/esp-idf/components/esp_https_ota /mnt/g/SDK/espressif-SDK/esp-idf/components/esp_https_server /mnt/g/SDK/espressif-SDK/esp-idf/components/esp_hw_support /mnt/g/SDK/espressif-SDK/esp-idf/components/esp_lcd /mnt/g/SDK/espressif-SDK/esp-idf/components/esp_local_ctrl /mnt/g/SDK/espressif-SDK/esp-idf/components/esp_netif /mnt/g/SDK/espressif-SDK/esp-idf/components/esp_partition /mnt/g/SDK/espressif-SDK/esp-idf/components/esp_phy /mnt/g/SDK/espressif-SDK/esp-idf/components/esp_pm /mnt/g/SDK/espressif-SDK/esp-idf/components/esp_psram /mnt/g/SDK/espressif-SDK/esp-idf/components/esp_ringbuf /mnt/g/SDK/espressif-SDK/esp-idf/components/esp_rom /mnt/g/SDK/espressif-SDK/esp-idf/components/esp_system /mnt/g/SDK/espressif-SDK/esp-idf/components/esp_timer /mnt/g/SDK/espressif-SDK/esp-idf/components/esp_wifi /mnt/g/SDK/espressif-SDK/esp-idf/components/espcoredump /mnt/g/SDK/micropython-SDK/micropython/ports/esp32/managed_components/espressif__esp-dsp /mnt/g/SDK/micropython-SDK/micropython/ports/esp32/managed_components/espressif__esp_tinyusb /mnt/g/SDK/micropython-SDK/micropython/ports/esp32/managed_components/espressif__mdns /mnt/g/SDK/micropython-SDK/micropython/ports/esp32/managed_components/espressif__nghttp /mnt/g/SDK/micropython-SDK/micropython/ports/esp32/managed_components/espressif__tinyusb /mnt/g/SDK/espressif-SDK/esp-idf/components/esptool_py /mnt/g/SDK/espressif-SDK/esp-idf/components/freertos /mnt/g/SDK/espressif-SDK/esp-idf/components/hal /mnt/g/SDK/espressif-SDK/esp-idf/components/heap /mnt/g/SDK/espressif-SDK/esp-idf/components/http_parser /mnt/g/SDK/espressif-SDK/esp-idf/components/idf_test /mnt/g/SDK/espressif-SDK/esp-idf/components/ieee802154 /mnt/g/SDK/espressif-SDK/esp-idf/components/json /mnt/g/SDK/espressif-SDK/esp-idf/components/log /mnt/g/SDK/espressif-SDK/esp-idf/components/lwip /mnt/g/SDK/micropython-SDK/micropython/ports/esp32/main_esp32s3 /mnt/g/SDK/espressif-SDK/esp-idf/components/mbedtls /mnt/g/SDK/espressif- [... cut ...]
-- Build files have been written to: /mnt/g/SDK/micropython-SDK/micropython/ports/esp32/build
Running ninja in directory /mnt/g/SDK/micropython-SDK/micropython/ports/esp32/build
Executing "ninja all"...
ninja: error: '../sdkconfig', needed by 'srmodels/srmodels.bin', missing and no known rule to make it
ninja failed with exit code 1, output of the command is in the /mnt/g/SDK/micropython-SDK/micropython/ports/esp32/build/log/idf_py_stderr_output_15747 and /mnt/g/SDK/micropython-SDK/micropython/ports/esp32/build/log/idf_py_stdout_output_15747`

crazyxingwenl, 2023-11-30:
我遇到了同样的问题，请问有解决方法吗？

JosephTang, 2023-12-01:
可以将esp-adf/components/esp-sr下的CMakelists.txt中 `esp32s3`的这一段调整为：

```cmake
elseif(${IDF_TARGET} STREQUAL "esp32s3")
    set(include_dirs
        src/include
        esp-tts/esp_tts_chinese/include
        include/esp32s3
        )
    set(srcs
        src/model_path.c
        src/esp_mn_speech_commands.c
        src/esp_process_sdkconfig.c
        )

    set(requires
        json
        spiffs
        )

    IF (IDF_VERSION_MAJOR GREATER 4)
        list(APPEND requires esp_partition)
    ENDIF (IDF_VERSION_MAJOR GREATER 4)

    idf_component_register(SRCS ${srcs}
                       INCLUDE_DIRS ${include_dirs}
                       REQUIRES ${requires}
                       PRIV_REQUIRES spi_flash)

    target_link_libraries(${COMPONENT_TARGET} "-L ${CMAKE_CURRENT_SOURCE_DIR}/lib/esp32s3")
    target_link_libraries(${COMPONENT_TARGET} "-L ${CMAKE_CURRENT_SOURCE_DIR}/esp-tts/esp_tts_chinese/esp32s3")

    add_prebuilt_library(esp_audio_processor "${CMAKE_CURRENT_SOURCE_DIR}/lib/esp32s3/libesp_audio_processor.a" PRIV_REQUIRES ${COMPONENT_NAME})
    add_prebuilt_library(wakenet "${CMAKE_CURRENT_SOURCE_DIR}/lib/esp32s3/libwakenet.a" PRIV_REQUIRES ${COMPONENT_NAME})
    add_prebuilt_library(multinet "${CMAKE_CURRENT_SOURCE_DIR}/lib/esp32s3/libmultinet.a" PRIV_REQUIRES ${COMPONENT_NAME})
    add_prebuilt_library(esp_audio_front_end "${CMAKE_CURRENT_SOURCE_DIR}/lib/esp32s3/libesp_audio_front_end.a" PRIV_REQUIRES ${COMPONENT_NAME})
    add_prebuilt_library(hufzip "${CMAKE_CURRENT_SOURCE_DIR}/lib/esp32s3/libhufzip.a" PRIV_REQUIRES ${COMPONENT_NAME})

    idf_component_get_property(esp_dsp_lib espressif__esp-dsp COMPONENT_LIB)

    target_link_libraries(${COMPONENT_TARGET} "-Wl,--start-group"
        hufzip
        dl_lib
        fst
        c_speech_features
        $<TARGET_FILE:${esp_dsp_lib}>
        esp_audio_front_end
        esp_audio_processor
        multinet
        esp_tts_chinese
        voice_set_xiaole
        wakenet
        "-Wl,--end-group")

    set(MVMODEL_EXE ${COMPONENT_PATH}/model/movemodel.py)
    idf_build_get_property(build_dir BUILD_DIR)
    set(image_file ${build_dir}/srmodels/srmodels.bin)

    partition_table_get_partition_info(size "--partition-name model" "size")
    partition_table_get_partition_info(offset "--partition-name model" "offset")

    if("${size}" AND "${offset}")
        add_custom_command(
            OUTPUT ${image_file}
            COMMENT "Move and Pack models..."
            COMMAND python ${MVMODEL_EXE} -d1 ${PROJECT_DIR} -d2 ${COMPONENT_PATH} -d3 ${build_dir}
            DEPENDS ${PROJECT_DIR}/sdkconfig
            VERBATIM)

        add_custom_target(srmodels_bin ALL DEPENDS ${image_file})
        add_dependencies(flash srmodels_bin)

        esptool_py_flash_to_partition(flash "model" "${image_file}")
    else()
        set(message "Failed to find model in partition table file"
                    "Please add a line(Name=model, Size>recommended size in log) to the partition file.")
    endif()
elseif(${IDF_TARGET} STREQUAL "esp32s2")

```

crazyxingwenl, 2023-12-01:
感谢回复，korvo2v3工程已成功编译。在使用音频MicroPython例程时，发现执行代码：import audio，会提示 audio module 不存在。github 的issus中，无相关的问题做参考。查看了工程在 esp-adf/micropython_adf/mod下，是有audio模块相关函数的定义，但是没有参与编译，是否需要在什么地方做编译配置吗？望回复，谢谢。

疯子
***@***.***

 

 elseif(${IDF_TARGET} STREQUAL "esp32s3")     set(include_dirs         src/include         esp-tts/esp_tts_chinese/include         include/esp32s3         )     set(srcs         src/model_path.c         src/esp_mn_speech_commands.c         src/esp_process_sdkconfig.c         )     set(requires         json         spiffs         )     IF (IDF_VERSION_MAJOR GREATER 4)         list(APPEND requires esp_partition)     ENDIF (IDF_VERSION_MAJOR GREATER 4)     idf_component_register(SRCS ${srcs}                        INCLUDE_DIRS ${include_dirs}                        REQUIRES ${requires}                        PRIV_REQUIRES spi_flash)     target_link_libraries(${COMPONENT_TARGET} "-L ${CMAKE_CURRENT_SOURCE_DIR}/lib/esp32s3")     target_link_libraries(${COMPONENT_TARGET} "-L ${CMAKE_CURRENT_SOURCE_DIR}/esp-tts/esp_tts_chinese/esp32s3")     add_prebuilt_library(esp_audio_processor "${CMAKE_CURRENT_SOURCE_DIR}/lib/esp32s3/libesp_audio_processor.a" PRIV_REQUIRES ${COMPONENT_NAME})     add_prebuilt_library(wakenet "${CMAKE_CURRENT_SOURCE_DIR}/lib/esp32s3/libwakenet.a" PRIV_REQUIRES ${COMPONENT_NAME})     add_prebuilt_library(multinet "${CMAKE_CURRENT_SOURCE_DIR}/lib/esp32s3/libmultinet.a" PRIV_REQUIRES ${COMPONENT_NAME})     add_prebuilt_library(esp_audio_front_end "${CMAKE_CURRENT_SOURCE_DIR}/lib/esp32s3/libesp_audio_front_end.a" PRIV_REQUIRES ${COMPONENT_NAME})     add_prebuilt_library(hufzip "${CMAKE_CURRENT_SOURCE_DIR}/lib/esp32s3/libhufzip.a" PRIV_REQUIRES ${COMPONENT_NAME})     idf_component_get_property(esp_dsp_lib espressif__esp-dsp COMPONENT_LIB)     target_link_libraries(${COMPONENT_TARGET} "-Wl,--start-group"         hufzip         dl_lib         fst         c_speech_features         $<TARGET_FILE:${esp_dsp_lib}&gt;         esp_audio_front_end         esp_audio_processor         multinet         esp_tts_chinese         voice_set_xiaole         wakenet         "-Wl,--end-group")     set(MVMODEL_EXE ${COMPONENT_PATH}/model/movemodel.py)     idf_build_get_property(build_dir BUILD_DIR)     set(image_file ${build_dir}/srmodels/srmodels.bin)     partition_table_get_partition_info(size "--partition-name model" "size")     partition_table_get_partition_info(offset "--partition-name model" "offset")     if("${size}" AND "${offset}")         add_custom_command(             OUTPUT ${image_file}             COMMENT "Move and Pack models..."             COMMAND python ${MVMODEL_EXE} -d1 ${PROJECT_DIR} -d2 ${COMPONENT_PATH} -d3 ${build_dir}             DEPENDS ${PROJECT_DIR}/sdkconfig             VERBATIM)         add_custom_target(srmodels_bin ALL DEPENDS ${image_file})         add_dependencies(flash srmodels_bin)         esptool_py_flash_to_partition(flash "model" "${image_file}")     else()         set(message "Failed to find model in partition table file"                     "Please add a line(Name=model, Size&gt;recommended size in log) to the partition file.")     endif() elseif(${IDF_TARGET} STREQUAL "esp32s2")

JosephTang, 2023-12-01:
编译时有加`-D USER_C_MODULES=${ADF_PATH}/micropython_adf/mod/micropython.cmake`吗

crazyxingwenl, 2023-12-01:
现在可以了，有个文件出了问题，感谢回复

疯子
***@***.***

closed 2023-12-19 by espressif-bot